tock/tock #1922: [CI] Failed to find source foo.rs in "CI-Job: Debug Support Targets"

gendx, 2020-06-09:
When running `make ci-nosetup` (after `make clean` to be sure), the output of "CI-Job: Debug Support Targets" contains plenty of warnings looking like `$HOME/.rustup/toolchains/nightly-2020-06-03-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump: warning: '/tock/target/thumbv7em-none-eabi/release/nrf52dk.elf': failed to find source arch/cortex-m4/src/lib.rs`.

I don't remember seeing any of these before the recent CI refactorings.

```
********************************************************************************
*  CI-Job: Debug Support Targets                                               *
********************************************************************************

# These are rules that build additional debugging information, but are
# also quite time consuming. So we want to verify that the rules still
# work, but don't build them for every board.
#
# The choice of building for the nrf52dk was chosen by random die roll.
make[2]: Entering directory '/tock/boards/nordic/nrf52dk'
   Compiling nrf52dk v0.1.0 (/tock/boards/nordic/nrf52dk)
    Finished release [optimized + debuginfo] target(s) in 3.83s
   text	   data	    bss	    dec	    hex	filename
  92672	   1244	  39716	 133632	  20a00	/tock/target/thumbv7em-none-eabi/release/nrf52dk
$HOME/.rustup/toolchains/nightly-2020-06-03-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump: warning: '/tock/target/thumbv7em-none-eabi/release/nrf52dk.elf': failed to find source arch/cortex-m4/src/lib.rs
```

ppannuto, 2020-06-14:
:(

I don't think this actually has anything to do with the CI infrastructure (which is running the same commands it was before); rather I think it's an artifact of the nightly bump. There were some non-trivial changes to llvm-objdump.

@brghena, any chance this has a similar / shared root cause to what you've been debugging lately?

brghena, 2020-06-15:
I don't think it's related. The objdump issue Emilio was running into has to do with large bin file sizes due to SRAM addresses starting before Flash addresses on his particular chip.

brghena, 2020-06-15:
Oops. To clarify, it was an `objcopy` issue, not objdump.

ppannuto, 2020-06-15:
Okay, I've looked into this a little bit more.

This has been a problem for a while, but this warning was merged into llvm on Aug 2019 (https://reviews.llvm.org/D62462), which is why it's now surfacing.

The warnings are correct that there are paths listed in the ELF file that it can't find:

> /Users/ppannuto/.rustup/toolchains/nightly-2020-06-03-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/bin/llvm-objdump: warning: '/Volumes/code/helena-project/tock/target/thumbv7em-none-eabi/release/nrf52dk.elf': **failed to find source arch/cortex-m4/src/lib.rs**

This one is perhaps our fault for rewriting paths during compilation to make panic strings shorter.

> failed to find source /cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.31/src/mem.rs

This one and ones like it *may* also be that problem.

gendx, 2020-06-16:
> The warnings are correct that there are paths listed in the ELF file that it can't find:
>
>
> This one is perhaps our fault for rewriting paths during compilation to make panic strings shorter.

That makes sense. The rationale for rewriting paths is both making panic strings shorter and making builds deterministic. Maybe these debug rules should depend on a separate non-deterministic build where `--remap-path-prefix` is not used? Or maybe objdump would work with a relative path, i.e. `./arch/cortex-m4/src/lib.rs`?

ppannuto, 2020-06-16:
Hmm..., this seems to be a shortcoming of llvm-objdump. [gnu objdump](https://linux.die.net/man/1/objdump) includes a `--prefix` option to fixup source paths, [llvm-objdump has no equivalent option](https://llvm.org/docs/CommandGuide/llvm-objdump.html).

[Several of the other tools in the llvm family](https://llvm.org/docs/genindex.html) seem to have options that do what we might need, so maybe they're amenable to adding. I'll look into asking upstream.

ppannuto, 2020-06-17:
Updates:

bad news: llvm-objdump doesn't currently support what we need
good news: reply indicated that they should and asked me to file an enhancement bug

upstream bug: https://bugs.llvm.org/show_bug.cgi?id=46368

I'm cautiously optimistic that this will be a quick fix for the llvm team, but realistically it will take a while to bubble down to the nightly toolchain. Not sure if we want to do anything locally in the short term to mitigate the warnings; I'm kind of inclined to say they're okay for now.

gendx, 2020-06-18:
Thanks for the investigation! The warnings are not so annoying per se, given that there is an explanation and a path to fix it at some point.

hudson-ayers, 2021-01-06:
It looks like the LLVM bug was fixed on October 16, but its not clear to me when Rust will update to a new enough version of LLVM to include that fix.

hudson-ayers, 2021-03-05:
Now that rust nightly is on LLVM 12, I think that updating to a newer nightly will fix this.

hudson-ayers, 2021-03-16:
Using nightly 2021-03-15, disassembling with `arm-none-eabi-objdump -D -S imix.elf` successfully shows tock source code inline with the assembly. Using a nightly from before Rust upgraded to LLVM 12, that was not the case.

ppannuto, 2021-03-24:
Continuing to close the loop on this, I just got the ping that the upstream llvm bug is now considered resolved: https://bugs.llvm.org/show_bug.cgi?id=46368 -- though I think we already had evidence of the from the new nightly.

bradjc, 2021-03-25:
Do we need to change anything, or is this fixed since we are on nightly nightly-2021-03-19?

hudson-ayers, 2021-03-29:
The warnings originally cited as justification for this issue still show up in CI (https://github.com/tock/tock/runs/2203324015). 

On one hand, the .elf files do point back to actual source (if you run objdump from the `tock/` root, source paths show up). On the other hand, we have not started using the `--prefix` option that was added in that bug report Pat filed, and I can't figure out how to use it to get rid of the warnings. Even if you run objdump from the `tock/` root, there are still other warnings for some rust files not being found (`/home/hudson/.rustup/toolchains/nightly-2021-03-19-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump: warning: 'imix.elf': failed to find source /cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.39/src/int/udiv.rs
`). Using `--prefix` does not seem to work here because `objdump` is looking for `cargo` but the file is actually in `.cargo`.