tock/tock #1923: [Tracking] Convert from llvm_asm!() to new asm!()

bradjc, 2020-06-09:
With a new `asm!()` [macro in Rust nightly](https://blog.rust-lang.org/inside-rust/2020/06/08/new-inline-asm.html) that has a path to stabilization, we should switch our uses of `llvm_asm` to the new format. This is part of #1654.

- [ ] arch/cortex-m0
- [ ] arch/cortex-m3
- [ ] arch/cortex-m4
- [ ] arch/rv32
- [ ] chips
- elsewhere?

gendx, 2020-06-09:
This is really great given that inline assembly is likely the main reason to use Rust nightly at the moment. And the new macro seems to be getting quite a bit of traction so we can expect it to stabilize in the medium term.

From a programming point of view this is IMO much cleaner than having to link external `.S` files, and if I understood correctly inline `asm!` also allows the compiler to apply more optimizations than with raw `.S` files being linked blindly at the linker step.

alistair23, 2020-06-12:
https://github.com/tock/tock/pull/1926 is a step in this direction and I think completes the swap for `libraries/riscv-csr``

hudson-ayers, 2021-01-02:
With the outstanding PRs merged, there are only 4 uses of `llvm_asm` remaining (ignoring all the uses in `cortex-m0`)

lschuermann, 2021-01-06:
As indicated in #2308, once that change is in, I'll do the migration of the rest of the `rv32i` crate.

bradjc, 2021-03-12:
Woo!

```
$ git grep llvm_asm | wc -l
       0
```

closed 2021-03-12 by bradjc