tock/tock #1937: Host OpenTitan boot rom?

ppannuto, 2020-06-15:
#1935 raised a good point that pointing to a random azure cloud build link isn't terribly reliable.

**Proposal:** We host a copy of the image on www.tockos.org.

I think the way to do this is to use git-lfs. Netlify has first-class support for large artifacts stored in LFS being served (i.e. I do this for ppts and videos on my personal website). It works great.

Holler if folks have other thoughts?

Attn: @alistair23

alistair23, 2020-06-15:
Does the license for the boot ROM allow this? My main concern was that we wouldn't be allowed to duplicate the boot ROM.

ppannuto, 2020-06-15:
@tock/opentitan-wg Any thoughts on this? Either Tock hosting a copy or providing something more canonical than an Azure build link for those who need images?

gkelly, 2020-06-15:
So long as the terms of the license are respected I don't think there are any issues. Currently the boot ROM isn't nearly as full-featured as it's going to end up being, and isn't changing very rapidly right now. We're looking to spin up development in about a month's time and we can start discussing what a reasonable release mechanism is then.

ppannuto, 2020-06-15:
Okay, I have to confess some OT ignorance here: what I know of this boot ROM is that you can download it from a magic Azure link Alistair has provided and (from a quick search) its documentation is here: https://docs.opentitan.org/sw/device/boot_rom/README/ (?). What license / potential license issues do we need to be looking at here?

gkelly, 2020-06-15:
The boot ROM code is Apache 2.0, so we're very aligned with the Tock side of things. The azure link you're getting is probably just a CI link that includes a boot ROM image as part of the normal build flow. We'll probably switch to tagged releases when the boot ROM stabilizes, and so there will be a stable GCS link for the artifacts.

bradjc, 2020-06-16:
I fully support this. I also think we should add the FPGA bitstream as well.

ppannuto, 2020-06-16:
Can you drop a link / copy of the bitstream here? I need to set up some things on the netlify side, but I'll do them together in a little bit.

lschuermann, 2020-06-16:
Has someone considered [Zenodo](https://zenodo.org/), maybe if only as a secondary mirror?
Given OpenTitan's nature and Tock's role in research I think this could be a reasonable platform.

ppannuto, 2020-06-16:
I've happily used Zenodo for research stuff before (indeed, it backs the datasets for [DATA](https://workshopdata.github.io/DATA2019/)). I probably don't want to upload someone else's artifact there though.

Happy to point to artifacts there if OT folks are willing to publish them that way.

hudson-ayers, 2020-09-29:
Relying on the OpenTitan CI artifact for our CI is continuing to cause issues, most recently in libtock-rs as the disappearing artifacts force libtock-rs to update to the latest upstream Tock submodule in order to run the QEMU tests. From reading this thread it doesn't sound like hosting a copy of the boot ROM on TockOS should be an issue, I think it would be great if we could move forward with that approach.

alistair23, 2020-09-29:
Just to clarify, to host it in the Tock repo?

brghena, 2020-09-29:
I believe the thinking was to host it in the Tock website somewhere: https://github.com/tock/tock-www. That way we could reliably access it and update it if needed through PRs. But also people who don't need it don't have to download it with the repo clone.

jrvanwhy, 2020-10-07:
We plan to host a copy of the Mask ROM (previously called the boot ROM) in a GCS bucket. I'll send a PR when that is available to point Tock at the GCS bucket rather than OpenTitan's CI artifacts. That should resolve the CI reliability issues currently caused by the boot ROM.

hudson-ayers, 2020-10-21:
Closed by #2170 🎉

closed 2020-10-21 by hudson-ayers