tock/tock #1940: Tacking: USB Stack

bradjc, 2020-06-15:
The USB stack is a work-in-progress. Some identified TODOs:

 - [ ] [This one should be addressed by adding an explicit `panic!`](https://github.com/tock/tock/pull/1902#discussion_r436592017) -- since it will panic either way, better to have something explicit than it happening from an array access.
 - [ ] [This one should be added to a tracking issue so we don't lose it](https://github.com/tock/tock/pull/1902#discussion_r436591970)
 - [ ] [This one could either be addressed by a comment, or added to a tracking issue](https://github.com/tock/tock/pull/1902#discussion_r435800117)

brghena, 2020-06-15:
Some thoughts I wrote up a couple weeks ago about on a rewrite of `usbc_client_ctrl.rs` and the USB HIL that might be useful for the future.

Goal: control_handler capsule for USB
  * All USB devices must support endpoint zero, which is a control endpoint
  * USB devices do not have multiple control endpoints: https://stackoverflow.com/questions/22417493/what-is-the-use-of-multiple-control-endpoints-non-ep0
  * control_handler should:
	* take a descriptor setup from the higher layer
	* automatically handle control requests from lower layer
	* automatically handle things without an application needing to exist
  * control_handler may:
    * send requests it doesn't know how to handle up the chain (if the higher layer wants them)
  * control_handler should not:
    * get in the way of handling the other (non-control) end points

Questions:
* Should the HIL have two interfaces, one for endpoints and one for control?
  * A good answer seems to me to be have one HIL interface, then one layer on top of it that handles the control endpoint and multiplexes. That layer's interface would be descriptors and endpoints.
* How would a board combine two desired USB capsules: e.g. CDC and HID or similar?
  * Can they be split into configurations?
    * No. The host picks one configuration that the entire device acts as, so a single configuration is best.
  * Ideally, they would be mutiplexed on top of the control layer that lives on top of the HIL. For now, they can be combined into a single capsule manually.
* Do the normal endpoints need to live on top of control, or in parallel?
  * They will need to provide descriptor configuration information to the control capsule.
  * The control capsule is not necessary to be between the HIL and something using bulk, interrupt, or isochronous endpoints.
* Why is the current interface for endpoints on top of buffers of VolatileCells? Can't it just be `[u8]`?
  * Existing USB capsules have a _very_ different interface from most of Tock. It seems that in the current design, the endpoint buffers are shared downwards once and then both sides hold on to references, using VolatileCell to make that work.

bradjc, 2020-06-15:
I would like to see the HIL split up so that there is a "down" path and a separate callback "up" path. Combining them tends to make things imperceptibly confusing (i.e. you can tell that things seem confusing and difficult to follow, but you can't quite pinpoint why).

namyoonw, 2020-09-02:
Questions: Is USB capsule be configurable?
- Can any project add more or less (non-control) endpoints than default ones in usb capsules?
- Can any project change DeviceDescriptor contents, like vendor ID, product ID, or strings?

brghena, 2020-09-02:
We discussed this on slack a little. The answer is yes, but also it's still pretty new and not necessarily easy to use yet.

The [CDC capsule](https://github.com/tock/tock/blob/master/capsules/src/usb/cdc.rs) is a good example of a capsule that has its own descriptors and endpoints. And it's used by the [nano33_ble](https://github.com/tock/tock/blob/master/boards/nano33ble/src/main.rs#L207) which creates strings and IDs for the product.

alistair23, 2020-09-03:
Also, what is the plan for supporting multiple ConfigurationDescriptors? I just opened a PR to star CTAP support for Tock (https://github.com/tock/tock/pull/2094) it would be great to support both CTAP and CDC at the same time.

alistair23, 2020-09-11:
I added two items to the TODO list