tock/tock #1944: Make Tock source code citable (Zenodo)

lschuermann, 2020-06-16:
Given Tock's (traditional) nature as a research operating system - it does have quite a few unique properties - I would like to be able to reliably cite or point to Tock code in scientific work.

[Zenodo](https://zenodo.org/) is a service funded by the European Union and CERN, and allows publishing arbitrary (scientific) material and making it available via a DOI. More importantly, it has a GitHub integration which automatically publishes new releases to Zenodo and assigns them a DOI.

This seems rather easy to integrate and could help people to refer to Tock source code and documentation reliably (in terms of availability & future proofing) in a widely accepted way.

Actually I suffered from this issue myself already: in one of my theses, explaining memory safety enforcement and therefore pointing towards a large piece of kernel code by Git revision hash was one of the main concerns of my advisors.

I therefore propose to integrate with Zenodo. I would like to keep this discussion free of the individual opinions regarding whether source code is acceptable literature material, or the quality of such in a scientific context; instead focusing on the organizational and technical concerns.

ppannuto, 2020-06-16:
https://guides.github.com/activities/citable-code/

Well that's pretty slick.

I've used zenodo happily for my own research stuff before, I'm an easy +1 on this.

bradjc, 2020-06-16:
One very selfish concern is I (and perhaps other authors as well) would rather have citations go to the SOSP'17 paper. Citing a particular line of code or a module is a very different story, but if someone just needs to cite "Tock" in general, citing the code isn't helpful in a very selfish, academic sort of way.

ppannuto, 2020-06-16:
We could add an explicit "how to cite" to the Readme -- those are fairly common. Would hopefully cover most cases?

lschuermann, 2020-06-16:
> One very selfish concern is I (and perhaps other authors as well) would rather have citations go to the SOSP'17 paper

I feel your concerns and if I had papers published somewhere I would probably feel very much the same. However I do think that the Tock source contains cite-worthy (or at least reference worthy) parts not explicitly mentioned in other conference papers. If an author deems these parts necessary to explain a certain concept or reason about why their concept is actually safe/secure/... in the context of Tock, they(I) will find ways to do that (as I have). This is an effort to _formally_ improve these specific sorts of citations.

Every scientific document I've written involving Tock has explained the high level concepts using the conference papers mentioned on the website. I don't think that can be substituted by code. The Markdown-documentation could probably be cited content-wise, but the authors must judge regarding quality (I would be careful here).

In the end I'm not the one to decide. I like @ppannuto's "how to cite" approach, but if this will make people investing time in Tock from a scientific background uncomfortable, I understand as well.

lschuermann, 2020-10-09:
*ping*

I'm still interested in this, since it'd be a low-hanging fruit given all concerns are addressed. I'm going to write an upcoming thesis which will likely look into the scheduler interface and driver architecture, hence this would be very welcome.

If the concerns raised are still issues and couldn't be solved by something like a "how to cite" guide, this can be closed.

ppannuto, 2020-10-11:
In the short term, I guess I don't see much harm in turning this on and but not advertising it explicitly. Those who really need to cite a Tock release will be able to find a mechanism to do so, most folks will still likely find the academic paper, which could promote more explicitly in the top-level readme.

lschuermann, 2020-10-23:
Works like a charm, the 1.6 tag now [has a DOI](https://doi.org/10.5281/zenodo.4124184)! Thanks for integrating this. :+1:

closed 2020-10-23 by lschuermann