tock/tock #1948: [arch, rv32i] Compilation error with "opt-level" set to 0 or 1

silvestrst, 2020-06-17:
I wasn't sure whether this is a known issue, I searched in open issues and couldn't find anything (maybe missed it).

```
V=\m/ make

PLATFORM = opentitan
TARGET = riscv32imc-unknown-none-elf
TOCK_KERNEL_VERSION = release-1.5-rc3-500-g7cf62ee3
TOOLCHAIN = "/home/svt/.rustup/toolchains/nightly-2020-06-03-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm"
cargo --version      = cargo 1.45.0-nightly (9fcb8c1d2 2020-05-25)                                                                                                                             
rustc --version      = rustc 1.45.0-nightly (fe10f1a49 2020-06-02)                                                                                                                             
rustup --version     = rustup 1.21.1 (2020-02-23)
...
error: invalid operand for inline asm constraint 'i'

error: aborting due to previous error

error: could not compile `opentitan`.
```

I also ran a downstream fork of Tock, and got a more specific error code:
```
error: invalid operand for inline asm constraint 'i'

error: aborting due to previous error

error: could not compile `rv32i`. 
```

The reason why I wanted to use `opt-level = 0` to have a run through some code with GDB, but it seems that a lot of functionality is optimised out:
```
95                      let bytes_to_send = total_bytes_to_send - total_bytes_sent;
(gdb) p bytes_to_send
$1 = <optimized out>
(gdb) p total_bytes_to_send
$2 = <optimized out>
(gdb) p total_bytes_sent
$3 = <optimized out>
```

lenary, 2020-06-17:
Which file is causing this issue?

silvestrst, 2020-06-17:
It doesn't really say:
```
error: invalid operand for inline asm constraint 'i'

error: aborting due to previous error

error: could not compile `opentitan`.

Caused by:
  process didn't exit successfully: `rustc --crate-name opentitan --edition=2018
boards/opentitan/src/main.rs --error-format=json --json=diagnostic-rendered-ansi
--crate-type bin --emit=dep-info,link -C panic=abort -C lto -C debuginfo=2 -C
link-arg=-L/home/svt/development/repos/tock/boards/opentitan -C debug-assertions=off
-C metadata=04ef4edb3cde02bd -C extra-filename=-04ef4edb3cde02bd --out-dir
/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps
--target riscv32imc-unknown-none-elf -L
dependency=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps
-L dependency=/home/svt/development/repos/tock/target/release/deps
--extern capsules=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps/libcapsules-d782f66ff5c6ec77.rlib
--extern components=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps/libcomponents-f49c66d2850fbc5a.rlib
--extern ibex=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps/libibex-f60301066f06a811.rlib
--extern kernel=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps/libkernel-25c504c2f12f561a.rlib
--extern lowrisc=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps/liblowrisc-a73e336efe21042a.rlib
--extern rv32i=/home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/deps/librv32i-da166b711ff3e739.rlib
-C link-arg=-Tlayout.ld -C linker=rust-lld -C linker-flavor=ld.lld
-C relocation-model=dynamic-no-pic -C link-arg=-zmax-page-size=512 -C link-arg=-icf=all -C force-frame-pointers=no
--remap-path-prefix=/home/svt/development/repos/tock/=` (exit code: 1)
make: *** [../Makefile.common:268: /home/svt/development/repos/tock/target/riscv32imc-unknown-none-elf/release/opentitan] Error 101
```
When `arch/rv32i` is built with `cargo build` directly, it ignores local profiles:
`warning: profiles for the non root package will be ignored, specify profiles at the workspace root:`

lenary, 2020-06-17:
The only place with an `i` constraint is `https://github.com/tock/tock/blob/2c1d35179cfb3f546b8552c078fb82712c13ab55/libraries/riscv-csr/src/csr.rs#L36-L46`, and it would seem that `self.value` is probably not a constant at -O1. 

I'm not sure how much we can change at `opt-level = 0` - the code requires a certain amount of constant propagation that is presumably just not happening at that optimisation level.

ppannuto, 2020-06-17:
This might actually be fixed by #1926, specifically this bullet:

> - Uses the new inline assembly. Unfortunately Rust won't let use access CSR dynamically like we used to (see rust-lang/rust#73220) so we now have a giant if statement to see which one to access. I couldn't see any other way to make the value const.

I'm wondering if what was there before was never really valid.

hudson-ayers, 2020-06-17:
I managed to "fix" this by moving `ReadWriteRiscvCsr` to use const generics to ensure the value will always be stored as a const. That allows building with opt-level=0 or 1, but it still eventually fails because the resulting binary does not fit in ROM.

hudson-ayers, 2020-06-17:
A link to my const-generics-csr patch, if you want to try to use it for debugging: https://github.com/hudson-ayers/tock/commit/ab4965cea5270ee3b5515513669b118a357ee7b6

lenary, 2020-06-17:
@hudson-ayers neat, I thought there might be a solution like that. Is the code-size problem just that it's causing the csr access functions to all be duplicated for each const parameter?

hudson-ayers, 2020-06-17:
@lenary I don't think so, I think that even without any csr code it probably would not fit, just because of using an optimization level not focused on size.

I don't think the const generics version should really be that different than what was working thanks to const propagation - the functions are marked inline, so the code for each function should be getting inlined into the caller's code regardless of the use of const generics

silvestrst, 2020-07-02:
> A link to my const-generics-csr patch, if you want to try to use it for debugging: hudson-ayers@ab4965c

Thank you @hudson-ayers

hudson-ayers, 2020-07-06:
I think the specific problem in this issue was fixed by #1926 , and the issue with the code not fitting because of size is more fundamental. Going to close this for now, but if I am wrong feel free to reopen.

closed 2020-07-06 by hudson-ayers