tock/tock #1954: [CI] Something is broken with CI + Clippy running locally

ppannuto, 2020-06-19:
While #1760 is failing in the cloud: https://github.com/tock/tock/pull/1760/checks?check_run_id=786552075

It's working locally:
```
[-bash] Thu 18 Jun 21:22 [[nikulshr-master $] /Volumes/code/helena-project/tock]
$ make ci-job-clippy

********************************************************************************
* CI-Job: Clippy                                                               *
********************************************************************************

    Finished dev [optimized + debuginfo] target(s) in 0.12s
```

My guess is that I have an out-of-date clippy on my machine, but will need to actually investigate what's going on at some point :/

bradjc, 2020-12-03:
I assume this issue has been resolved.

closed 2020-12-03 by bradjc

hudson-ayers, 2020-12-03:
Oh, I semi-regularly observe that `make clippy` will pass unless I run `make clean` and then `make clippy`, and then the error manifests. I assume this is what Pat ran into.

This is a clippy issue, not a Tock issue, and is tracked here: https://github.com/rust-lang/rust-clippy/issues/4612 .