tock/tock #1966: Question for add sdcard support on board stm32f446re

takoyakiyo, 2020-06-20:
I tried to add 'sdcard capsule' on board `stm32f446re` to read and write a sdcard. I modified the code of `main.rs` and `spi.rs` to enable the `sdcard capsule`, and used a 32G sdcard for testing. But I found that the read data lost a few bits, so it caused a misalignment. The reading result is as follows:
| Data Source  | Hex  | Binary  |
|---|---|---|
| real  | eb58904d53444f53352e30000220820c | 11101011010110001001000001001101010100110100010001001111010100110011010100101110001100000000000000000010001000001000001000001100 |
| tock  | dac4826a9a227a99a971800011041060 | xxx11011010110001001000001001101010100110100010001001111010100110011010100101110001100000000000000000010001000001000001000001100000 |

What is the reason for this situation？

brghena, 2020-06-20:
Unfortunately, there are several possibilities for where the problem could be coming from. My very high-level guess is that the SPI driver is failing to receive the first few bits for some reason. Maybe it doesn't respond quickly enough? But that's just a possibility.

If it's not an issue with the SPI driver, then it could be with the SD Card capsule itself. The capsule needs to wait an arbitrary amount of time for data to be ready from the SD Card. It checks for data every 1 ms [in a big state machine here](https://github.com/tock/tock/blob/master/capsules/src/sdcard.rs#L789). That definitely worked fine for the SD Cards I was testing on, but maybe it's not good enough for the SD Card you're on for some reason?

To debug this kind of an issue, I would first try to get a logic analyzer connected to the SPI wires and see what's actually going over the wire. If the correct data is on the wires, the next thing to determine is whether the SPI is reading those bits at the correct time. Next, I would try to add some `debug!()` calls into the SPI driver and see if it is receiving all the same bits you are seeing on the physical SPI wires. It's hard to tell if the SPI driver received at the wrong time due to an implementation issue or because the SD Card capsule told it to start reading at the wrong time. But my first guess would be an issue with the SPI driver since the SD Card capsule was known to be working on another chip at one point.

Sorry, that's not a simple answer, but unfortunately you're not having a simple problem.

bradjc, 2020-12-03:
Please re-open if any updates.

closed 2020-12-03 by bradjc