tock/tock #1972: Debugging over USB serial

brghena, 2020-06-22:
For platforms with only USB virtual consoles, we need to support debugging over that interface.

On a panic, we traditionally use the synchronous UART to print debugging information. However, this synchronous method seems very hard to impossible to implement for USB CDC. Crashes in applications should still be debuggable since the kernel is trustworthy, but would require some changes in the kernel for how they are handled. Crashes in the kernel are a harder question.

ppannuto, 2020-06-22:
I don't think the goal of the panic / emergency path is usually to be terribly efficient. While I certainly don't have the whole state machine in my head, couldn't the synchronous write just start the write event for a byte, then spin on `EVENTS_ENDEPOUT[X]` until the byte is sent?

hudson-ayers, 2020-10-26:
I am going to take a stab at this. If synchronous USB CDC output looks too difficult, I am going to try using the log functionality to log panic dump info to flash, and have the board print any existing panic dump at boot, such that any panics which occur after initialization can at least be debugged by just resetting the board.

commit c7083df9 (2020-10-28): Merge #2178

2178: Synchronous panic output over USB for nano33ble r=bradjc a=hudson-ayers

### Pull Request Overview

This pull request fixes #1972 by implementing a "synchronous" panic output via USB CDC for the nano33ble. It does so by spin-waiting for USB interrupts to allow the USB and CDC state machines to operate as intended despite the fact that the core kernel is no longer running and dispatching interrupts. It also requires some creative use of `static mut` global variables to give the panic handler access to the resources it needs.

Of course, this approach will not provide panic ou [... cut]
commit 9ec147aa (2020-10-28): Merge #2178

2178: Synchronous panic output over USB for nano33ble r=bradjc a=hudson-ayers

### Pull Request Overview

This pull request fixes #1972 by implementing a "synchronous" panic output via USB CDC for the nano33ble. It does so by spin-waiting for USB interrupts to allow the USB and CDC state machines to operate as intended despite the fact that the core kernel is no longer running and dispatching interrupts. It also requires some creative use of `static mut` global variables to give the panic handler access to the resources it needs.

Of course, this approach will not provide panic ou [... cut]
PR #2178 (2020-10-28): Synchronous panic output over USB for nano33ble
### Pull Request Overview

This pull request fixes #1972 by implementing a "synchronous" panic output via USB CDC for the nano33ble. It does so by spin-waiting for USB interrupts to allow the USB and CDC state machines to operate as intended despite the fact that the core kernel is no longer running and dispatching interrupts. It also requires some creative use of `static mut` global variables to give the panic handler access to the resources it needs.

Of course, this approach will not provide panic output if the USB connection is not yet establ [... cut]
closed 2020-10-28 by bors[bot]