tock/tock #1977: Tracking: Full Support for Nano33BLE

bradjc, 2020-06-24:
With Tock running on this board with virtual_uart sitting on top of USB, the Nano 33 BLE is a reasonably promising board to be a standard, well supported Tock board. Additionally, if the board turns out to not be great, the infrastructure which would make the Nano 33 BLE work well would apply to any other board that has the nRF52840 + USB combination.

- [ ] Make the bootloader and tockloader experience "just work". This means `tockloader install` should work on a compiled app.
  - [ ] Replace the bootloader on the board.
    - [ ] Trojan horse kernel that replaces the bootloader.
    - [ ] tock-bootloader ported on top of the USB-CDC-ACM stack.
      - Need some way for the bootloader to detect it should _stay_ in bootloader mode. I'm not sure how Arduino does this.
  - [ ] Need USB serial name that tockloader understands so that it can auto-detect this is a nRF52840 + USB board.
  - [ ] Need the CDC driver to detect when tockloader is trying to connect, and reset the board into the bootloader mode. This should be doable with the baud rate trick that Arduino does.
    - [ ] CDC capsule parses CDC messages looking for a baud rate setting of 1200.
    - [ ] Need a reset() function passed into the CDC capsule. Perhaps with a capability.
- [ ] Make panic debugging work over USB #1972 
- [ ] Implement sensor drivers.
- [ ] Implement BLE.
- [ ] Implement 15.4.

bradjc, 2020-12-02:
One thought on this topic: proper Tock support really requires breaking backwards compatibility with other nrf52 SDKs and workflows (particularly anything based on the softdevice). That is, by asking users to flash our bootloader, their board will no longer work with any adafruit or arduino instructions. Now we can write a good readme that explains what is happening, and they certainly can run different code on the board of course, but it might not run out of the box anymore.

Do you think this is a concern? Is it worth it to have a mechanism for switching the bootloader back?

alexandruradovici, 2020-12-02:
I am speaking now from a teaching / beginner point of view and this is only my personal opinion: any action that might "damage" (please see the quotes) the board might be something that might scare people. If the "student" breaks the board and is not able to get it back to its factory state is not something that I as a teacher want to use.

I have been searching for a while for a device that can be used with some scripting language and with Tock. A good candidate has been nano ble 33, as it run CircuitPython [1] until I got to this [2]. I have to replace the boot loader by soldering pins.

People usually want something that can work out of the box and is not easily brick-able.

[1] https://circuitpython.org/board/arduino_nano_33_ble/
[2] https://forums.adafruit.com/viewtopic.php?f=60&t=158279

bradjc, 2020-12-17:
The tock bootloader is now working on the nrf52 over USB/CDC. Next step is to start testing how to get the bootloader on to the board. Currently I _think_ there is a sane way to both add the Tock bootloader and support getting the board back to its original state.