tock/tock #1993: IPC: Return error if shared memory region cannot be protected due to MPU alignment issues

bradjc, 2020-07-01:
Right now the IPC `allow()` call can succeed, and a notify callback can be triggered, even if the `expose_to()` call fails.

bradjc, 2021-04-26:
If we re-implement IPC to use message passing rather than shared memory, we can revisit the MPU implementation (re: #1532).

alexandruradovici, 2021-08-08:
I was playing with the IPC API and noticed that applications fault due to memory protection if the shared buffers are not multiple of 64. The buffer's memory region does not seem to be mapped.

I cannot figure out the reason for this, all the API calls seem to return success.

If this is a requirement, maybe we should add this to the kernel and refuse the buffer?

lschuermann, 2021-12-13:
While debugging tock/tock#2906 I've noticed a more intricate issue of app load order and, what appears to me as a racy relationship of the IPC service registering its callback and the client notifying the service. This is in the context of a failing `rot13_{service,client}` CI test. When debugging these issues I used to change the app's order in the TBF list and inserted `printf` statements. With the `rot13_service` and `rot13_client` installed in this order, what seems to be happening in the normal (good) case is:
1. `rot13_client` performs initialization and all of its IPC calls including `ipc_notify_service` in one go, which will enqueue a task in the target process' queue
2. `rot13_service` performs initialization and the call to `ipc_register_service_callback`, which will register its callback
3. the kernel processes the task enqueued in (1) which calls back to `IPC::schedule_upcall`, which will ultimately schedule the proper upcall on the `rot13_service` process
4. the IPC service callback scheduled in (3) is called by the kernel scheduler

Now, if the kernel were to execute (3) before (2), the IPC task is converted into a callback (`FunctionCall`) _before_ the service had a chance to subscribe. When service subscribes, this pending upcall is removed from its queue. It appears one can trigger this behavior through a combination of load-order and scheduler behavior, but it is most obvious when inserting a simple `printf` statement at the beginning of `ipc_service`'s main.

We currently have no way to tell whether notifying an IPC service (i.e. scheduling a callback in the service process) actually worked. Whereas the source code used to indicate that we might have a method of identifying a null-callback (`EOFF`), `Process::enqueue_task` does not actually provide us any information about whether the enqueued IPC task will result in a callback being scheduled or not. This usually makes sense and is a logical consequence of our architecture based around the fact that a null-callback should be treated exactly the same as a proper "do nothing" callback, but here it's not good.

LawrenceEsswood, 2023-01-26:
To add something to point (6) of the current IPC issues: restarting an application resets app_break/kernel_memory_break//allow_high_water_mark . The result is that an application (A) can:

1) Make a very large malloc(), causing a brk system call to claim all remaining grant/heap memory for itself.
2) Allow this to another process (call it B)
3) Restart itself in place (i.e., by faulting) (and so reset the watermarks)
4) Make syscalls to capsules causing grant allocations that exceed the size of allocation its previous life

This will result in application B having direct access to kernel data structures. Spicy. This can be leveraged even from safe userspace Rust.

pqcfox, 2025-05-02:
IIRC from the Tock Strategy Workshop there was a good bit of interest in specifically a new "message passing"-type IPC architecture (and @LawrenceEsswood I believe you might have mentioned having a draft of something similar you developed that you've found works well?).

If I wanted to help kickstart getting a sketch of the improved IPC functionality together from the requirements discussed in the Strategy Workshop, is that something that could actively use hands?

lschuermann, 2025-05-06:
@pqcfox That sounds super interesting. I think it would be great to start from an existing example, either just for inspiration or as a basis for development. I'd suggest that, based on the feedback we got from the workshop, starting this effort in a more smaller, focused, and high-bandwidth group might be best. Pinging @alevy who I think is also interested in that.

lschuermann, 2025-05-06:
I created a GitHub Project to track existing IPC issues, potential IPC use cases, and efforts around a revised IPC interface: https://github.com/orgs/tock/projects/5

bradjc, 2026-08-19:
I believe our current IPC is more problematic for memory safety than maybe we thought. For example:

1. Process A and Process B collude to break the kernel.
2. Process A calls sbrk to get as much memory as it can, while using as few grants as possible.
3. Process A defines an IPC buffer at the end of its memory region.
1. Process A shares its IPC buffer with Process B.
2. Process A calls restart and is restarted.
3. Process B still has an MPU region for the memory in Process A.
4. Process A now allocates many grants.
5. The region Process B has access to is now kernel owned memory and can manipulate the grant structure for Process A.