tock/tock #1995: RFC: Have `Process` include a reference to a `Debugger` object

bradjc, 2020-07-01:
## Background

To help with debugging processes, the `Process` struct currently includes a `Debug` struct that holds different values to help with debugging a process if it crashes. We intentionally sectioned off the `Debug` items with the thought that eventually we wouldn't want to make including debug state mandatory for a userspace process.

## What's New

There is interest in changing how process debugging works on two fronts:
1. Removing it to reduce overhead (see #1830).
2. Increasing it to make debugging even easier (see #1698). Also, I would like to add a "logging" feature that records when certain things fail (for example an sbrk call returns an error).

## Proposal

At a high-level, I'm imagining a `ProcessDebugger` trait, with different implementations that could (optionally) be passed to `Process::create()` and stored as a reference in the `Process` struct.

```rust
struct Process {
...
    debugger: Option<&ProcessDebugger>
...
}
```

The trait would specify all of the hooks that `Process` calls if the debugger reference is present.

```rust
trait ProcessDebugger {
    fn syscall_called(&self, syscall);
...
}
```

The ProcessDebugger implementation could then decide what to store, and how to display it on a panic.

I don't know exactly how the `Process` trait would change.

I also don't know if this is actually possible. But I'm looking for something that gives us the flexibility to permit the maximum amount of debugging possible, while also allowing us to use less or none at all if desired.

hudson-ayers, 2020-07-01:
I think that this approach sounds good if we can use generics rather than trait objects to store the `ProcessDebugger` in the `Process`. Otherwise the compiler will not be able to optimize away the function calls in the case of the minimal `ProcessDebugger` being used so this will add a baseline overhead that the proposal in #1830 will not.

However I am concerned that `ProcessType` already being stored behind a trait object may make this difficult

bradjc, 2021-04-26:
Not much progress here, this might get revisited from a code size perspective.

closed 2021-04-26 by bradjc