tock/tock #2007: RFC: Mark core component "changability" in source header comments

bradjc, 2020-07-08:
Tock has a general procedure for incorporating new code to core components (e.g. kernel/ libraries/ arch/) that _roughly_ flows through the following stages:

1. Use case is described in an issue, on the mailing list, during a call, or on slack, and generally there is agreed upon merit or interest, and any immediate issues are noted.
2. A PR is opened and merged which includes a reasonable implementation and at least an initial use case.
3. The implementation is subsequently refined as bugs or other issues come up. Sometimes new platforms require some changes.
4. Overtime, more and more rough edges are found, and that culminates in a substantial re-write which includes all of the learning from using it over time. Many different cases and hardware platforms are considered. This triggers often a lengthy review process, generally because now the code is more familiar and more widely used.
5. We try to avoid changing the code since the re-write likely considered many subtle factors which are not always obvious. However, we do loop back to (3) if new hardware platforms emerge that challenge some of the assumptions used in previous re-writes.

Code that has hit stage 4 is not usually that obvious, and is really only known by the developers engaged in the discussions when the stage 4 PR was under review. It seems like we should better capture when core components have reached that stage, so it is less surprising to future developers when we are hesitant to accept PRs that change those components.

## Proposal

We add a `## Changability` section (probably with a better name) in the top header comments of files that have reached stage 4 indicating them as such. We then document this process somewhere in docs and refer to it.

I don't think we want to document every file, and a file without this section in its comments can just be assumed to not be in stage 4.

## Precedent

We have marked some `Driver` implementations in capsules with their stability.

alistair23, 2020-07-08:
It seems like a bad idea to have a "this code is old so don't re-write it" mentality.

I agree that is kind of what happens, but codifying it seems like a step backwards.

alistair23, 2020-07-08:
A better idea would be to have "experimental" files. These don't require as much scrutiny to change as they are considered experimental.

Maybe it's worth having subsystems and then marking some as experimental?

bradjc, 2020-07-08:
"Experimental" is another great case. Both networking and USB are in this category, and should be marked as such.

alistair23, 2020-07-09:
If there are subsystems and they can be marked experimental then we can mark the USB subsystem as experimental then. That gives users an idea that it isn't well tested and tells developers that it's in flux.

bradjc, 2020-07-15:
I was waiting to see if there are other comments, but the intent wouldn't be "this code is old...", but rather "this code was carefully reviewed and factored in many concerns...", and instead of "...don't rewrite it", "...factor those concerns in as well when you re-write it". These implicit requirements keep coming up and confusing contributors, and I think it would make sense to write them down.

ppannuto, 2020-07-21:
There's some precedent for this elsewhere, more in packaging:

 - [Node has](https://nodejs.org/api/documentation.html) Deprecated, Experimental, and Stable
 - [Python has](https://pypi.org/classifiers/) Planning, Pre-Alpha, Alpha, Beta, Stable, Mature, Inactive
 - Rust _sort of_ captures this with nightly/beta/stable, though that's different

I might suggest that we have categories of:

 - Experimental -- New stuff
 - Stable -- Been around for a while, no longer reasonable to call experimental, but not necessarily the next category
 - Mature -- This is code that has many hidden, subtle concerns that have been factored in over time
 - Deprecated -- Legacy stuff

bradjc, 2020-07-27:
I agree, except I think "Stable" should be named "Normal", and is the default (all unlabeled code would be assigned "Normal"). For example, if we had to label `process.rs` or `sched.rs`, what would we label it?

ppannuto, 2020-07-27:
Yeah, on second thought I agree that `Stable` is perhaps too strong of a word for what we're looking to convey. "Normal" is a bit weird, but I also don't have anything better, so it sounds good to me :).

bradjc, 2021-04-23:
We can probably close this RFC. I think this idea has merit, but probably not enough momentum to try to add to the code base and maintain going forward.

closed 2021-04-23 by bradjc