tock/tock #2023: Proposal: Runtime CI Testing

alistair23, 2020-07-14:
Recently I have been making changes to the Tock CI to enable run time testing. The idea is that this will slowly progress to cover more and more, including [running user space apps](https://github.com/tock/tock/pull/2012) and testing the kernel interface (as well as other things).

This will lead to CI failures for things that aren't regressions. For example a change in the userspace/kernel ABI will cause a app test to break, but it's not considered a regression. Another pain point is the failure caused by failing to download the OpenTitan ROM.

My worry is that because the CI will sometime have false negatives, it will be disabled, removed or ignored and then not be usable.

This proposal is to split the test CI into two parts:
 1. Mostly Tock CI
    - This will run CI tests where Tock is the almost the only part. For example running Tock on a pinned version of QEMU for the HiFive1. Just testing that Tock prints something.
 2. Other components CI
    - This runs more complex CI tests, but involves other components.
    - This might download the OpenTitan ROM and run Tock with that
    - This might build/download libtock-rs and run apps on top of Tock

The idea with the spit is that the tests in the first category must always pass. The tests in the second category are not blocking. There should still be a requirement (manually enforced by reviewers) that any failures in 2 must be justified. For example "The libtock-rs tests are failing because this changes the kernel interface. Once this PR is merged I'll update libtock-rs to work then fix the tets".

My worry though is that the category 2 tests will break (for a good reason or not) and then bit rot as no one wants to fix them. Unfortunately this type of non-blocking split might be the only way to maintain the OT boot test (the ROM download is a pain) and make progress on running userspace apps.

Thoughts?

bradjc, 2020-07-15:
My 2¢

- Tock CI should be versioned, and Tock should have PRs that update the CI to a new version when new CI versions are released. This is what we do with elf2tab and tockloader. I would expect this would lessen issues around CI fatigue.
- There should be a tool that we can call in a PR (like bors and the tool rust uses to check a change against all crates.io crates) to do more advanced checks for PRs that warrant it. (Updating the porting.md file does not need to run in QEMU.)

This doesn't address all of your points or questions by any means.

ppannuto, 2020-07-21:
> There should be a tool that we can call in a PR (like bors and the tool rust uses to check a change against all crates.io crates) to do more advanced checks for PRs that warrant it. (Updating the porting.md file does not need to run in QEMU.)

To this point, we can condition github actions on labels for PRs. i.e. if a PR is labeled `documentation`, it can skip the heavyweight tests. We could be more pedantic, adding `CI-full` or `CI-light` labels to each PR. 

---

> CI should be versioned..

I'm not sure I understand mechanistically how this would be implemented? CI is sort of versioned already, in that its operation is stored in the repository itself. Are you envisioning an external CI repo (submodule!? :D) that's pinned or some such?

bradjc, 2020-07-21:
> Are you envisioning an external CI repo (submodule!? :D) that's pinned or some such?

Yes. For example, consider our clippy test. We don't include clippy in the repo, and we use a very specific version. All we have is a script that knows how to run it, and the relevant github action config file. I imagine the same pattern would work for a tool we do develop ourselves: all that is in tock/tock is how to download it and run it, plus a github action to run that for us. The mirror would exist within the tock-ci tool, aka a quick script that can download the latest tock kernel and run the tests.

bradjc, 2021-04-26:
I think we absolutely want CI to test on actual hardware. We also should leverage QEMU to the extent its useful. We are using QEMU with risc-v.

I think this particular RFC is uncontroversial and therefore hasn't generated much discussion.

At this point, someone needs to put together a tracking issue with a plan for implementing the improved support.

closed 2021-04-26 by bradjc