tock/tock #2030: Release 1.6

bradjc, 2020-07-17:
## Release Date: Summer 2020

Milestones to be included in the release:

- [ ] Scheduler options #1441 
- [ ] Alarm HIL updates
- [ ] Fixed address support

## Testing

To test the release we use the following applications and run them on all applicable boards and ensure they execute successfully. This list should be copied for each platform and will be tested once a release candidate is ready.

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`
  - [ ] `examples/tests/hail` and `examples/sensors`

- Imix
  - [ ] `examples/tests/imix`
  - [ ] `examples/ble-uart`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  - [ ] `boards/imix/src/test/aes_ccm_test.rs`
  - [ ] `boards/imix/src/test/aes_test.rs`
  - [ ] `boards/imix/src/test/i2c_dummy.rs`
  - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
  - [ ] `boards/imix/src/test/ipv6_lowpan_test.rs`
  - [ ] `boards/imix/src/test/linear_log_test.rs`
  - [ ] `boards/imix/src/test/log_test.rs`
  - [ ] `boards/imix/src/test/rng_test.rs`
  - [ ] `boards/imix/src/test/spi_dummy.rs`
  - [ ] `boards/imix/src/test/spi_slave_dummy.rs`
  - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
      - [ ] `start()`
      - [ ] `start_rx()`
      - [ ] `start_with_app()`
      - [ ] `start_dual_rx()`
  - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`

- Miscellaneous
  - [ ] Test the apps compiled at 1.0
  - [ ] process console works as expected
  - [ ] restarting apps works as expected

---

## Sign off

- [ ] Niklas Adolfsson, @niklasad1
- [ ] Hudson Ayers, @hudson-ayers
- [ ] Brad Campbell, @bradjc
- [ ] Branden Ghena, @brghena
- [ ] Philip Levis, @phil-levis
- [ ] Amit Levy, @alevy
- [ ] Pat Pannuto, @ppannuto (2020-04-29, 17:30 UTC)
- [ ] Johnathan Van Why, @jrvanwhy

bradjc, 2020-10-09:
### Hail

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`
  - [ ] `examples/tests/hail` and `examples/sensors`

lschuermann, 2020-10-09:
Linking #2143 here, which for me causes a kernel-loop if used with 64-bit wide alarms in userspace through the alarm driver. **If** that issue can be confirmed, it should probably be a blocker for 1.6 since OpenTitan / EarlGrey seems to use a 64-bit wide alarm. [Core call discussion](https://github.com/tock/tock/blob/2434c33f4e0df2a723924d9e093e3bb0f706a04e/doc/wg/core/notes/core-notes-2020-10-09.md#16-release)

phil-levis, 2020-10-12:
@lschuermann Thanks -- I will take a look at #2143 tomorrow.

alistair23, 2020-10-12:
- OpenTitan
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

- Apollo3
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

hudson-ayers, 2020-10-12:
# Imix

- All Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

 

- Imix-specific
  - [ ] `examples/tests/imix`
  - [ ] `examples/ble-uart`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  -  [ ] **In-Kernel Tests**
      - [ ] `boards/imix/src/test/aes_ccm_test.rs`
      - [ ] `boards/imix/src/test/aes_test.rs`
      - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
      - [ ]  `boards/imix/src/test/ipv6_lowpan_test.rs`
      - [ ] `boards/imix/src/test/linear_log_test.rs`
      - [ ] `boards/imix/src/test/log_test.rs`
      - [ ] `boards/imix/src/test/rng_test.rs`
      - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
          - [ ] `start()`
          - [ ] `start_rx()`
          - [ ] `start_with_app()`
          - [ ] `start_dual_rx()`
      - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`

jrvanwhy, 2020-10-13:
Normally, I would check H1 by updating tock-on-titan to the new release candidate. Unfortunately, because we are stuck with an old `libtock-rs` (which requires an older Rust toolchain), I am unable to do that check this time.

hudson-ayers, 2020-10-13:
@lebakassemmerl are you able to complete release testing for the MSP432? You can ignore tests that use peripherals not yet implemented for that board/chip/.

@alexandruradovici @valexandru @cosmindanielradu19 @rajivr anyone who is able to complete release testing for one of the stm32 boards or one of the nucleo boards would be greatly appreciated, as I do not think any of the core team currently has access to stm hardware.

alexandruradovici, 2020-10-14:
Sure, as we never did this before, can you direct us to some documentation on how to run those tests?

hudson-ayers, 2020-10-14:
Sure thing! I am not sure we have documentation on this, but here is a description:

First, you want to check out current Tock master and flash the kernel onto your board.

Next, checkout master in https://github.com/tock/libtock-c , and flash each of the apps listed in the "all boards" section at the top of this issue. After flashing the app, look at the console output and verify that the app is executing correctly. For lines where multiple apps are listed, flash multiple apps simultaneously to verify that those combinations of apps work. Most apps should have a `README.md` file describing what a success case looks like.

Some boards may not implement peripherals that certain tests require (e.g. the stm32f3 board does not have an RNG peripheral driver yet), so those tests will fail. This is fine, just note that in a comment under the test along with the note that the test failed.

Once you have started testing a board, put a comment in this thread with the name of the board and the list of tests you are going to do. Then as you do the tests, check off each test that succeeds / place a note describing the failure for each test that fails. Make sure to place a comment before you start testing to prevent multiple people running tests for the same platform.

Let me know if you have any other questions!

hudson-ayers, 2020-10-14:
# nrf52dk

* All Boards
  * [ ]  `examples/sensors`
  * [x]  `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    * Fails with error code 16385 as expected (no serialization)
  * [ ]  `examples/c_hello` and `examples/tests/printf_long`
  * [ ]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    Doesn't work - as expected.
  * [ ]  `examples/blink`
  * [ ]  `examples/rot_client` and `examples/rot_service`
    * [x] With both: `12: Uryyb Jbeyq!  12: Hello World!`
  * [ ]  `examples/blink` and `examples/c_hello` and `examples/buttons`
  * [ ]  `examples/lua-hello`
  * [ ]  `examples/tests/console_timeout`
  * [ ]  `examples/tests/malloc_test01`
  * [ ]  `examples/tests/stack_size_test01`
  * [ ]  `examples/tests/stack_size_test02`
  * [ ]  `examples/tests/mpu_stack_growth`
  * [ ]  `examples/tests/mpu_walk_region`
     regular loop does not crash, button presses crash both reads as expected
  * [ ]  `examples/tests/multi_alarm_test`
     slow, peaceful blinking of each LED in order
  * [ ]  `examples/tests/adc`
     `No ADC driver!`
  * [ ]  `examples/tests/adc_continuous`
     `No ADC driver!`
  * [ ]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
     `Number of LEDs: 4` .. then each LED randomly turns on and off
  * [ ]  `examples/tests/gpio` with mode set to 0
    Touch `P0.03` to Vdd (randomly tested with a few other pins to _not_ fire an interrupt):
 
* nRF specific
  * [ ]  `examples/ble_advertising`
  * [ ]  `examples/ble_passive_scanning`

alexandruradovici, 2020-10-15:
# STM32F412G Discovery

  * All Boards    
      * [ ]  `examples/sensors`
      * [ ]  `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
      * [ ]  `examples/c_hello` and `examples/tests/printf_long`
        * uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
      * [ ]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
        * We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
      * [ ]  `examples/blink`
        * blink: blinks LEDs
      * [ ]  `examples/rot_client` and `examples/rot_service`
        * rot_ipc: tests IPC with a simple service
      * [ ]  `examples/blink` and `examples/c_hello` and `examples/buttons`
      * [ ]  `examples/lua-hello`
      * [ ]  `examples/tests/console_timeout`
      * [ ]  `examples/tests/malloc_test01`
      * [ ]  `examples/tests/stack_size_test01`
      * [ ]  `examples/tests/stack_size_test02`
      * [ ]  `examples/tests/mpu_stack_growth`
      * [ ]  `examples/tests/mpu_walk_region`
      * [ ]  `examples/tests/multi_alarm_test`
      * [ ]  `examples/tests/adc`
      * [ ]  `examples/tests/adc_continuous`
      * [ ]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
      * [ ]  `examples/tests/gpio` with mode set to 0
      * [ ]  `examples/tests/whileone`

alexandruradovici, 2020-10-15:
# STM Nucleo  F429ZI

  * All Boards    

      * [ ] ❌ `examples/sensors`
         * FAILS: temperature sensor driver is not exposed, submitted PR (#2158)
      * [ ]  `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
         * fails as expected as there is no BLE
      * [ ] `examples/c_hello` and `examples/tests/printf_long`
           * Work when flashed separately and when flashed like | printf_long  c_hello |.  If flashed the other way around, `printf_long` app fails to load with `App flash length not supported by MPU`. Submitted PR to fix the flash size (#2159)
      * [ ]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
        * long failed with -2, probaly as expected
      * [ ]  `examples/blink`
      * [ ]  `examples/rot_client` and `examples/rot_service`
      * [ ]  `examples/blink` and `examples/c_hello` and `examples/buttons`
      * [ ]  `examples/lua-hello`
      * [ ]  `examples/tests/console_timeout`
      * [ ]  `examples/tests/malloc_test01`
      * [ ]  `examples/tests/stack_size_test01`
      * [ ]  `examples/tests/stack_size_test02`
      * [ ]  `examples/tests/mpu_stack_growth`
      * [ ]  `examples/tests/mpu_walk_region`
      * [ ]  `examples/tests/multi_alarm_test`
      * [ ]  `examples/tests/adc`
        * fails as expected at timed samples as this is not implemented by this board
      * [ ]  `examples/tests/adc_continuous`
        * fails as expected at timed samples as this is not implemented by this board
      * [ ]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
        * does not actuall work as RNG is not available, faults due to this 
      * [ ]  `examples/tests/gpio` with mode set to 0
        * works, EXTI line has to be associated in `main.rs`
      * [ ]  `examples/tests/whileone`

I am receiving a lot of MPU allocation failures, depending on the apps order.

lebakassemmerl, 2020-10-16:
## MSPEXP432P401R

    * [ ]  `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    * [ ]  `examples/c_hello` and `examples/tests/printf_long`
    * [ ]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    * [ ]  `examples/blink`
    * [ ]  `examples/rot_client` and `examples/rot_service`
    * [ ]  `examples/blink` and `examples/c_hello` and `examples/buttons`
    * [ ]  `examples/lua-hello`
    * [ ]  `examples/tests/console_timeout`
    * [ ]  `examples/tests/malloc_test01`
    * [ ]  `examples/tests/stack_size_test01`
    * [ ]  `examples/tests/stack_size_test02`
    * [ ]  `examples/tests/mpu_stack_growth`
    * [ ]  `examples/tests/mpu_walk_region`
    * [ ]  `examples/tests/multi_alarm_test`
    * [ ]  `examples/tests/adc`
    * [ ]  `examples/tests/adc_continuous`
    * [ ]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
    * [ ]  `examples/tests/gpio` with mode set to 0
    * [ ]  `examples/tests/whileone`

krady21, 2020-10-16:
# STM32F3 Discovery

[] examples/sensors
[] examples/services/ble-env-sense and examples/services/ble-env-sense/test-with-sensors
[] examples/c_hello and examples/tests/printf_long
[] examples/tests/console_recv_short and examples/tests/console_recv_long
[] examples/blink
[] examples/rot_client and examples/rot_service
[] examples/blink and examples/c_hello and examples/buttons
[] examples/lua-hello
[] examples/tests/console_timeout
[] examples/tests/malloc_test01
[] examples/tests/stack_size_test01
[] examples/tests/stack_size_test02
[] examples/tests/mpu_stack_growth
[] examples/tests/mpu_walk_region
[] examples/tests/multi_alarm_test
[] examples/tests/adc
[] examples/tests/adc_continuous
[] examples/tutorials/05_ipc/led and examples/tutorials/05_ipc/rng and 
[] examples/tutorials/05_ipc/logic
[] examples/tests/gpio with mode set to 0
[] examples/tests/whileone

hudson-ayers, 2020-10-21:
Dug into the potential scheduling bug I mentioned on the core call, it was actually an application logic bug exposed by the switch to a round robin scheduler. Fixed here: https://github.com/tock/libtock-c/pull/116. 

I think the only remaining blocker is #2146

hudson-ayers, 2020-10-24:
@lebakassemmerl @alexandruradovici @krady21 thanks so much with your help testing!

The 1.6 release has been tagged here: https://github.com/tock/tock/releases/tag/release-1.6

closed 2020-10-24 by hudson-ayers