tock/tock #2057: Context Switch and Interrupt Checking Race Condition

bradjc, 2020-07-30:
After an entertaining journey through the BLE serialization library and new scheduler code, @hudson-ayers and I believe we have tracked down why running the imix/hail app with the `whileone` app no longer works.

Short answer: it is possible for an interrupt to fire _after_ the scheduler has _committed_ to running the whileone app, but before the context switch actually occurs. The interrupt is then not serviced until _after_ the whileone app exhausts its timeslice.

Ok, but why does this break the BLE serialization library? Well, the interrupt in question is related to the USART TX, which eventually triggers a "tx done" callback to userspace. However, since the interrupt handling is so delayed, the nRF51822 sends back a response, triggering a DMA RX done interrupt. Both are pending when whileone is preempted, but the RX DMA has a lower NVIC index (yeah, that problem again!) so that is handled first triggering a "rx done" callback in userspace which throws off the BLE serialization state machine (since it is expected a TX done callback first).

Fix
---

We need to disable interrupts between our last check for any pending interrupts and the actual context switch.

hudson-ayers, 2020-08-11:
Should this be a blocker for 1.6, given that it makes the Imix and Hail apps fail? Or should it not be considered a blocker because really those apps should not rely on specific interrupt ordering?

pqcfox, 2024-06-20:
Is this issue still present in the current version of Tock? If so I'd love to implement the mentioned fix.

lschuermann, 2024-07-03:
> Is this issue still present in the current version of Tock? If so I'd love to implement the mentioned fix.

From the issue description above I'm having a hard time telling whether this is actually a _kernel_-issue:

> Both are pending when whileone is preempted, but the RX DMA has a lower NVIC index (yeah, that problem again!) so that is handled first triggering a "rx done" callback in userspace which throws off the BLE serialization state machine (since it is expected a TX done callback first).

I do not know whether this issue still exists or not, but my feeling is that this is a mix between an incorrect capsule and missing interrupt priority support (#3576). @bradjc should correct me if I'm wrong, but I don't think that this points to a fundamental issue in our core-kernel loop, the cortex-m arch crates, or even the USART driver?

bradjc, 2024-07-03:
All of this does make sense why this has been here for four years. I think both are issues and both are still issues.

I do think it is a problem that a scheduler which wants to prioritize interrupts over userspace code can't if this case happens. The scheduler shouldn't have to let an app run for its full timeslice before it can handle an interrupt. That seems like a bug.

I think the reason this is still open is that it isn't entirely clear how to disable and re-enable interrupts correctly. @pqcfox if you want to give it a go I think this would be an improvement. However, it isn't a _huge_ improvement, so if the code is really terrible then it might not be worth it.

lschuermann, 2024-07-03:
FWIW, this is most certainly an ARM-specific assembly. Our current RISC-V context switch & trap handling assembly will happily go straight back to the `_start_trap` symbol on a pending interrupt the instant we run an `mret` instruction, if an interrupt arrives between the check for pending bottom-half ISRs in the kernel loop and the context switch to an app.

ppannuto, 2024-07-03:
If it's just wrapping a truly small critical section, the quintessential arm answer is `cpsid i` .. critical code .. `cpsie i` — it's a big hammer, disables all interrupts, but code footprint is very light

[Could generalize as a `enter/exit_critical_section()`, though it sounds like risc-v doesn't need this here?]