tock/tock #2058: Flashing kernel using adafruit-nrfutil

dzervas, 2020-07-30:
Hello,

I have a particle xenon (nrf52840) with Adafruit's bootloader which allows me to flash it using DFU or mass storage (UF2). I don't want to hook up a J-Link.

Is the tock kernel actually the bootloader? shoud I overwrite it with `adafruit-nrfutil --bootloader <binary>`?

bradjc, 2020-07-30:
You probably do not want to overwrite the bootloader. Instead, place the kernel in flash wherever the adafruit bootloader expects application code to go. You just have to make sure that the address in the `layout.ld` file for the `boards/` crate you are using matches that address.

dzervas, 2020-07-30:
I'm not quite sure that I get it right but I did the following:
updated `nrf52840_chip_layout.ld`:

```ld
/* Memory Space Definitions, 1M flash, 256K ram */
MEMORY
{
  /* rom (rx)  : ORIGIN = 0x00000000, LENGTH = 192K */
  rom (rx)  : ORIGIN = 0x00026000, LENGTH = 16K
  prog (rx) : ORIGIN = 0x00030000, LENGTH = 832K
  ram (rwx) : ORIGIN = 0x20000000, LENGTH = 256K
}

MPU_MIN_ALIGN = 8K;
```

as the bootloader states that the application address is at `0x26000` [here](https://github.com/adafruit/Adafruit_nRF52_Bootloader#making-your-own-uf2)

bradjc, 2020-07-30:
That looks right so far. `prog` is for applications. You will need to move the prog region to a later address in flash (maybe to 0x0040000). Of course, if you flash an application you need to make sure it is at that address.

dzervas, 2020-07-30:
Ha! found the official linker script! https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/cores/nRF5/linker/nrf52840_s140_v6.ld - it's used by the arduino core for that board, the supported way to do it is with the adafruit bootloader.

My new ld script (compiles fine):
```ld
/* Memory Space Definitions, 1M flash, 256K ram */
MEMORY
{
  rom (rx)  : ORIGIN = 0x00026000, LENGTH = 0x58000 - 0x26000
  prog (rx) : ORIGIN = 0x00058000, LENGTH = 0xED000 - 0x58000
  ram (rwx) : ORIGIN = 0x20006000, LENGTH = 0x20040000 - 0x20006000
}

MPU_MIN_ALIGN = 8K;
```

Now the problem seems to be the the flashing program, I get:
```
[+] Converting ELF file to hex
[+] Generating package
Zip created at /home/dzervas/Tools/tock/target/thumbv7em-none-eabi/release/nrf52840dk.hex.zip
[+] Flashing package over UART
Upgrading target on /dev/ttyACM0 with DFU package /home/dzervas/Tools/tock/target/thumbv7em-none-eabi/release/nrf52840dk.hex.zip. Flow control is disabled, Single bank, Touch 1200
Touched serial port /dev/ttyACM0
Opened serial port /dev/ttyACM0
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 204804
Sending DFU start packet
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: list index out of range
Traceback (most recent call last):
  File "/home/dzervas/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 294, in serial
    dfu.dfu_send_images()
  File "/home/dzervas/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 235, in dfu_send_images
    self._dfu_send_image(HexType.APPLICATION, self.manifest.application)
  File "/home/dzervas/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 199, in _dfu_send_image
    self.dfu_transport.send_start_dfu(program_mode, softdevice_size, bootloader_size,
  File "/home/dzervas/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 179, in send_start_dfu
    self.send_packet(packet)
  File "/home/dzervas/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 243, in send_packet
    ack = self.get_ack_nr()
  File "/home/dzervas/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 291, in get_ack_nr
    return (data[0] >> 3) & 0x07
IndexError: list index out of range

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
```

I'll take a closer look and report back. Thank you! (closing for now)

dzervas, 2020-07-30:
Nope, I was using wrong port, flashed successfully. Problem is that I see no USB device enumerating - even `dmesg` does not report any activity. Do I have to configure tock kernel to raise a serial interface on boot?

bradjc, 2020-07-30:
Short answer: yes. It sounds like this board has the USB directly connected to the nRF52? You should look at the nano33ble board. That is also based on the nRF52840 and is probably a better starting point. It sets up serial over the Tock USB stack.

dzervas, 2020-07-30:
Yeap, just figured that out. It does not have a Serial to USB, so the USB initialization is missing from the reset handler :)

AMAZING support, wow, keep it up

closed 2020-07-30 by dzervas