tock/tock #2093: Can tock be trated to become General Purpose (GP - OS )

AloyASen, 2020-09-02:
I have been following this project since and have received its development and ambitious goals but I am quite inquisitive about whether it would be possible to make a GUI for the same considering the adoption of TOCK. Just a random question

alistair23, 2020-09-02:
Can you be more descriptive about what you mean by a general purpose OS?

Tock can support displays, there is currently work going on to add screen support, see https://github.com/tock/tock/pull/2079 for example.

lschuermann, 2020-10-12:
As far as I know, and based on the current uses, Tock is not designed to be a general purpose operating system such as Linux or Windows. While probably possible to make it run on conventional desktop / server hardware given sufficient time and resources, it won't replace conventional operating systems any time soon. Most projects using Tock target embedded systems such as IoT devices or security chips. While these systems may use displays as indicated by @alistair23, they may not have the notion of windows, buttons and other elements commonly associated with a GUI.

Does this answer your question?

alistair23, 2020-10-23:
Closing due to inactivity

closed 2020-10-23 by alistair23