tock/tock #2096: USB: Need functions to reset data toggle in UsbController trait

namyoonw, 2020-09-03:
USB data toggle should be reset at the device side by host request in these cases:
- Standard Device Request, SetConfiguration => the data toggle should be zero for all endpoints.
- Standard Endpoint Request, ClearFeature => the data toggle should be zero for the specified endpoint (both in and out).

Without this, possibly it may happen that the device never gets a packet from the host because it is dropped for data toggle mismatch reason, and the host hangs.

The proposed change is
- UsbController trait defines these functions:
   -  fn set_config(&self, _conf_val: usize);
      // usb device driver resets data toggle for all endpoint.
   -  fn endpoint_reset(&self, endpoint: usize);
      // usb device driver resets data toggle for the given endpoint.
- ctrl_setup() of capsules::usb::usbc_client_ctrl::ClientCtrl handles these requests:
  - handle_standard_device_request() should call self.controller.set_config() on StandardRequest::SetConfiguration.
  - Add handle_standard_endpoint_request(), and let it call self.controller.endpoint_reset() on StandardRequest::ClearFeature.

bradjc, 2020-09-03:
I'm not sure I understand what the data toggle is. Is that a USB parameter?

namyoonw, 2020-09-03:
I meant a binary data packet ID (PID) in the low level. One party (host or dev) sends a packet with a PID, and sends next packet with PID toggled.

https://www.beyondlogic.org/usbnutshell/usb4.shtml

Sometimes, a sender and a receiver might not agree each other on a PID because of some accidental factors either physically (by power status change or connection change) or logically (by false firmware logic).

bradjc, 2021-02-22:
Moved to #1940

closed 2021-02-22 by bradjc