tock/tock #2099: Interest in a CTAP capsule in mainline Tock?

gendx, 2020-09-08:
In https://github.com/tock/tock/pull/2094, it was discussed to introduce a CTAP HID implementation in Tock. In the context of OpenSK ([patches/tock/02-usb.patch](https://github.com/google/OpenSK/blob/master/patches/tock/02-usb.patch)), we've already implemented and tested that, but also defined an actual `Driver` interface for userspace applications to send/receive CTAP packets via USB.

Is there an interest in integrating that patch in mainline Tock (including reserving a driver number in `capsules/src/driver.rs`)? It would obviously simplify the maintenance burden on OpenSK's side, and could benefit from testing on more boards supported by Tock, and more generally for applications that also use a USB HID protocol.

If yes, I can prepare a pull request for that.

alistair23, 2020-09-08:
Can you apply any missing features on top of https://github.com/tock/tock/pull/2094?

gendx, 2020-09-09:
> Can you apply any missing features on top of #2094?

What is the list of these "any" missing features?

To be clear, I'd propose to send the OpenSK patch (refactored to support the latest USB changes in Tock), and we can discuss in the review some potential shortcomings. And once that is merged, anyone is of course free to extend it with missing features that come up.

bradjc, 2021-02-22:
We support CTAP now, so I would say the answer is yes :)

closed 2021-02-22 by bradjc