tock/tock #2123: [sam4l] Uniformization of USB packet lengths across platforms?

gendx, 2020-09-23:
In `capsules/src/usb/usbc_client.rs`, we define platform-dependent maximum USB packet lengths.

https://github.com/tock/tock/blob/77598c10a7ff40e1911e66731d10381ae4255244/capsules/src/usb/usbc_client.rs#L28-L33

The distinction for sam4l seems to come from the current configuration, i.e. using `EPSIZE::Bytes8` in e.g. the following.

https://github.com/tock/tock/blob/77598c10a7ff40e1911e66731d10381ae4255244/chips/sam4l/src/usbc/mod.rs#L1525-L1528

However, it seems that sam4l can support many more packet lengths.

https://github.com/tock/tock/blob/77598c10a7ff40e1911e66731d10381ae4255244/chips/sam4l/src/usbc/mod.rs#L164-L173

Other platform would also work with other packet lengths I think, it's mainly a matter of setting the length in the `DeviceDescriptor`.

https://github.com/tock/tock/blob/ea4acbe3423253a3e5b67fc4af6cb661d98f1b41/capsules/src/usb/descriptors.rs#L310-L311

Can we uniformize across platforms by always using 64-byte USB packets? This would greatly simplify things by removing the "platform-specific" constants (which really are not platform-specific but configuration-specific), at least until a compelling use case arises for another packet length.

Does anyone has context on the sam4l to confirm it would work? If nobody has context, I haven't seen much support for sam4l's USB lately, so as a default can we assume that 64 bytes is fine?

---

I also don't understand why these constants are duplicated in the CDC capsule:

https://github.com/tock/tock/blob/ea4acbe3423253a3e5b67fc4af6cb661d98f1b41/capsules/src/usb/cdc.rs#L35-L40

hudson-ayers, 2020-10-19:
I tried to look into testing this on Imix, but the USB tests fail on my machine even without changing the lengths to 64. That being said, I cannot think of a reason that 8 byte buffers should be required, though the test applications in `tools/usb/` might need to be updated to reflect the length change.

I am personally unconvinced that it is important we maintain a requirement for 8 byte buffers throughout the USB stack. Properly supporting low speed USB will require a number of changes beyond what we have today -- ex: the sam4l stack exclusively uses bulk endpoints, which are not low speed compatible. Very few types of USB devices work properly at low speed, and the lack of const generics support in Rust means that supporting multiple buffer sizes imposes overhead and adds complexity. I am not convinced this complexity is worthwhile, and think that switching sam4l to also use 64 byte packets would be a good change. It would be nice if someone was able to test the sam4l with 64 byte packets, though.

Also, agreed that the duplication of constants is undesirable.

bradjc, 2021-02-22:
Moved to #1940

closed 2021-02-22 by bradjc