tock/tock #2124: CI-Job: Rustdoc Documentation not enforced on GitHub?

gendx, 2020-09-24:
I just ran `make ci-nosetup` and this triggered the following error:

```rust
********************************************************************************
* CI-Job: Rustdoc Documentation                                                *
********************************************************************************

...
Documenting nano33ble
make[3]: Entering directory '/tock/boards/nano33ble'
...
error: unused attribute
 --> boards/nano33ble/src/main.rs:6:1
  |
6 | #![no_main]
  | ^^^^^^^^^^^
  |
  = note: `-D unused-attributes` implied by `-D warnings`
```

ppannuto, 2020-09-25:
[image]

Looks like 
```
boards/Makefile.common: $(Q)RUSTDOCFLAGS='-Z unstable-options --document-hidden-items -D warnings' $(CARGO) --color=always doc $(VERBOSE) --release --package $(PLATFORM) --target-dir=$(TARGET_DIRECTORY) 2>&1 | tee /dev/tty | grep -q warning && (echo "Warnings detected during doc build" && if [[ $$CI == "true" ]]; then echo "Erroring due to CI context" && exit 33; fi) || if [ $$? -eq 33 ]; then exit 1; fi
```

isn't working like it should..

hudson-ayers, 2020-09-25:
that looks like our classic "works in bash, but not in the default ubuntu shell (dash)" problem.

alistair23, 2020-09-25:
Can we force the CI to use Bash?

hudson-ayers, 2020-09-25:
> Can we force the CI to use Bash?

We actually do do this in the main Makefile (`SHELL := bash` is line 13 of the Makefile), but not in `boards/Makefile.common`, so perhaps we should apply it there as well. Unfortunately it seems that the Makefile calling a shell script in tools which then calls make again is enough to run some code with the system shell unless every Makefile specifies the shell as bash

alistair23, 2020-09-25:
Could we run this as part of the CI setup?
```shell
rm /bin/sh
ln -s /bin/sh /bin/bash
```

hudson-ayers, 2020-09-25:
Probably, but we would have to carefully gate it such that it only actually is executed as part of a github action -- I can imagine developers being upset if we changed their default shell on them when they were just trying to test the Tock CI on a PR. Also, who knows if any other ubuntu-specific aspects of the actions workflow depend on the default shell being dash (unlikely, but possible..?)

Changing the default shell in Makefile.common works (just tested) so I would probably prefer that

commit fd1cdd88 (2020-09-26): Merge #2129

2129: fix rustdoc CI, force bash for Makefile.common r=ppannuto a=hudson-ayers

### Pull Request Overview

This pull request fixes #2124 by uniting `boards/Makefile.common` and `Makefile` such that `bash` is the default shell for both. This fixes an issue where when `Makefile.common` was called by a shell script from the `Makefile` on the ubuntu CI machine, it would use the default ubuntu shell `dash` which does not support the syntax used to detect warnings in rustdoc.

### Testing Strategy

This pull request was tested by running the CI locally on Ubuntu 20.04 before and after t [... cut]
PR #2129 (2020-09-26): fix rustdoc CI, force bash for Makefile.common
### Pull Request Overview

This pull request fixes #2124 by uniting `boards/Makefile.common` and `Makefile` such that `bash` is the default shell for both. This fixes an issue where when `Makefile.common` was called by a shell script from the `Makefile` on the ubuntu CI machine, it would use the default ubuntu shell `dash` which does not support the syntax used to detect warnings in rustdoc.

### Testing Strategy

This pull request was tested by running the CI locally on Ubuntu 20.04 before and after this change.

### TODO or Help Wanted

N/A

### [... cut]
closed 2020-09-26 by bors[bot]