tock/tock #2132: HiFive: userspace console writes <= 8 bytes dropped

jrvanwhy, 2020-09-28:
Note: I do not have a real HiFive to test against, so I have only tested in QEMU. It's possible this is a QEMU issue rather than a Tock issue. I'd appreciate help in reproducing this on a real HiFive.

While working on a `libtock-rs` example, I found that console writes of 8 bytes or less are dropped silently (that is, no bytes are transmitted). The `command` call that triggers the write returns `SUCCESS`, and the write callback is invoked with the correct (nonzero) number of bytes, but I never see the output appear in QEMU's console. Writes of 9 bytes or larger seem to work.

Reproduction steps:

1. Check out the `console-bug` branch of https://github.com/jrvanwhy/libtock-rs
1. `make kernel-hifive`
1. `cd core`
1. `./test.sh`

Expected output:

```
HiFive1 initialization complete.
Entering main loop.
Hello, World!
```

Actual output:
```
HiFive1 initialization complete.
Entering main loop.
```

bradjc, 2020-09-29:
This app works for me on hifive1b hardware:

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>

#include <console.h>

char hello[] = "Helloo\n";

static void nop(
  int a __attribute__((unused)),
  int b __attribute__((unused)),
  int c __attribute__((unused)),
  void* d __attribute__((unused))) {}

int main(void) {
  putnstr_async(hello, sizeof(hello), nop, NULL);
  return 0;
}
```

compiled with `make RISCV=1 APP_HEAP_SIZE=512 STACK_SIZE=1024`.

I get:

```
Bench Clock Reset Complete

ATE0--> Send Flag error: #0 #0 #0 #0 AT+BLEINIT=0-->AT+BLEINIT=0
OK
AT+CWMODE=0-->AT+CWMODE=0
OK

HiFive1 initialization complete.
Entering main loop.
Helloo
␀
```

I also enabled strace and rejiggered the app and got:

```
Bench Clock Reset Complete

ATE0--> Send Flag error: #0 #0 #0 #0 AT+BLEINIT=0-->AT+BLEINIT=0
OK
AT+CWMODE=0-->AT+CWMODE=0
OK

HiFive1 initialization complete.
Entering main loop.
[0] function_call @0x20040068(0x20040040, 0x80002800, 0x1704, 0x80003400)
[0] memop(0, 0x80003544) = 0x0 = SUCCESS
[0] memop(10, 0x80002a00) = 0x0 = SUCCESS
[0] memop(11, 0x80003544) = 0x0 = SUCCESS
[0] allow(0x1, 1, @0x8000339c, 0x8) = 0x0 = SUCCESS
[0] remove_pending_callbacks[0x1:1] = 0 callback(s) removed
[0] subscribe(0x1, 1, @0x200400bc, 0x0) = 0x0 = SUCCESS
[0] cmd(0x1, 1, 0x8, 0x0) = 0x0 = SUCCESS
[0] yield
Helloo
␀[0] schedule[0x1:1] @0x200400bc(0x8, 0x0, 0x0, 0x0) = true
[0] function_call @0x200400bc(0x8, 0x0, 0x0, 0x0)
[0] yield
```

and made the print smaller and got:

```
Bench Clock Reset Complete

ATE0--> Send Flag error: #0 #0 #0 #0 AT+BLEINIT=0-->AT+BLEINIT=0
OK
AT+CWMODE=0-->AT+CWMODE=0
OK

HiFive1 initialization complete.
Entering main loop.
[0] function_call @0x20040068(0x20040040, 0x80002800, 0x1704, 0x80003400)
[0] memop(0, 0x80003544) = 0x0 = SUCCESS
[0] memop(10, 0x80002a00) = 0x0 = SUCCESS
[0] memop(11, 0x80003544) = 0x0 = SUCCESS
[0] allow(0x1, 1, @0x8000339c, 0x6) = 0x0 = SUCCESS
[0] remove_pending_callbacks[0x1:1] = 0 callback(s) removed
[0] subscribe(0x1, 1, @0x200400bc, 0x0) = 0x0 = SUCCESS
[0] cmd(0x1, 1, 0x6, 0x0) = 0x0 = SUCCESS
[0] yield
Helo
␀[0] schedule[0x1:1] @0x200400bc(0x6, 0x0, 0x0, 0x0) = true
[0] function_call @0x200400bc(0x6, 0x0, 0x0, 0x0)
[0] yield
```

jrvanwhy, 2020-09-30:
Turns out this isn't a kernel bug at all. For some reason, strings of 8 or fewer bytes are not populated in `.data` in libtock-rs apps! I'm still trying to determine *why* they're not populated correctly.

jrvanwhy, 2020-10-01:
The issue is that strings of <= 8 bytes are put in `.sdata`, not `.data`, which is not included in the `libtock-rs` linker script. I'll send a PR for the fix. I'm closing this issue because it is in the wrong repository.

closed 2020-10-01 by jrvanwhy