tock/tock #2135: Grant API is unsound?

hudson-ayers, 2020-10-01:
After some investigation, I think the grant API is unsound.

The following code compiles in a Tock capsule today. Incidentally, it currently prints "1" , but I believe it could print "11" if no-aliasing optimizations were enabled in Rust (which has happened in the past and will happen in the future).

```rust
struct App {
    dummy: u32,
}
self.apps.enter(appid, |app, _| {
    let mutable_ref = &mut app.dummy;
    *mutable_ref = 10;
    self.apps.enter(appid, |app, _| {
        app.dummy = 0;
    }).unwrap();
    *mutable_ref += 1;
    kernel::debug!("dummy: {:?}", mutable_ref);
});
```

The problem here is that it is possible to enter a grant region that has already been entered -- `Grant::enter()` and `Grant::each()` do not check if the grant region being entered has already been entered when providing mutable access to the contents. 

https://github.com/tock/tock/blob/e9c5508d2debd1d63eff1d0cad94248dab85df8f/kernel/src/grant.rs#L310 is the particular problematic line of unsafe in `enter()`. 

https://github.com/tock/tock/blob/e9c5508d2debd1d63eff1d0cad94248dab85df8f/kernel/src/grant.rs#L68 is the problematic line of unsafe that is made accessible by `each()` (Confusingly, `each()` is not just syntactic sugar for `for app in apps.iter() { app.enter() }`).

Unfortunately, code already exists that utilizes this API in a way that could be unsound.

For example, in the ble_advertising_driver: https://github.com/tock/tock/blob/2acd4c56ab56b9d98c9501256b63594165a8e1ca/capsules/src/ble_advertising_driver.rs#L574

A closure passed to `Grant::enter()` contains a function call that calls `enter()` itself.

I found another example in the UDP driver: https://github.com/tock/tock/blob/e9c5508d2debd1d63eff1d0cad94248dab85df8f/capsules/src/net/udp/driver.rs#L531

I suspect there are others.

So far, I can think of two possible solutions:

1. Change the definition of `Grant` and `Grant::enter()` as follows:
```rust
pub struct Grant<T: Default> {
    pub(crate) kernel: &'static Kernel,
    grant_num: usize,
    ptr: PhantomData<T>,
    accessed: Cell<bool>,
}
    pub fn enter<F, R>(&self, appid: AppId, fun: F) -> Result<R, Error>
    where
        F: FnOnce(&mut Borrowed<T>, &mut Allocator) -> R,
        R: Copy,
    {
        appid
            .kernel
            .process_map_or(Err(Error::NoSuchApp), appid, |process| {
                if let Some(untyped_grant_ptr) = process.get_grant_ptr(self.grant_num) {
                        // ... allocate code omitted
                    } else {
                        untyped_grant_ptr as *mut T
                    };
                    if !accessed.get() {
                         accessed.set(true);
                         // Dereference the typed GrantPointer to make a GrantRegion
                         // reference.
                        let region = unsafe { &mut *typed_grant_pointer };
                        let mut borrowed_region = Borrowed::new(region, appid);

                        // Call the passed in closure with the borrowed grant region.
                        let res = fun(&mut borrowed_region, &mut allocator);
                        accessed.set(false);
                        Ok(res)
                    } else {
                        Err(Error::Busy)
                    }
                } else {
                    Err(Error::InactiveApp)
                }
            })
    }
```
This requires capsules to detect and handle reentrant calls at runtime.

2. Change `grant.enter()` to provide the passed closure with an immutable reference to the type stored in the grant, and fields of that type can only be mutated if they are behind some type of cell. So you would see code like this:

```rust
struct App {
    num: Cell<u32>,
    buf: TakeCell<'static, [u8]>,
    appslice: RefCell<AppSlice<Shared, u8>>,
}
```
This allows for primitives to be accessed and modified safely despite multiple simultaneous accesses to grants (I think Cell should be sufficient because of the single threaded nature of the Tock Kernel). For things like `AppSlice`, simultaneous access would lead to panics rather than unsoundness, thanks to the semantics of `RefCell`.

3. Make `grant.enter()` and `grant.each()` require `&mut self` instead of `&self`, to make reentrant calls impossible. This would probably require large scale changes to capsule code, but maybe could be made easier by storing the Grant in something like a `TakeCell`?

hudson-ayers, 2020-10-01:
https://github.com/hudson-ayers/tock/tree/safe-grant-take-2 has a working fix in the style of option 1, without using const generics. It imposes an artificial limit of 32 processes at a time, but then again IPC already imposes an artificial limit of 8 processes at a time.

gendx, 2020-10-02:
> The problem here is that it is possible to enter a grant region that has already been entered -- `Grant::enter()` and `Grant::each()` do not check if the grant region being entered has already been entered when providing mutable access to the contents.

I think this is problematic. I thought the assumption was that `enter` APIs can only be entered once at a time.

> 1. Change the definition of `Grant` and `Grant::enter()` as follows.

This sounds like a reasonable fix in the short term. In the longer term, which should take these concerns into account when looking further into #1905.

> 2. [...] This allows for primitives to be accessed and modified safely despite multiple simultaneous accesses to grants.

I don't think it should be possible to `enter` twice the same resource simultaneously.

> 3. Make `grant.enter()` and `grant.each()` require `&mut self` instead of `&self`, to make reentrant calls impossible.

This would IMO be nice and the most "Rusty" way of doing things - i.e. leverage exclusive access rules of mutable references to ensure *at compile time* that things only happen once.

This is contrary to the widespread practice of using all kinds of `*Cell`s in Tock, which circumvent the rules but moves bugs to runtime errors:
- either we have a `panic` due to `unwrap`ing a `None`/`Err` value when taking a cell,
- or code is silently not being run when entering an already entered resource (assuming the resource implements entering semantics properly) - assuming the result is ignored instead of being `unwrapped`.

It's also hard to tell how entangled things are without actually starting to implement a prototype. Last time I tried to move some things away from `Cell`s to use references instead, it ended up being a large-scale rabbit hole. IIRC it was somewhat related to making lifetimes generic as opposed to `'static` (#1854), although I don't remember the details. But this doesn't mean that `Cell`s are unavoidable - although it would likely mean re-architecting the kernel and boards initialization to have clear ownership in line with Rust rules.

alevy, 2020-10-02:
@hudson-ayers excellent work, this is subtle and super bad.

I think something along the lines of option 1 is the right solution, but can be done with no artificial limits (e.g. a up to 32 grants or whatever). Specifically, grants should simply have `TakeCell` semantics and an implementation to match.

Currently, each grant is stored using a pointer indirection table at the top of the grant region. This table would simply contain `Option`al etnries instead of one-word poitner entries: each entry is a pointer as well as a `taken` field (maybe just implemented as the magic value `0` or something, as with `Option<&mut T>`). Then, accessing grants would work just like accessing a TakeCell works: before calling the closure we set the `taken` field to true iff it is not already taken, then restore the taken field after the closure returns.

hudson-ayers, 2020-10-02:
> Currently, each grant is stored using a pointer indirection table at the top of the grant region. This table would simply contain `Option`al etnries instead of one-word poitner entries: each entry [...]

Currently, the pointer being `Null` is used to indicate that a grant has not been setup yet, so we would probably need an additional magic value for "setup, but taken", but would probably want to do so without increasing the size of every entry in the table past a word.

alevy, 2020-10-02:
> but would probably want to do so without increasing the size of every entry in the table past a word.

I see. Well... increasing the size of the entry is definitely _less_ important that preserving soundness. And yeah, two magic vaules (e.g. `0` and `!0`) is probably reasonable.

closed 2021-02-15 by bors[bot]