tock/tock #2138: tock-registers: Next release to crates.io?

andre-richter, 2020-10-04:
Hi all, @ppannuto,

I see that tock-registers got a fair share of changes since the 0.5.0 release. Especially the switch to `UnsafeCell` looks important.

Can you we make a new release to crates.io, or are there any gating issues?

Thanks,
Andre

ppannuto, 2020-10-05:
No gating issues.. just hasn't hit the top of the priority queue here :/

We're very near a feature freeze for Tock 1.6 release. The release testing should more thoroughly cover all of the register changes -- not that we've seen any problems yet. Let's take advantage of that, and I'll plan on kicking out a tock-registers release contemporaneously with Tock 1.6 [likely timeline >1 week, <1 month].

hudson-ayers, 2020-10-24:
@ppannuto gentle reminder about this

PR #2191 (2020-11-13): tock-registers: release v0.6
Changes since v0.5:

 - #2095: Fix syntax errors and inconsistencies in documentation
 - #2071: Clarify bit widths in documentation examples
 - #2015: Use UnsafeCell in registers (see issue #2005)
 - #1939: Make the Field::mask and FieldValue::mask fields private
 - #1823: Allow large unsigned values as bitmasks + add bitmask! helper macro
 - #1554: Allow lifetime parameters for `register_structs! { Foo<'a> { ..`
 - #1661: Add `Aliased` register type for MMIO with differing R/W behavior

Thanks to @namyoonw, @brghena, @daboross, @gendx, @hudson-ayers, and @pfmooney [... cut]
commit 72a1e1ea (2020-11-13): tock-registers: release v0.6

Changes since v0.5:

 - #2095: Fix syntax errors and inconsistencies in documentation
 - #2071: Clarify bit widths in documentation examples
 - #2015: Use UnsafeCell in registers (see issue #2005)
 - #1939: Make the Field::mask and FieldValue::mask fields private
 - #1823: Allow large unsigned values as bitmasks + add bitmask! helper macro
 - #1554: Allow lifetime parameters for `register_structs! { Foo<'a> { ..`
 - #1661: Add `Aliased` register type for MMIO with differing R/W behavior

Thanks to @namyoonw, @brghena, @dabross, @gendx, @hudson-ayers, and @pfmooney [... cut]
closed 2020-11-13 by bradjc