tock/tock #2143: Alarm driver does not work with 64-bit alarms

lschuermann, 2020-10-09:
While porting to a new board that features a 64-bit HW Timer and hence a 64-bit SW Alarm peripheral, I've noticed weird behavior where an Alarm would be set, fire, and then be immediately reset to the same alarm time (now in the past).

I've managed to trace this down to the Alarm driver, specifically the Alarm callback function that checks for each app whether a set alarm has expired.

Since the alarm times (`reference` and `dt`) are stored as 32-bit values, adding `dt` to `reference` will potentially wrap at reaching `u32::MAX`, however the comparison to the current time will still indicate that the current time is in between `reference` and `dt`, even though the alarm time would have passed.

https://github.com/tock/tock/blob/ba867c2c269325133f212b757b125b114ab9f2ad/capsules/src/alarm.rs#L260-L270

This will cause `num_armed` to not be decreased, which will in turn call `reset_active_alarm()`, setting it to the original alarm time  and hence firing immediately, resulting in a loop.

https://github.com/tock/tock/blob/ba867c2c269325133f212b757b125b114ab9f2ad/capsules/src/alarm.rs#L284-L291

## Current behavior

Given a 64-bit alarm peripheral, once an app-alarm is set in a way that `reference < u32::MAX && (reference + dt) > u32::MAX`, the alarm driver will call `set_alarm(` on the alarm with constant arguments in a loop. 

## Expected behavior

Handle alarm times correctly and don't hang in a loop.

## Example

For example, given

| Field                                        | Value                    |
|---|---|
| `reference`                              | `4_294_967_263` |
| `dt`                                          | `64`                       |
| `reference.wrapping_add(dt)` | `32`                       |
| `now`                                      | `4_294_967_370`  |

the problem is evident, since technically `reference < reference + dt < now` (and hence the alarm should fire), but given the particular wrapping semantics in the comparison as done currently, `reference < now < reference.wrapping_add(dt)` (`4_294_967_263 < 4_294_967_370 < 32`).

While changing this code to read
```rust
                if !now.within_range(
                    A::Ticks::from(reference),
                    A::Ticks::from(reference).wrapping_add(A::Ticks::from(dt)),
                ) {
```
and hence use 64-bit wrapping both in the comparison and the addition does solve the alarm looping issue, it's technically incorrect since it will compare the 64-bit absolute time to the 32-bit alarm time set in userspace and hence at some point fire *before* the userspace alarm time.

## Proposed approach

I believe the only reasonable way to reliably (and comprehensibly!) support 64-bit Alarms would be to rewrite the driver to use the native Alarm::Ticks underneath. I can offer to do this, but wanted to open this issue nonetheless to prevent people from also spending a few days (on a chip with little debug infrastructure established) debugging this.

phil-levis, 2020-10-14:
I'm sorry it took me a while to get to this. I'm looking at it now.

Would a simpler fix to be to just move down to 32 bits? The issue is that 64 and 32 bits are being used at the same time. So something like this:

```rust
                 let now: Ticks32 = Ticks32::from(self.alarm.now().into_u32()); 
                 if !now.within_range( 
                     Ticks32::from(reference), 
                     Ticks32::from(reference.wrapping_add(dt)), 
                 ) {
```

That way we are basically just pretending that the underlying 64-bit counter is 32 bits. The only time we need to scale back up to 64 is when we set the actual value for the alarm. This is handled in line 135.

phil-levis, 2020-10-14:
> I believe the only reasonable way to reliably (and comprehensibly!) support 64-bit Alarms would be to rewrite the driver to use the native Alarm::Ticks underneath.

We can't do this: the Driver needs to be independent of the underlying ticks and be in 32 bits. So we could have one implementation for each underlying width (a 24, a 32, and a 64) but I think 95% of the code will be identical. My problem is that I haven't been able to test a 64-bit userspace. Your example failure case is very useful. I will manually test with these values and try to come up with some other test cases.

phil-levis, 2020-10-14:
Can you try this:

https://github.com/tock/tock/tree/alarm-virt-64-fixes

lschuermann, 2020-10-15:
> We can't do this: the Driver needs to be independent of the underlying ticks and be in 32 bits. So we could have one implementation for each underlying width

I believe that thanks to the trait abstractions of `Ticks` we can write the code generic for all widths and store the pending alarms as the native `<Alarm as kernel::hil::time::Time>::Ticks` type. I've just opened #2160 as a proof of concept for this approach.

I do not mean to force one approach or the other, but think it could be helpful to compare both and check which one is preferable (if they both work, I have only performed a basic set of tests for #2160 and will test your approach shortly). I've outlined the advantages I see in the `A::Ticks` approach in the PR description.

lschuermann, 2020-10-15:
> If this fixes the problem I'll make it into a PR.

Based on my quick test from userspace, your solution does seem to fix the issue and behave identical to my proposal, both using commands `4` and `6`.

hudson-ayers, 2020-10-15:
Closed by #2164

closed 2020-10-15 by hudson-ayers

lschuermann, 2020-10-15:
It's unfortunate that #2160 did not even seem to be considered at all, but I guess that's just how it is. At least the issue does appear to be fixed by now, thanks.

hudson-ayers, 2020-10-15:
> It's unfortunate that #2160 did not even seem to be considered at all, but I guess that's just how it is. At least the issue does appear to be fixed by now, thanks.

I did not mean to indicate that #2160 will not be considered when I closed this, I just closed it because a fix for this particular issue was merged. I did not merge #2164, but generally I think that #2164 was clearly a bug fix to the already agreed upon alarm driver (and seems like how the alarm driver was originally intended to work), while #2160 proposes a new approach. Generally we do not merge new PRs during the release testing window, only bug fixes, so I would expect any discussion on #2160 to come after the 1.6 release.

lschuermann, 2020-10-15:
This makes sense, thank you for the explanation. The merge of #2164 seemed arbitrary given the very short time it was open and the lack of a last-call phase; however it is understandable since fix was confirmed (by me) and solves an issue which would otherwise be a release blocker. Nonetheless a comment would have been good, indicating that this is a quick fix in light of the 1.6 release and not necessarily a hard decision on the two solution approaches discussed here.