tock/tock #2195: Nano33Ble drops first few debug messages

hudson-ayers, 2020-11-12:
If you flash a nano33ble with no apps, then run `tockloader listen`, you do not see the "Initialization Complete..." message printed to the console. In fact, I found that I had to copy/paste this message 5 times in a row before I could see one printed to the console. Once one message has been printed, all subsequent messages are printed.

If the second debug statement occurs later (triggered from an app or in response to an interrupt), then it does successfully print. So the issue seems just to be with messages printed very near in time to initialization.

This could be a Tock problem or a tockloader problem, I have not looked into it further.

bradjc, 2020-11-12:
What OS are you using?

hudson-ayers, 2020-11-12:
Ubuntu 20.04

bradjc, 2020-11-13:
The issue appears to be that the CDC-ACM driver on the nano33 sends data to the host too early for some operating systems (namely Ubuntu 20.04 and MacOS 10.15). I just tried on a 20.04 VM and I can see the "Init complete" string in wireshark going over USB, but yeah I can't get it to display in a terminal.

Looking through RIOT and nRF5 SDK, it seems the common thing to do is wait for the `set line state` USB command before declaring that the host connected and it is ok to send data. Tock is actually waiting for a `set line coding` message with a baud rate of 115200 before sending data to the host, and that happens after the set line state message. I feel like there is a special way to get this to work, but I don't know what the trick is.

bradjc, 2020-11-13:
I'm just now remembering that I wrote this commit:

https://github.com/tock/tockloader/commit/f1899bc3460c13102f293a7e764de2acf5c4be82

which makes a big difference. With that patch to tockloader I now get all output from the nano33 on my 20.04 vm. And it works on mac 10.15 as well.

hudson-ayers, 2020-11-13:
ah, awesome, thanks. I will build tockloader from source then

closed 2020-11-13 by hudson-ayers