tock/tock #2201: Make netlify a required check

hudson-ayers, 2020-11-16:
The netlify CI is currently having issues, so I made it a non-required check in the Github UI. We should require it again once netlify is fixed. This issue serves as a reminder.

alistair23, 2020-11-17:
Any idea how to fix it? It seems like I can't even see the error without a login.

hudson-ayers, 2020-11-19:
I don't have access either, and haven't really looked into it.

ppannuto, 2020-11-20:
There are two issues, one is hopefully transient and had to do with the netlify cache doing bad things when a submodule updates (that's all the PR's hanging at `Preparing Git Reference refs/heads/master`) -- a clear cache and rebuild seems to fix those that still have a bad cache. I don't have the support thread open on this computer, but I think Netlify is working on or had a fix.

The other is that something changed in the tty/term config causing hangs, which I started diagnosing here, but ran out of time for. Hopefully will be able to get to it soonish.

hudson-ayers, 2021-01-06:
Netlify has been back working correctly for awhile, so I made it a required check again, which closes this.

closed 2021-01-06 by hudson-ayers