tock/tock #2222: RAM initialization code violates pointer provenance

jamesmunns, 2020-11-25:
RAM initialization currently passes pointers to these statics:

https://github.com/tock/tock/blob/master/chips/stm32f4xx/src/lib.rs#L65-L79

to `tock-rt0`:

https://github.com/tock/tock/blob/master/libraries/tock-rt0/src/lib.rs#L14-L34

But since r0 will fill the entire memory range between the pointers, it ends up calling offset with values that create pointers well beyond the u32 the static is declared to contain. According to the [docs of `offset`](https://doc.rust-lang.org/std/primitive.pointer.html#method.offset), this is 

> Both the starting and resulting pointer must be either in bounds or one byte past the end of the same allocated object. Note that in Rust, every (stack-allocated) variable is considered a separate allocated object.

(every `static` also counts as its own "allocated object", I believe)

Moreover, the `ptr::write` and `ptr::write_volatile` calls in r0 also cause UB, because this invariant is 

> dst must be valid for writes.

Where the [validity](https://doc.rust-lang.org/std/ptr/index.html#safety) of a pointer requires this:

> For a pointer to be valid, it is necessary, but not always sufficient, that the pointer be dereferenceable: the memory range of the given size starting at the pointer must all be within the bounds of a single allocated object. Note that in Rust, every (stack-allocated) variable is considered a separate allocated object.

...but again, we are crossing between potentially many allocated objects (every static in the .data/.bss section).

Furthermore, writing to a static through a pointer not derived from that static might violate aliasing rules, but it isn't yet clear if this is the case today (I've asked the folks working on the unsafe code guidelines for clarification / guidance).

The most robust solution for this issue is to write the RAM init code in assembly instead of Rust, and run no Rust code at all until RAM is initialized.

(this issue was recently discovered to affect most -rt crates in the ecosystem; we do not believe it to cause practical issues at the moment)

jrvanwhy, 2020-11-25:
> Furthermore, writing to a static through a pointer not derived from that static might violate aliasing rules, but it isn't yet clear if this is the case today (I've asked the folks working on the unsafe code guidelines for clarification / guidance).

Do you have a link to this discussion, or is it a private communication?

`libtock-rs`' runtime performs memory setup by casting `usize` values into `*mut u8` headers, then writing the memory using `core::ptr::{copy_nonoverlapping, write_bytes}`. I think that whether `libtock-rs`' runtime is sound depends on the answer to your question.

jamesmunns, 2020-11-25:
(sorry, I copy and pasted that issue report from @jonas-schievink's other issue reports. I'm not sure where the UCG discussion is going on, probably on the Rust-Lang Zulip).

This is the initial bug report we got, that caused us to investigate the issue: https://github.com/rust-embedded/embedonomicon/issues/69. It was also discussed on matrix in the wg meeting last night.

jonas-schievink, 2020-11-25:
The discussion is happening here: https://rust-lang.zulipchat.com/#narrow/stream/136281-t-lang.2Fwg-unsafe-code-guidelines/topic/The.20least.20incorrect.20init.20code.20.3A)

...but it hasn't resulted in any interesting conclusions yet

jrvanwhy, 2020-11-25:
Awesome, thank you both!

jonas-schievink, 2020-11-25:
> EDIT: More generally, jumping into Rust code before statics are initialized is probably UB.

This is currently our conclusion as well, yeah. To fix this, I've rewritten the RAM initialization in assembly, so feel free to take a look: https://github.com/rust-embedded/cortex-m-rt/pull/301

bradjc, 2021-01-08:
I think it would be a little bit of a shame to lose the feature of Tock where the reset handler is implemented directly in the top level main.rs (no hidden startup code! how cool is that?), but it also wouldn't be too hard to implement the rt0 functions in the /arch crate in asm and _then_ jump to the function in main.rs (which we should rename in that case).

bradjc, 2021-03-24:
Fixed in #2461

closed 2021-03-24 by bradjc