tock/tock #2231: Unable to load apps to qemu for Earlgrey

jackieke724, 2020-12-03:
Hello,

I am trying out tock on Earlgrey using QEMU, by following this [guide](https://github.com/tock/tock/tree/master/boards/earlgrey-nexysvideo). I am able to get Boot ROM initialization and kernel initialization working shown by the output below. However, I do not see "hello world" printed by loading the app.

```
make OPENTITAN_BOOT_ROM=/mnt/d/ubuntu_workspace/opentitan/build-bin/sw/device/boot_rom/boot_rom_fpga_nexysvideo.elf APP=/mnt/d/ubuntu_workspace/libtock-rs/rv32imc.tbf qemu-app
```
> info: This is the version for the rustup toolchain manager, not the rustc compiler.
> info: The currently active `rustc` version is `rustc 1.49.0-nightly (ffa2e7ae8 2020-10-24)`
>     Finished release [optimized + debuginfo] target(s) in 0.06s
>    text    data     bss     dec     hex filename
>   83681    5316   11064  100061   186dd /mnt/d/ubuntu_workspace/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo
> qemu-system-riscv32 -M opentitan -kernel /mnt/d/ubuntu_workspace/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo.elf -bios /mnt/d/ubuntu_workspace/opentitan/build-bin/sw/device/boot_rom/boot_rom_fpga_nexysvideo.elf -device loader,file=/mnt/d/ubuntu_workspace/libtock-rs/rv32imc.tbf,addr=0x20030000 -nographic -serial mon:stdio
> I00000 boot_rom.c:52] Version:    opentitan-snapshot-20191101-1-3310-g0b6831f6
> Build Date: 2020-12-02, 10:34:10
> 
> I00001 boot_rom.c:61] Boot ROM initialisation has completed, jump into flash! HELLOOO
> OpenTitan initialisation complete. Entering main loop HELLOOOOO
> QEMU: Terminated

The APP is generated in the libtock-rs repo as said in the guide.
```
make flash-opentitan EXAMPLE=hello_world
tar -xf target/riscv32imac-unknown-none-elf/tab/opentitan/hello_world.tab
```

Any insights on my issue would be really appreciated!

alistair23, 2020-12-03:
You shouldn't be running tar. The `make flash-opentitan` command will generate a tbf file that you should point to from from `APP`.

jackieke724, 2020-12-04:
Hi, @alistair23 , thanks for the suggestion. I have tried using `target/riscv32imc-unknown-none-elf/tab/opentitan/hello_world/rv32imc.tbf` directly. The result is the same. There is no difference when I diff those two files.

alistair23, 2020-12-04:
Ok. Can you enable debugging in the kernel? Set these to `true`: https://github.com/tock/tock/blob/master/kernel/src/config.rs#L48

jackieke724, 2020-12-04:
Thanks. The outputs are shown below:

> info: This is the version for the rustup toolchain manager, not the rustc compiler.                                                                               
> info: The currently active `rustc` version is `rustc 1.49.0-nightly (ffa2e7ae8 2020-10-24)`                                                                       
>    Compiling kernel v0.1.0 (/mnt/d/ubuntu_workspace/tock/kernel)                                                                                                  
>    Compiling rv32i v0.1.0 (/mnt/d/ubuntu_workspace/tock/arch/rv32i)                                                                                               
>    Compiling capsules v0.1.0 (/mnt/d/ubuntu_workspace/tock/capsules)                                                                                              
>    Compiling lowrisc v0.1.0 (/mnt/d/ubuntu_workspace/tock/chips/lowrisc)                                                                                          
>    Compiling earlgrey v0.1.0 (/mnt/d/ubuntu_workspace/tock/chips/earlgrey)                                                                                        
>    Compiling components v0.1.0 (/mnt/d/ubuntu_workspace/tock/boards/components)                                                                                   
>    Compiling earlgrey-nexysvideo v0.1.0 (/mnt/d/ubuntu_workspace/tock/boards/earlgrey-nexysvideo)                                                                 
>     Finished release [optimized + debuginfo] target(s) in 14.65s                                                                                                  
>    text    data     bss     dec     hex filename                                                                                                                  
>   88817    5972   10408  105197   19aed /mnt/d/ubuntu_workspace/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo                               
> #qemu-system-riscv32 -M opentitan -kernel /mnt/d/ubuntu_workspace/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo.elf -bios /mnt/d/ubuntu_work
> space/opentitan/build-bin/sw/device/boot_rom/boot_rom_fpga_nexysvideo.elf -device loader,file=/mnt/d/ubuntu_workspace/libtock-rs/target/riscv32imc-unknown-none-el
> f/tab/opentitan/hello_world/rv32imc.tbf,addr=0x20030000 -nographic -serial mon:stdio -append 'console=ttyS0' -s -S                                                
> qemu-system-riscv32 -M opentitan -kernel /mnt/d/ubuntu_workspace/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo.elf -bios /mnt/d/ubuntu_works
> pace/opentitan/build-bin/sw/device/boot_rom/boot_rom_fpga_nexysvideo.elf -device loader,file=/mnt/d/ubuntu_workspace/libtock-rs/target/riscv32imc-unknown-none-elf
> /tab/opentitan/hello_world/rv32imc.tbf,addr=0x20030000 -nographic -serial mon:stdio                                                                               
> I00000 boot_rom.c:52] Version:    opentitan-snapshot-20191101-1-3310-g0b6831f6                                                                                    
> Build Date: 2020-12-02, 10:34:10                                                                                                                                  
>                                                                                                                                                                   
> I00001 boot_rom.c:61] Boot ROM initialisation has completed, jump into flash! HELLOOO                                                                             
> Loading processes from flash=0x20030000-0x200FFFFF into sram=0x10004000-0x1000FFFF                                                                                
> Loaded process[0] from flash=0x20030000-0x20030FFF into sram=0x10004000-0x10005013 = "hello_world"                                                                
> OpenTitan initialisation complete. Entering main loop HELLOOOOO

alistair23, 2020-12-04:
Hmmm... Your application is loading, but QEMU is stuck in an interrupt loop.

jackieke724, 2020-12-04:
I was just learning to use gdb to debug the kernel. Any tips for tracing, i.e. where to insert breakpoints?

alistair23, 2020-12-04:
Thank you for trying to help debug the issue.

I have figured it out though, it's an issue with how we save pending interrupts in the Ibex PLIC model inside QEMU. Unfortunately the Ibex PLIC isn't standards compliant so we have a custom implementation in QEMU. I have been trying to test and catch all of the bugs (in the code I wrote), but somehow missed this, thanks for helping test.

I will send a patch to QEMU, if you would like to be CCed on the patch and follow the discussion I can, but I will need an email address. In the mean time, applying this diff to QEMU should fix the issue:

```diff
diff --git a/hw/intc/ibex_plic.c b/hw/intc/ibex_plic.c
index 341c9db405..bc422fc681 100644
--- a/hw/intc/ibex_plic.c
+++ b/hw/intc/ibex_plic.c
@@ -43,6 +43,10 @@ static void ibex_plic_irqs_set_pending(IbexPlicState *s, int irq, bool level)
 {
     int pending_num = irq / 32;
 
+    if (!level) {
+        s->hidden_pending[pending_num] &= ~(1 << (irq % 32));
+    }
+
     if (s->claimed[pending_num] & 1 << (irq % 32)) {
         /*
          * The interrupt has been claimed, but not completed.
```

jackieke724, 2020-12-04:
Thank you so much for the quick fix! Please add me to the discussion jackieke724@hotmail.com. 
I am now able to see the Tock World :)

closed 2020-12-04 by jackieke724

brghena, 2020-12-04:
Hi @jackieke724. You can add yourself to whatever groups you'd like.

We have an email group (~weekly conversation): https://groups.google.com/forum/#!forum/tock-dev

And we have a Slack workspace (~daily conversation): https://join.slack.com/t/tockos/shared_invite/enQtNDE5ODQyNDU4NTE1LWVjNTgzMTMwYzA1NDI1MjExZjljMjFmOTMxMGIwOGJlMjk0ZTI4YzY0NTYzNWM0ZmJmZGFjYmY5MTJiMDBlOTk

jackieke724, 2020-12-04:
Joined, thank you @brghena !

alistair23, 2020-12-04:
I just sent the patch to the QEMU mailing list, you should be included in CC.

alistair23, 2020-12-04:
If you would like you can help get the patch merged quicker by replying saying that you have tested it.

You would do this by replying in plain text with the following:

```
Tested-by: First Last <jackieke724@hotmail.com>
```

You are also free to review it if you feel comfortable with that.

jackieke724, 2020-12-05:
Replied!