tock/tock #2235: Porting system call capsules to Tock 2.0 system call API

phil-levis, 2020-12-04:
We've now reached a point when we're almost ready to port all of the system call capsules to the new Tock 2.0 system call API. All of this work is going on in the [tock-2.0-dev branch](https://github.com/tock/tock/tree/tock-2.0-dev).

This issue is tracking who is porting which capsules and their status. I've clustered them by complexity. Please sign up to port and check off when done and tested. We have a [porting guide](https://github.com/tock/tock/blob/tock-2.0-dev/doc/Porting_Capsules_2.md) that you are free to update/refine as you learn new things.

**Complex/Tricky**

- [ ]  adc (**unclaimed**): tricky because of the buffering/semantics of high speed sampling, plus there are two implementations (virtualized and dedicated)
- [ ] alarm (@phil-levis): tricky because virtualization is not FIFO, traversing app states in a complex way
- [ ] app_flash_driver (**unclaimed**): tricky because there's careful buffer management/copies
- [ ] ble_advertisement (**unclaimed**): tricky because there's a complex state machine and buffer management
- [ ] ctap (**unclaimed**): tricky because there are many different operations and buffer management
- [ ] hd44780 (**unclaimed**): complex protocols with the display
- [ ] i2c_master_slave_device (**unclaimed**):  Handles many different I2C cases

** Medium/Non-trivial**
- [ ] button (**unclaimed**): Needs a bunch of extra checks and issues of pending events
- [ ] crc (**unclaimed**): has buffers
- [ ] gpio (**unclaimed**): nothing super-complicated but has all of the moving parts and lots of operations
- [ ] gpio_async (**unclaimed**): nothing super-complicated but has all of the moving parts and lots of operations
- [ ] hmac (**unclaimed**): buffers and some complex logic
- [ ] i2c_master (**unclaimed**): Call and response but not complex
- [ ] l3gd20 (**unclaimed**): A lot of calls, but pretty simple structure/control flow
- [ ] lsm303dlhc (**unclaimed**): A lot of calls, but pretty simple structure/control flow
- [ ] ltc294x.rs
** Easy/Simple **
- [ ] ambient_light (**unclaimed**): Just a call and a callback, virtualization is just multiple serving, no queueing
- [ ] analog_comparator (**unclaimed**): Just a call and a callback
- [ ] buzzer_driver (**unclaimed**): Incorporates alarms
- [ ] dac (**unclaimed**): As simple as it gets!
- [ ] ft6x06 (**unclaimed**): Doesn't actually do anything.
- [ ] humidity (**unclaimed**): Simple

brghena, 2020-12-04:
Is this list meant to be complete? At the minimum, it's missing the SD Card capsule: https://github.com/tock/tock/blob/master/capsules/src/sdcard.rs#L1486

phil-levis, 2020-12-04:
I thought it better to save it in progress than risk it all disappearing, then add more. I was going alphabetically.

brghena, 2020-12-04:
I was just too quick. Got it! Thanks for writing this list up Phil.

hudson-ayers, 2020-12-04:
I may also just be too quick, but this seems to still be missing all of the drivers that are hidden behind folders (e.g. `ieee802154/driver.rs`, `net/udp/driver.rs`, `low_level_debug/mod.rs`, etc.)

phil-levis, 2020-12-04:
> I may also just be too quick, but this seems to still be missing all of the drivers that are hidden behind folders (e.g. `ieee802154/driver.rs`, `net/udp/driver.rs`, `low_level_debug/mod.rs`, etc.)

Ah right, will add them tonight.

phil-levis, 2020-12-05:
I claimed most of the ones that can be tested on imix. Depending on how they go I might pick up more.

phil-levis, 2020-12-07:
@bradjc @alevy @niklasad1 -- What are the calling conventions/behaviors of the traits in `hil::sensors`? For example, if I call `hil::sensors::HumidityDriver::read_humidity` what can it return? I can look through examples to see what they do return, but I'm not sure this will cover the set. In particular, is there a SuccessWithValue or merely a Success? Is this documented somewhere?

hudson-ayers, 2020-12-07:
(porting from mailing list)

I think there is a little ambiguity in the current porting guide about how far ports should go towards moving capsules away from use of `ReturnCode`. The LED example removes uses of `ReturnCode` and replaces them with directly constructed `CommandResult`s. The console example, on the other hand, sticks with the use of `ReturnCode` everywhere and just converts it to the appropriate `CommandResult` at the end. What is the expectation here? In many cases it is probably much easier to leave the body of `command()` mostly untouched, and convert the `ReturnCode` to a `CommandResult` at the end. But it is more efficient to switch things over earlier, and switching things earlier is the first step towards moving to `Result` based error handling.

Response:

Okay to do whatever is easiest in each case, and worry about larger-scale refactorings later

phil-levis, 2020-12-07:
> I think there is a little ambiguity in the current porting guide about how far ports should go towards moving capsules away from use of `ReturnCode`. The LED example removes uses of `ReturnCode` and [...]

In those cases, I tried to change the capsules as little as possible. For LED, the easiest thing seemed to be to just change the underlying methods, while for the more complex console I wanted to keep its core logic untouched. For now, don't worry about refactoring/rewriting capsules. The goal is to make the smallest set of changes possible to port them to the new API. Once we have that, we can always improve them later. But if we start a full refactoring pass, we will never finish.

hudson-ayers, 2020-12-07:
I am finding that the "swap" based approach for subscribe is very easy to get wrong, because `Subscribe` implements `Copy`. There are a number of capsules which are not yet virtualized across multiple applications, and as a result do not use grants yet -- take `analog_comparator`, for example. Instead, they currently store `Callback`s as follows

```rust
pub struct AnalogComparator {
    //...
    callback: Cell<Option<Callback>>,
}
```

Based on the changes to Callback for 2.0, it is logical to change this to
```rust
pub struct AnalogComparator {
    //...
    callback: Cell<Callback>,
}
```
as the `Option` is now internal to the `Callback` type.

My instinct when porting this to new syscall design was to do the following, based on the porting guide:
```rust
fn subscribe(...) {
    mem::swap(&mut self.callback.get(), &mut callback);
}
```
which compiles fine! But it ends up not working, because the memory you end up swapping is a *copy* of the stored callback with the passed callback (returned by `Cell::get()`). Not using `Cell` is not an option, because then you cannot get the mutable reference to `Callback` required for `mem::swap()`. The correct approach is actually the following:

```rust
fn subscribe(...) {
    self.callback.swap(Cell::from_mut(&mut callback));
}
```

which somewhat confusingly involves creating a `Cell` out of the passed callback just so that `core::cell::Cell::swap()` can be used.

Unfortunately, legacy uses of `Callback` rely on it being `Copy`, so it will not be easy to remove this footgun while people are porting capsules. I supposed we will just need to be careful that this is done correctly.

phil-levis, 2020-12-07:
> Unfortunately, legacy uses of `Callback` rely on it being `Copy`, so it will not be easy to remove this footgun while people are porting capsules. I supposed we will just need to be careful that this is done correctly.

I will take a look at AnalogComparator tonight to see the pattern and see if I can come up with a good solution. It's an open question whether `Callback` can be `Copy`. It doesn't have the same restrictions as an application slice does. But we might want to remove `Copy` so buffer swaps become the standard approach.

Since `Callback` implements `Copy`, you don't need any magic here. You should just be able to copy in and out, right? E.g.:

```rust 
    let returned_cb = self.callback.get();
    self.callback.set(passed_cb);
    return Ok(returned_cb);
```

hudson-ayers, 2020-12-07:
> I will take a look at AnalogComparator tonight to see the pattern and see if I can come up with a good solution. It's an open question whether `Callback` can be `Copy`. It doesn't have the same [...]

Hmm, okay, yes I suppose that does work. I was operating under the assumption that the swap technique was important -- `mem::swap()` ensures that the passed callback is immediately replaced with the returned one, but I guess the returned callback is passed as the return value anyway so that is not necessary. 

Even better:
```rust
return Ok(self.callback.replace(passed_cb));
```

phil-levis, 2020-12-08:
> Even better:
> 
> ```rust
> return Ok(self.callback.replace(passed_cb));
> ```

Yes!

alexandruradovici, 2020-12-11:
Is there any way to find out if a callback has been set? The `Option` is now internal to `Callback`, so `callback.is_some()` does not work anymore.

hudson-ayers, 2020-12-11:
Looks like no, not without trying to actually schedule a callback. I see no problem with adding an `is_set()` method though.

Also, if you are taking on porting any more capsules feel free to post them here and I can edit the top-level comment so we avoid duplicate work :)

alexandruradovici, 2020-12-11:
Thanks, I added the `is_set` function, just wanted to make sure I did not miss anything. I am porting right the `touch` driver and I am planning to do `l3gd20` and `lsm303xx`.

alexandruradovici, 2020-12-11:
> Looks like no, not without trying to actually schedule a callback. I see no problem with adding an `is_set()` method though.
> 
Scheduling a callback will not give you that information, as it returns `true` if `cb` is `None`:

```rust
pub fn schedule(&mut self, r0: usize, r1: usize, r2: usize) -> bool {
        self.cb.map_or(true, |mut cb| cb.schedule(r0, r1, r2))
}
```

Should this be 

```rust
pub fn schedule(&mut self, r0: usize, r1: usize, r2: usize) -> bool {
        self.cb.map_or(false, |mut cb| cb.schedule(r0, r1, r2))
}

```

phil-levis, 2020-12-11:
You should just invoke the callback: if it is the Null Callback, `schedule` will do nothing.

alexandruradovici, 2020-12-11:
Normally I would do that, but the touch panel is enabled and disabled based on whether there are callbacks registered or not. I write do a counter, but I have no way of knowing if an app provided a NULL callback or not.

hudson-ayers, 2020-12-16:
One thing that I expect will cause unanticipated issues:

The type signature of `schedule` has not changed:
```rust
 pub fn schedule(&mut self, r0: usize, r1: usize, r2: usize) -> bool;
```

`schedule()` is frequently used to report errors via callbacks when operations cannot be performed asynchronously. A frequent pattern involves passing a `ReturnCode` as r0, like so:

```rust
cb.schedule(retcode.into(), 0, 0);
```

This still compiles today, as we are still supporting `ReturnCode` after 2.0. However, we now also support `ErrorCode`, and are encouraging movement towards it -- I could easily see people switching some calls to `schedule()` to pass an `ErrorCode` in the same way -- `into()` still works -- and not realize that userspace will be receiving a different type to decode. 

Similarly, it is now less clear how apps should indicate success in a callback, given that we are discouraging new uses of `ReturnCode` going forward. I propose that it might make sense for `schedule()` to take a `CommandResult` rather than three `usize` values, so that it is still straightforward to use callbacks as a mechanism for returning the results of a command that is executed late because of virtualization.

phil-levis, 2020-12-18:
> Similarly, it is now less clear how apps should indicate success in a callback, given that we are discouraging new uses of `ReturnCode` going forward. I propose that it might make sense for [...]

Seems like a good thing to talk about on the call tomorrow.

alexandruradovici, 2020-12-22:
I added the microbit v2 board into `master` and the PR modifies the `lsm303xx` driver. What would be the steps required to update the `tock-2.0-dev` branch. Is it a rebase from master or just hand picking new files?

hudson-ayers, 2020-12-22:
We have been merging `master` into `tock-2.0-dev` every so often (rather than using rebase, though that is our normal practice in Tock). If there are conflicts that you think might be difficult for someone unfamiliar with the driver to resolve, feel free to submit a PR merging `master` into `tock-2.0-dev` -- it was last merged in only a few days ago so hopefully there won't be many other conflicts.

lschuermann, 2020-12-22:
@alexandruradovici I have just merged master, as I'm also relying on some hardware introduced into master recently for testing with `tock-2.0-dev`. In general, when merging, I try to make sure that the new patchset compiles and passes CI. For Tock 2.0 we also change the Driver trait to LegacyDriver for newly introduced capsules and change the platform-trait of new boards accordingly. Looking into `lsm303xx` I don't see any obvious merge-mishaps, but feel free to test :smiley:.

alexandruradovici, 2020-12-23:
I see that `adc` is still unclaimed, I'll take it, I just need some help testing the high speed part as I do not have the hardware.

hudson-ayers, 2020-12-23:
Thanks! I'm happy to help testing high speed

phil-levis, 2020-12-24:
I'm not quite sure what to do with `spi_peripheral`; using it on imix, for example, requires a new board, because you have to change from a mux'ed SPI controller to a singleton SPI peripheral. This means, among other things, cutting out the RF233 and IPv6. I'm wondering if perhaps I should create a very simple imix_peripheral board, which has just the bare bones (GPIO, UART, Alarm, and SPI peripheral)? This would give a template for implementing an imix-based SPI peripheral as well as give us a testing example.

hudson-ayers, 2020-12-24:
Creating such a board for testing would be fine for the 2.0 update.

I'm not sure we want to clutter the upstream repo with it though, as it would be yet another board to update every time we change the board/kernel interface or board/chip interface, and there are already a number of tests that require changes to the board to execute.

Alternatively, there are multiple SPI peripherals on the nrf52840, which can be mapped to arbitrary GPIO pins. So we could just modify the nrf52840dk to have one sets of pins be SPI controller, and another set be spi peripheral, and both enabled by default...unfortunately noone has written a Tock driver for SPI peripheral mode for the nrf52 yet :/

hudson-ayers, 2020-12-24:
FWIW, I do think @mmurray22 has a working SPI peripheral Imix board implementation somewhere

phil-levis, 2020-12-29:
> Alternatively, there are multiple SPI peripherals on the nrf52840, which can be mapped to arbitrary GPIO pins. So we could just modify the nrf52840dk to have one sets of pins be SPI controller, and [...]

This is a good idea. The SAM4L USART can be configured to be a SPI master; I think I'll try having this drive the SPI peripheral. I'd kinda rather work on this and fix it than toss it away.

phil-levis, 2020-12-29:
@bradjc -- Do you have hardware for the  ltc294x? It looks like the initial commit is yours.

hudson-ayers, 2021-01-01:
There are 4 remaining unclaimed drivers: 

1. `ltc294x`, `pac9544a`, and `max17205` -- all of which were sensors used on signpost, but not used by any upstream boards today. It seems likely that no-one on the core team has hardware. I am happy to port these to 2.0, as they are mostly straightforward, without testing. Is that okay, or should we archive them instead?

2. `hd44780`: added by @teonaseverin and @alexandruradovici , probably too complex to port without testing. Is either of you able to port this, or to test a port authored by someone else?

alexandruradovici, 2021-01-01:
>     2. `hd44780`: added by @teonaseverin and @alexandruradovici , probably too complex to port without testing. Is either of you able to port this, or to test a port authored by someone else?
Submitted #2321, @teonaseverin still has to test it.

alexandruradovici, 2021-01-18:
#2365  deletes the `ft60x6` and `st77xx` drivers as they should not be directly used

phil-levis, 2021-01-27:
I'm cutting `i2c_master_slave_driver`. It looks like only the SAM4L supports slave mode. But because slave mode is rarely if ever used, it currently doesn't seem operational or usable. First off, it's easy to trigger panics. E.g. some I2C modules only support master mode, but still implement slave mode; if you try to use slave mode on them, they panic. If you try to use slave mode on a module that does support it, (e.g., I2C1), then the kernel panics. So it looks like slave support in the SAM4L no longer works correctly. Since this capsule only works on that chip, we should archive it until there's a desire to use it again.

hudson-ayers, 2021-01-27:
> I'm cutting `i2c_master_slave_driver`. It looks like only the SAM4L supports slave mode. But because slave mode is rarely if ever used, it currently doesn't seem operational or usable. First off, [...]

Is the kernel panic fixed by https://github.com/tock/tock/pull/2380 ? Even if it is, I think its fine to archive the driver if nothing uses it.

phil-levis, 2021-02-14:
OK, looks like all that remains is IPC!

closed 2021-02-17 by phil-levis