tock/tock #2246: Flash Controller Failed tests on OpenTitan

KaisserbenDll, 2020-12-08:
The initial support for the OpenTitan Flash Controller, as documented here: https://docs.opentitan.org/hw/ip/flash_ctrl/doc/index.html is failing. 

I used the non volatile storage capsule provided by Tock (latest version in master). I tried to run  the libtock-c application in ` examples/tests/nonvolatile_storage`  and i expected the following output `All tests succeeded`. Instead, the test fails outputting: 
Inconsistency between data written and read at index 1. 
 
## Used Components: 

1) I used the master branch of the repository of lowrisc/opentitan (commit: 23b882050e91bd9ef392bde871d2cdd64db30646)
 to build the opentitan bitstream to try the non_volatile_storage capsule of Tock.  Note that the latest version of Opentitan has a changed  address of the Flash Controller. I receive the same failed test. 

2) I tested it also with the bitstream provided in the Makefile tock offers in the ci_setup_qemu_opentitan target (https://github.com/tock/tock/blob/master/Makefile#L517) 

@towoe 
## Log
```
I00438 boot_rom.c:61] Boot ROM initialisation has completed, jump into flash!
OpenTitan initialisation complete. Entering main loop
[Nonvolatile Storage] Test App
                              Have 32768 bytes of nonvolatile storage
                                                                     Test with size 256 ...
                Finished write! 14
                                        Finished read! 14
                                                                Inconsistency between data written and read at index 1
                                      Failed with code -1
```
## Meta
Hardware platform:  Earlgrey on Nexys Video

alistair23, 2020-12-08:
Thanks for raising this.

First, I have seen some issues using the OpenTitan Flash controller with the latest bitstreams. It seems like there are a few regression with the latest bitstreams.
That is mostly related to interrupts not occurring though. I don't think I have seen data loss with the latest master commit of Tock. I did push a fix to Tock recently that fixes some issues with the newer bitstreams.
From my understanding the failing case would need to be replicated with the lowRISC embedded software in order to have them investigated or fixed. From memory that isn't interrupt driven though, so replicating bugs will be difficult.
I am pretty sure I ran these tests when I wrote the flash controller, so I am guessing this is a hardware bug, in which case you might want to look at replicating it with the OpenTitan baremetal software.

It could also be a Tock bug though ;)
Can you provide more logs? It would be nice to see the actual data being written and read from flash. Is the data not being written to flash or are the reads not working correctly?

KaisserbenDll, 2020-12-08:
I tried to print both write_buffer and the read_buffer from userspace using printf for the first test  
`  if ((r = test(readbuf, writebuf, 256, 0,  14)) != 0) return r; ` 
And apparently the read operation fails. But one cannot be sure that only the read operation fails. This is the output 
`
OpenTitan initialisation complete. Entering main loop
[Nonvolatile Storage] Test App
Have 32768 bytes of nonvolatile storage
Test with size 256 ...
Finished write! 14
Finished read! 14
readbuff[ 0 ] = 0 , writebuff[ 0 ] = 0
readbuff[ 1 ] = 0 , writebuff[ 1 ] = 1
readbuff[ 2 ] = 0 , writebuff[ 2 ] = 2
readbuff[ 3 ] = 0 , writebuff[ 3 ] = 3
readbuff[ 4 ] = 0 , writebuff[ 4 ] = 4
readbuff[ 5 ] = 0 , writebuff[ 5 ] = 5
readbuff[ 6 ] = 0 , writebuff[ 6 ] = 6
readbuff[ 7 ] = 0 , writebuff[ 7 ] = 7
readbuff[ 8 ] = 0 , writebuff[ 8 ] = 8
readbuff[ 9 ] = 0 , writebuff[ 9 ] = 9
readbuff[ 10 ] = 0 , writebuff[ 10 ] = a
readbuff[ 11 ] = 0 , writebuff[ 11 ] = b
readbuff[ 12 ] = 0 , writebuff[ 12 ] = c
readbuff[ 13 ] = 0 , writebuff[ 13 ] = d
`
Also I am curious of why the address accesible by userspace is 0x20000000. According to this link: https://github.com/lowRISC/opentitan/blob/master/hw/top_earlgrey/sw/autogen/top_earlgrey_memory.h, that is the actual address of eFlash in Opentitan. Nevertheless, that is where _start function and other usefull functions are stored. So why exactly does the userspace store data in that region ?

alistair23, 2020-12-08:
You will need to print them in the kernel, just before reading/writing to flash to check if it's an issue with the driver/hardware.

Good point, that is wrong and should be fixed.

KaisserbenDll, 2020-12-08:
So this is another log, where i am printing the read_buffer and writer_buffer from the kernel in /chips/lowriscv/src/flash_ctrl.rs, specifically in the handle_interrupt and following each interrupt, where OP_DONE is where the read and write are occuring 
```bash
INTR::OP_DONE  
CONTROL::OP::READ
Read Buffer 16 
Read Buffer 0  
Read Buffer 0  
Read Buffer 32 
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
INTR::RD_LVL   
INTR::OP_DONE  
CONTROL::OP::PROG
Write Buffer 0 
Write Buffer 1 
Write Buffer 2 
Write Buffer 3 
Write Buffer 4 
Write Buffer 5 
Write Buffer 6 
Write Buffer 7 
Write Buffer 8 
Write Buffer 9 
Write Buffer 10
Write Buffer 11
Write Buffer 12
Write Buffer 13
Finished write! 14
INTR::RD_LVL
INTR::RD_LVL   
INTR::OP_DONE  
CONTROL::OP::READ
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
Read Buffer 0  
```

In order to write, there must be always a read first, if not page aligned. I also checked the addresses where is read from and written to. And those are exactly the same values 0x20000000 (later given to the driver's register).

alistair23, 2020-12-08:
Great! That is super helpful.

Reading those bufs I have a pretty good idea what is wrong. You haven't performed an erase on the flash, you should be getting 0xFF for every byte.

KaisserbenDll, 2020-12-08:
Can you elaborate more on that  please?

alistair23, 2020-12-08:
Which part?

I'm assuming the erase part. All flash needs to be erased before it can be written. The erase function will set all bits to `1`, which means all bytes will be `0xFF`. A flash write can only convert a `1` to a `0` so writing a byte to `0x00` will have no effect.

At one point in the past the OpenTitan flash write would first erase the region first. This is probably why it worked for me as all writes were guaranteed to have already erased the region. This behaviour didn't follow Tock's async model and didn't match the HIL so I fixed it. Now you need to call erase on a region before calling write.

KaisserbenDll, 2020-12-08:
So I tried calling earse_page in the nonvolatile_to_pages capsule before doing a write. First in the `write` function if the write is page aligned and in the read_complete function before calling write_page. This raises the exception `Environment call from U-mode`.

hudson-ayers, 2020-12-08:
I think one issue here is ambiguity regarding what the Tock Flash HIL means. The sam4l implementation of `hil::Flash::write_page()` calls `flashcalw_erase_page()` internally, erasing the target page whenever `write_page()` is called. The `nonvolatile_to_pages` capsule seems to be written under the assumption that chip drivers will internally handle erasing the page before writing.

The OpenTitan implementation interprets the requirements of `hil::Flash::write_page()` differently, expecting that higher layers in the stack will call `erase_page()` before a write.

alistair23, 2020-12-08:
@KaisserbenDll Yeah, the non_volatile_storage_driver doesn't seem to support erasing pages. I would say in that case that it doesn't work and you won't be able to use it. Your best bet is to add support for letting an app call erase_page. Note that it is async, so you will need to wait for the callback before you start the next operation.

@hudson-ayers That seems like a bug right? That breaks Tocks async model as now the erase operation has to be called synchronously. It also has a large speed disadvantage as flash can usually be written at very small granularities (sometimes u8/32) while erased at large granularities. Even without the granularities, erasing a page on every write is slow and not always required.

hudson-ayers, 2020-12-08:
The sam4l implementation handles the erasing asynchronously, it has an internal state machine that receives interrupts from the flash controller and then dispatches additional commands to the flash controller until the write_unlock -> erase -> write sequence is complete, at which point it delivers the `write_complete()` callback to the client that originally called `write_page()`

alistair23, 2020-12-08:
That's pretty cool!

I just checked, the nRF is not async and spin waits on the erase.

The STM doesn't call erase from the write, the same as OT.

I'll open a PR to update the documentation and we can continue the discussion there.

alistair23, 2020-12-09:
Ok, I added that and some other changes in: https://github.com/tock/tock/pull/2248

KaisserbenDll, 2020-12-19:
@alistair23  Every time I am trying to add support to the erase_page functionality, I get hit with an OP::ERROR from the flash controller. I even tried to point to a region that was initialized with 0xFF. The write operation always fails. 
Any ideas on why is that ?

alistair23, 2020-12-24:
Strange. I don't have an idea on the top of my head. I think I mentioned earlier that I have been seeing bugs in the Flash Controller. It might be best to raise an issue with the OpenTitan people and see if they can help.

alistair23, 2020-12-24:
I just opened https://github.com/tock/tock/pull/2309 which I had already queued up. I'll be back at work in about a week and a half and I can look into it then.

I just checked my testing repo (https://github.com/alistair23/tock/tree/alistair/tickfs-integration.wip) and I am using flash address 0x20040000 which might help.

closed 2021-10-15 by bradjc