tock/tock #2254: Console Operator Capsule

kevtan, 2020-12-10:
Hello everybody! This issue is a RFC for the design of a "console operator" capsule that will implement UART multiplexing in order to avoid issues like #1336 and make interacting with boards over serial communication programs like `minicom` or `tockloader listen` a more polished user experience. If you look at the issue, you'll see that both the number guessing game and the process console are receiving the same UART input, responding to it, and echoing whatever they receive (where it says 11, the user typed 1 and where it says 1100, the user typed 10), which makes it difficult to either play the game or manage your processes. Ideally, the user should be able to establish isolated communication channels with a specific capsule or application. I'd like to introduce a `ConsoleOperator` capsule to implement this feature. Here is what, declaratively, I'm currently imagining the user interface to be:

```
console_operator % help
I'm the console operator! Here is an exhaustive list of the capsules
and applications I can help you set up a UART connection with:
  0 [C] process_console
  1 [C] log_console
  2 [A] number_guessing_game
  3 [A] console
Type in the number or name of one of these capsules or applications
to establish a dedicated communication channel. Use Ctrl-D to
terminate the connection and return to the console operator.
console_operator % 0
process_console % help
Welcome to the process console.
Valid commands are: help status list stop start fault
process_console % status
Total processes: 2
Active processes: 2
Timeslice expirations: 0
process_console % ^D
console_operator % 2
number_guessing_game %
Welcome to the number guessing game!

OK I have selected a random number between 0 and 65535
Enter your guesses, and I will let you know when you are correct!
Enter a number: 
number_guessing_game % 50000
Not quite. Please choose a lower number.
Enter a number: 
number_guessing_game % 20000
Nope. You are too low
Enter a number: 
number_guessing_game % 29000
Not quite. Please choose a lower number.
Enter a number: 
number_guessing_game % 28500
Nope. You are too low
Enter a number: 
number_guessing_game % 28950
Not quite. Please choose a lower number.
Enter a number: 
number_guessing_game % 28925
Hooray! You guessed my number.
You are very good at this.
Thank you for playing the number guessing game.^D
console_operator %
```

Even from the declarative end-goal view, there are a lot of things I'd like feedback on. For instance, are the command prompts (to make it seem like a shell) useful? If so, when should they be printed? Every line seems useful for the process console but screws up the interaction with the guessing game. Also, should the output of capsules and applications be buffered? In the example above, the output of the number guessing game was not shown until the user switched to that communication channel. I don't know much about it, but it seems to me like buffering the output would be complicated and potentially not resource-friendly.

And what about buffering inputs? Should we only allow the user to send input to an application that is in blocked on a `getch()`, `getnstr`, or similar function? If so, then the number guessing game would be quite hard to play as it would be impossible to maintain a constant connection with it (the program isn't continually blocked on a UART reception operation). If not, then is the overhead incurred by buffering a necessary undertaking?

jrvanwhy, 2020-12-10:
Another option would be to have a packetized interface. Each communication going across the UART would contain the capsule or application that sent or should receive it. This would be better suited to programmatic access rather than direct use by hand. If we were to give it a human-readable interface then it might look something like this:

```
[console_operator] I'm the console operator! Here is an exhaustive list of the capsules
[console_operator] and applications I can help you communicate with:
[console_operator]  0 [C] process_console 
[console_operator]  1 [C] log_console
[console_operator]  2 [A] number_guessing_game
[console_operator]  3 [A] console
[console_operator] Type a name or number followed by a colon, space, and your
[console_operator] message to send a message to a capsule or application.
process_console: help
[process_console] Welcome to the process console.
[process_console] Valid commands are: help status list stop start fault
process_console: status
[process_console] Total processes: 2
[process_console] Active processes: 2
[process_console] Timeslice expirations: 0
number_guessing_game: start
[number_guessing_game] Welcome to the number guessing game!
[number_guessing_game] OK I have selected a random number between 0 and 65535
[number_guessing_game] Enter your guesses, and I will let you know when you are correct!
[number_guessing_game] Please enter your first guess.
number_guessing_game: 50000
[number_guessing_game] Not quite. Please choose a lower number.
number_guessing_game: 25000
[number_guessing_game] Hooray! You guessed my number.
[number_guessing_game] You are very good at this.
[number_guessing_game] (almost like you are tired of typing this GitHub comment...)
```

This would require applications to be ready to receive messages at any time. That seems very reasonable for `libtock-rs` -- is it difficult for `libtock-c` applications?

kevtan, 2020-12-10:
@jrvanwhy Thanks for your feedback! I agree that the stateless packetized interface would probably be more amenable to programmatic interaction. I've added it to my original comment. I'm not sure what you mean when you say "seems very reasonable for `libtock-rs`." I'm looking through the repository and I don't see a way you can actually listen to UART input? None of the examples seem to demonstrate this behavior.

jrvanwhy, 2020-12-10:
> I'm not sure what you mean when you say "seems very reasonable for `libtock-rs`." I'm looking through the repository and I don't see a way you can actually listen to UART input? None of the examples seem to demonstrate this behavior.

That's because I haven't written it yet :-). I'm in the middle of rewriting `libtock-rs` right now. The reason I say it seems reasonable is that `libtock-rs` will have a highly-asynchronous design, so listening for an event continuously in the background will be a reasonable requirement.

bradjc, 2020-12-17:
> * Should the output of capsules and applications be buffered? In the example above, the output of the number guessing game was not shown until the user switched to that communication channel. I [...]

If the kernel is responsible for keeping track of which console is active, then no. There is no way for the kernel to allocate memory for this, and so it would end up dropping messages anyway.

In the version of this where the kernel only tags messages (as @jrvanwhy describes) then I think the capsule just sends everything, and the receive is responsible for buffering/filtering.

> * What about buffering inputs? Should we only allow the user to send input to an application that is blocked on a `getch()`, `getnstr`, or similar function? If so, then the number guessing game would be quite hard to play as it would be impossible to maintain a constant connection with it (the program isn't waiting on a `getch()`, `getnstr`, or similar function at all times). If not, then is the overhead incurred by buffering a necessary evil?

The kernel wouldn't be able to implement this anyway. If the capsule or application is not actively listening for UART then the kernel will have to drop the incoming message.

> * As @jrvanwhy mentioned below, would it be useful to also have a stateless packetized interface that is more amenable to programmatic access? Or should this be the only interface?

I don't know. The issue with the packetized approach is the host-side tool needs to know about it, and now it is no longer a simple serial terminal which is very ubiquitous and easy to use. The ascii operator version provides a lot of the benefits but is also easy for users to just use manually. It's not clear to me exactly how this should look.

bradjc, 2021-04-26:
Moving discussion to tracking issue #1336

closed 2021-04-26 by bradjc