tock/tock #2260: STM32F3DISCOVERY debug! Macro and Process Console Don't Work

kevtan, 2020-12-11:
Was wondering if anyone else was having issues with the UART1 connection for the STM32F3DISCOVERY board? For me, I am unable to get any sort of debugging output from the board. At first, I thought it had something to do with the SB10 solder bridge as described [here](https://docs.rust-embedded.org/discovery/06-hello-world/index.html) but even after shorting SWO and PB3, I'm still unable to establish a working serial connection.

alexandruradovici, 2020-12-12:
There is a bug that is preventing stm32f3discovery from working, most probably due to this https://github.com/tock/tock/pull/2188. Increasing the stack size did not help for this boards.

kevtan, 2020-12-13:
That PR does not seem the `stm32f303xc` or `stm32f3discovery` modules so I don't see how it could affect the operation for the STM32F3DISCOVERY board unless I'm missing something—did you mean to reference a different PR?

krady21, 2020-12-14:
It was probably this one #2187.

hudson-ayers, 2020-12-14:
Yeah, that was my PR, but I don't have stm32f3 hardware to test what broke, so any debugging help is appreciated.

hudson-ayers, 2022-07-22:
This board passed the release testing for Tock 2.0, so this was fixed at some point (probably byhttps://github.com/tock/tock/pull/2784)

closed 2022-07-22 by hudson-ayers