tock/tock #2270: subscribe: validate callback address boundaries?

lschuermann, 2020-12-14:
While developing the swap constraints of the `subscribe` operation for the Tock 2.0 system call interface, I noticed that a subscribe operation does not validate the address boundaries of the callback in any way.

## Example

A simple dummy application
```c
int main(void) {
  putnstr_async(hello, strlen(hello), (subscribe_cb *) 0x00000004, NULL);
  return 0;
}
```

will actually cause the CPU to actually jump to the specified address (`0x00000004`) in usermode, which in turn panics the application as this is catched by the MPU accordingly:

```
---| Fault Status |---
Instruction Access Violation:       true
Forced Hard Fault:                  true
Fault Status Register (CFSR):       0x00000001
Hard Fault Status Register (HFSR):  0x40000000

---| App Status |---
App: c_hello   -   [StoppedFaulted]
 Events Queued: 0   Syscall Count: 7   Dropped Callback Count: 0
 Restart Count: 5
 Last Syscall: Some(YIELD)

  [...]

  R0 : 0x0000000E    R6 : 0x0003030C
  R1 : 0x00000000    R7 : 0x20004000
  R2 : 0x00000000    R8 : 0x00000000
  R3 : 0x00000000    R10: 0x00000000
  R4 : 0x0003002C    R11: 0x00000000
  R5 : 0x20004000    R12: 0x00000000
  R9 : 0x20004800 (Static Base Register)
  SP : 0x200047D0 (Process Stack Pointer)
  LR : 0x00030141
  PC : 0x00000004
 YPC : 0x00030140

```

## Safety

For chips that have proper memory protection support this is not a safety/security issue, but much rather a question of whether the kernel should catch such invalid values beforehand. I have a strong feeling that returning something like `EINVAL` would be much more appropriate than simply jumping to the specified address, no questions asked. It could be beneficial for a userspace application to be able to handle these errors gracefully, instead of just crashing as soon as the callback is scheduled.

## Outlook for Tock 2.0

Tock 2.0 requires some handling in order to prevent swapping of Callback structs anyways. My question is whether the kernel should sanity-check the values received from userspace and only accept (and invoke the `subscribe` methods) if the callbacks are located in process flash or RAM, or whether we should continue to blindly accept any value passed from userspace here.

If we want to handle this, I can simply add checks as to whether the passed address is in process accessible memory as part of the Tock 2.0 callback swap constraint checks.

alistair23, 2020-12-14:
Just from a debugging perspective I think a sanity check would be nice. With RISC-V especially it's not uncommon that mis-aligned addresses (although it's not as bad as it used to be) cause jumps to strange addresses that are hard to debug. Printing and error instead would be very useful.

jrvanwhy, 2020-12-14:
I'm not sure this check is very valuable, because I can't imagine a mistake that would result in a correctly-linked application passing an invalid function pointer to the kernel.

> Just from a debugging perspective I think a sanity check would be nice. With RISC-V especially it's not uncommon that mis-aligned addresses (although it's not as bad as it used to be) cause jumps to strange addresses that are hard to debug. Printing and error instead would be very useful.

How do you end up with such a mis-aligned function pointer? Bad linker script?

lschuermann, 2020-12-14:
I was just thinking of some pointer-arithmetic stunts in C, but your point is indeed valid: userspace should always pass a valid pointer, if it doesn't cast something bad explicitly. If there's memory protection on the board it should catch it, if not the application could jump to the memory anyways. This check isn't very expensive though, so I'm fine either way.

alistair23, 2020-12-14:
Yes, for RISC-V the linker script is hard to get right as the addresses need to match. Brad has done great work adding checks to the kernel, so this is mostly avoided, but it can still happen.

I would say this probably should be a config option in the kernel so that it can be turned off to squeeze out more performance, but a quick sanity check would be useful.

hudson-ayers, 2020-12-14:
Personally I don't think that we should include a check that is only valuable on targets without an MPU -- Tock's design explicitly states that hardware is assumed to provide an MPU (https://github.com/tock/tock/blob/master/doc/Overview.md#tock-overview).

jrvanwhy, 2020-12-14:
> Yes, for RISC-V the linker script is hard to get right as the addresses need to match. Brad has done great work adding checks to the kernel, so this is mostly avoided, but it can still happen.

The ARM entry point in `libtock-rs` checks that the code is at the correct location. IIRC the RISC-V entry point does not do so. I'm planning to write a RISC-V entry point for libtock_platform this week, and I was planning to include the address checks.

Would that resolve your concerns? That would avoid a check on every `subscribe` call -- I'm hoping `subscribe` is small enough to be consistently inlined.

alistair23, 2020-12-15:
Yeah, that's fine.

I don't think it's enough of a problem that I'm that fussed. I was just pointing out that it might be useful to verify addresses.

phil-levis, 2021-02-15:
Yeah, the idea is that if a process provides an invalid function pointer, it will just fault, in the same way that you can pass an invalid function pointer to `sigaction(2)`.

bradjc, 2021-03-03:
I think the conclusion is "no". For normal operation passing in a bad upcall address to subscribe is a hard mistake to make, and for improperly linked/aligned processes adding a check on every subscribe call is not the right place to catch the issue.

closed 2021-03-03 by bradjc