tock/tock #2320: Tock 2.0: callbacks, return code, and errorcode

hudson-ayers, 2021-01-01:
Tock 2.0 changes system calls such that success vs. failure is now passed in a separate register from the register used to communicate error codes in the case of a failure. This works fine because we increased the number of registers used for system calls, and meshes well with our desire to eventually remove `ReturnCode` from the Tock kernel in favor of `Result<T, ErrorCode>`.

However, callbacks still use `ReturnCode` to communicate errorcodes + the possibility of success in a single register. And the number of registers used to communicate values in callbacks has not changed for Tock 2.0, so using the same approach we do for `Command` is not feasible -- several callbacks exist that already use all of the available registers. Callbacks are generally a contract between a capsule and a corresponding userspace driver, so it is okay for callbacks to still use ReturnCode or some similar format for communicating errors. But I anticipate that there will be some confusion as some capsules start to report ErrorCode's to communicate errors while others still use ReturnCode's -- especially because the values used to represent the same errors do not match.

After some discussion with @lschuermann , I believe the best solution would be to change `ReturnCode` to represent errors using the same underlying values as `ErrorCode` does, after changing `ErrorCode` to not represent any errors using `0`. This would mean `ReturnCode` and `ErrorCode` would be identical types, except that `ReturnCode` would also have a `Success: 0` variant. Even once `ReturnCode` is deprecated, we could have a function like the following:

```rust
impl From<Option<ErrorCode>> for u32 {
    ...
}
```
so that a `Result<T, ErrorCode>` can easily be communicated as success/failure in a callback via:

```rust
self.callback.schedule(e.ok().into(), something as u32, something as u32)
```

Unfortunately, this would require changing a substantial amount of error handling code in `libtock-c` apps which relies on the fact that, in callbacks, errors are negative and success is positive.

Technically, there is no reason we have to make this change as part of 2.0 -- as the types of values passed in callbacks are not part of the ABI contract. Is this a change we should try to make before delviering 2.0, or schedule for sometime afterwards (perhaps whenever `ReturnCode` is actually deprecated in the kernel?)

phil-levis, 2021-01-06:
Is this a libtock-rs problem? I expect that libtock-c will use ReturnCode, because C has poor mechanisms for more complex types. The standard POSIX semantics is that < 0 is an error, 0 or >0 is success. So I'd suggest that ReturnCode is the negative of ErrorCode (sure, remove 0). This is kind-of how POSIX does it with `errno`; our `ErrorCode` is like `errno` while our `ReturnCode` is like return values.

Or am I missing something?

hudson-ayers, 2021-01-06:
That seems to suggest keeping `ReturnCode` around in the kernel forever, just for the use of passing errors in callbacks. I thought our goal was to completely remove `ReturnCode` in favor of `Result<T, ErrorCode>` throughout the kernel. 

Generally it seems silly to have two types in the kernel that represent the same errors using different underlying values. It also seems odd to sometimes communicate errors across the ABI using positive values (as we do in the new system call API) and other times using negative values (as we are doing in callbacks)

phil-levis, 2021-01-06:
Looking at the callbacks that use all three arguments, I think we can transition them away from it. For example, it's 3-axis devices (which do not need 32 bits for each axis), the SDCard, which wants to pass both a callback identifier as well as two values. This seems like something to talk about on Friday?

bradjc, 2021-01-08:
I agree error numbers should be the same in userspace and in the kernel (i.e. there my be multiple lists/definitions, but the same error type has the same number). I think this will make debugging easier, and is worth changing userspace drivers for.

And having a convention where capsules use `Option<ErrorCode>` in their callbacks seems like an intuitive approach that will work for most/all capsules.

hudson-ayers, 2021-01-22:
Update: I have implemented the kernel changes required for this, based on the state of the discussion on the January 8 core wg call. That was simple.

The challenging part is that now every callback in libtock-c which gets an error back from the kernel needs to negate that error before passing it up to higher levels (as much of libtock-c uses `err < 0` to check if an operation succeeded). There are about 125 callbacks in libtock-c, based on a very quick grep. 

Most of the changes will look something like this:
```c
static void text_screen_callback(
                                 unsigned int error, // previously this was just `int error`
                                 int data1,
                                 int data2,
                                 void* ud) {
  TextScreenReturn *ret = (TextScreenReturn*) ud;
  ret->error = error * -1; // * -1 is new
  ret->data1 = data1;
  ret->data2 = data2;
  ret->done  = true;
}
```

Some confirmation that this is definitely what we want would be nice before I start working my way through all 125 callbacks.

bradjc, 2021-01-28:
Can we change `<` to `!=` in the check? That is almost certainly a harder to change to make in terms of finding all of the uses, but if we are going to do this then I think we should do it all the way.

hudson-ayers, 2021-01-28:
Unfortunately there are still lots of library functions that pass positive values for success (when a value is passed along with that success in a different register) and negative for error. So it would be hard to determine where it is safe to use !=

bradjc, 2021-01-28:
It does seem in the 2.0 spirit to fix that as well.

hudson-ayers, 2021-01-28:
Yes, but the inability to create anything like `Result` in C complicates things relative to the matching fix in libtock-rs. Positive values for success and negative for error is pretty idiomatic C code. A different approach would require returning some manually constructed Result struct, or having functions take a pointer that is manipulated to communicate errors.

lschuermann, 2021-01-29:
> ```c
> ret->error = error * -1; // * -1 is new
> ```

While technically we're negating the value, it only works by coincidence (choosing the right values for `ErrorCode`) I suspect we're going to use the already existing `tock_error_to_rcode`, right? I also side with @bradjc in that we should, wherever easily possible, we should try to avoid storing values > 0 in the return code.

hudson-ayers, 2021-01-29:
> While technically we're negating the value, it only works by coincidence (choosing the right values for `ErrorCode`) I suspect we're going to use the already existing `tock_error_to_rcode`, right? 

yes, I think using `tock_error_to_rcode` would be better there.

>I also side with @bradjc in that we should, wherever easily possible, we should try to avoid storing values > 0 in the return code.

I agree this is the nicer solution, but I think that verifying this will require manually reviewing every single library function in `libtock-c`, *and* changing a good few of them, *and* updating every use of each changed function to match the new interface. C's type system does not catch refactoring errors nearly as effectively as Rust's, so I am a little wary of that..

bradjc, 2021-01-29:
Does libtock-c actually return that many positive success values? Probably everywhere we used successwithvalue (e.g. `led_count()`)? I think we could find those at least fairly easily.

Perhaps the way to answer this is to think of what we would build starting at 2.0 if libtock-c did not currently exist. I agree with @hudson-ayers that return errors in C are negative. A C interface user likely expects that errors will be negative values.

I do think we can say, however, that all tock C library functions return error values. So even in the case of `led_count()`, the number should be returned via a pointer argument, and not as a positive "error" value. I think C convention is not as clear on this? For things like `num_leds()` we could certainly get away with combining them (how many leds can a board have), but it would be more consistent with 2.0 to say they have to be separated, and that is just a Tock convention.

So I can support doing the `*-1` trick (perhaps in some wrapper function), but also changing functions like `led_count()` to only return errors and to set the count with a pointer.

hudson-ayers, 2021-01-29:
> Does libtock-c actually return that many positive success values? Probably everywhere we used successwithvalue (e.g. `led_count()`)? I think we could find those at least fairly easily.

Yeah, I suppose everywhere that we now return `success_u32()` in the kernel. Maybe a couple more places as well, where system calls are issued that could error, and if they don't some value stored in the library is returned.

> Perhaps the way to answer this is to think of what we would build starting at 2.0 if libtock-c did not currently exist. I agree with @hudson-ayers that return errors in C are negative. A C interface user likely expects that errors will be negative values.

Yeah I think we should change tock_error_to_rcode to just use the *-1 trick as part of this.

> I do think we can say, however, that all tock C library functions return error values. So even in the case of `led_count()`, the number should be returned via a pointer argument, and not as a [...]

Okay, I guess I am on board with that. It will mean a lot of changes to libtock-c -- but I don't technically think 2.0 has to block on any of this. We can change the values passed in callbacks as part of a minor release, yes?

bradjc, 2021-01-29:
> Okay, I guess I am on board with that. It will mean a lot of changes to libtock-c -- but I don't technically think 2.0 has to block on any of this. We can change the values passed in callbacks as part of a minor release, yes?

Why does this require changes to capsules? I was thinking all changes would be in libtock-c.

We can change the capsule specific values in any syscall at any time except for ones that we stabilize.

hudson-ayers, 2021-01-29:
right now capsules still return negative values when passing errors (every capsule callback in Tock still uses `ReturnCode`, which has the old negative values). we want to change that at-2.0 or sometime soon post-2.0 because of the reasons at the top of this thread, but can't do so without breaking most libtock-c apps.

phil-levis, 2021-02-15:
I think we should do this after 2.0 is merged; it involves specifying the values passed in upcalls, which it outside the scope of the syscall TRD. We should write a separate upcall TRD that specifies how this works.

I think we do want to keep upcalls as flexible as possible, as we did with commands. So one approach would be to just follow the command return value approach and specify how errors are passed. The other would be to just let capsules use ReturnCode if they want to. There's nothing stopping a capsule from defining its own error type and values. But I think this starts to get messy, because standardized tools and output analysis will become hard. The amount of time I spent searching nRF softdevice code to understand what 0x4001 meant as an error makes me shy away from this approach. Of course capsules can provide richer error data than what `ErrorCode` does, but I'd argue it should supplement, rather than replace.

> Yeah I think we should change tock_error_to_rcode to just use the *-1 trick as part of this.

This is exactly what Linux does, for example. It's important that conceptually code doesn't assume this happens (it should work fine if there is a randomized remapping between error codes and rcodes), but we do it this way for the efficiency of `tock_error_to_rcode`. 

Does this mean that within the kernel we want to define a type that is

```rust
Result<(), ErrorCode>
```

which has a `From` for `u32`  such that `Ok(())` is 0 and `Err<ErrorCode>` to the value of `ErrorCode`? We could then specify that the first argument of an upcall is this type and leave the others as `u32`. 

So a follow-up TRD could specify that upcalls look like this:

```rust
type UpcallResult = Result<(), ErrorCode>;

fn schedule(&mut self, r0: UpCallResult, r1: usize, r2: usize) -> bool {...}
```

I think that porting all of the capsules to use this new API should not be very difficult. Only the ones that use all 3 arguments will be a little tricky.

hudson-ayers, 2021-02-15:
Towards the goal of making upcalls as flexible as possible, I would prefer we continue to allow arbitrary values in all 3 registers, and just make it a convention that if an error is passed in a register, it will be an `ErrorCode` or 0. It is up to individual capsule authors to get this right, but we make it straightforward using a `From` implementation similar to what you just described, plus the `tock_error_to_rcode` * -1 trick in userspace. If libtock-c users forget to use tock_error_to_rcode, they will get positive errors instead of negative, but that seems like the only downside. The upside is that it will become much more clean in libtock-c when a value is an error.

Once `ReturnCode` is gone from the kernel for good, I suspect this becomes pretty much a non-issue. And I would prefer we design around that future rather than around minimizing errors in the time period before the switch is complete.

bradjc, 2021-02-15:
If a capsule wants to return SUCCESS, what would the code look like? `app.schedule(0, 0, 0)`?

```rust
let success: Option<ErrorCode> = None;
app.schedule(success.ok().into(), 0, 0)
```
?

phil-levis, 2021-02-15:
> Towards the goal of making upcalls as flexible as possible, I would prefer we continue to allow arbitrary values in all 3 registers, and just make it a convention that if an error is passed in a register, it will be an `ErrorCode` or 0.

Then at the very least we should provide some standard support for doing this (e.g., the `UpcallResult` type). But what's the case we're concerned with? I see three possible cases:

    1. The upcall can't have an error (seems unlikely, given almost all HILs have errors in callbacks to support virtualization).
    2. The upcall needs more that 64 bits, so wants to combine some data with error bits. E.g., uses the high 28 bits of `r0` for data.
    3. The upcall wants to pass back richer error information, so would like to use an alternate set of error codes.

Which of these seem important?

hudson-ayers, 2021-02-15:
I think they all seem somewhat important. Today it is possible to have a capsule that allows for registering two types of callbacks -- one for error reporting, and the other for reporting results in the case of success. For something that needs to pass lots of data via callbacks (say, streaming 9DOF values) it might be nice to use all 3 values to pass the results. In the rare case of an error, that can be reported via a separate error callback.

Passing richer error information with an alternate set of error code also seems useful -- for example we already do this with SDCard today. This can be especially useful if the userspace app is a third party library that expects particular error types. If we ever did try to use a userspace networking library, we would probably want an alternate set of error codes to accurately describe different failures without having to spend 2 registers on errors.

phil-levis, 2021-02-16:
> Today it is possible to have a capsule that allows for registering two types of callbacks -- one for error reporting, and the other for reporting results in the case of success.

Sort of -- in that case, you can choose to not register for the error callback, which seems like a problem. In TinyOS, we considered separate error/success callbacks because they were statically bound and so you *had* to implement both or it was a compile-time error. The thought was having a separate error callback would force the programmer to consider it rather than just ignore the error code. But we found that there was typically a lot of repeated code between the two (e.g., putting a buffer back), so having a single callback was generally better.

> For something that needs to pass lots of data via callbacks (say, streaming 9DOF values) it might be nice to use all 3 values to pass the results. 

9DOF does seem to be the one example, but it doesn't need 96 bits. It's pretty rare that sensors give more than 16 bits. But you're right, if I want to grab both linear as well as angular acceleration, it would be nice to fit it in 3 words.

> Passing richer error information with an alternate set of error code also seems useful -- for example we already do this with SDCard today. This can be especially useful if the userspace app is a [...]

Makes sense.

My major worry is we are making the base case more cumbersome in order to better support edge cases. What if had two `schedule` methods, one which takes an `UpcallResult` for `r0`, and one which lets you pass an arbitrary `u32`? That way for the common case capsules can pass an error code/success in `r0` but more complex cases can use `r0` however they want? The first variant is therefore just a slightly more ergonomic approach than manually converting `ErrorCode` to a `u32` and passing 0 to denote success.

hudson-ayers, 2021-02-16:
> Sort of -- in that case, you can choose to not register for the error upcall, which seems like a problem. In TinyOS, we considered separate error/success events because they were statically bound and so you _had_ to implement both or it was a compile-time error. The thought was having a separate error event would force the programmer to consider it rather than just ignore the error code. But we found that there was typically a lot of repeated code between the two (e.g., putting a buffer back), so having a single event handler was generally better.

Makes sense. I would expect this usage pattern to be rare though.

> My major worry is we are making the base case more cumbersome in order to better support edge cases. What if had two `schedule` methods, one which takes an `UpcallResult` for `r0`, and one which [...]

I like the idea of two `schedule` methods, but would doing that negate most of the benefit on the userspace side? I was imagining that if we went the `UpcallResult` route that userspace would automatically handle the error conversion for all callbacks somehow (I haven't thought about how exactly that would look), but you can't do that if there are two `schedule` methods in the kernel. Maybe what I was imagining isn't really possible anyway, I am not familiar with how userspace handles callbacks exactly.

closed 2021-04-23 by bradjc