tock/tock #2326: Kernel not loading on earlgrey

jackieke724, 2021-01-07:
Hello,

I am testing Tock on Earlgrey which I implemented on Genesys 2 instead of Nexys Video. When I used spiflash to load the kernel, I did not see `OpenTitan initialisation complete. Entering main loop`. I suspect the system hanged before the reset_handler. Please check the terminal and uart outputs posted below.

Could anyone please shed some light? Thank you!

**terminal output**
```
$ make flash-app APP=~/libtock-rs/target/riscv32imc-unknown-none-elf/tab/opentitan/hello_world.tab 

info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.49.0-nightly (ffa2e7ae8 2020-10-24)`
    Finished release [optimized + debuginfo] target(s) in 0.11s
   text	   data	    bss	    dec	    hex	filename
  88753	   5972	  10408	 105133	  19aad	/home/junmin/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo
/tools/riscv/bin/riscv32-unknown-elf-objcopy --update-section .apps=/home/junmin/libtock-rs/target/riscv32imc-unknown-none-elf/tab/opentitan/hello_world.tab /home/junmin/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo.elf /home/junmin/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo-app.elf
/tools/riscv/bin/riscv32-unknown-elf-objcopy --output-target=binary /home/junmin/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo-app.elf /home/junmin/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo-app.bin
/home/junmin/genesys2/debug/opentitan//build-out/sw/host/spiflash/spiflash --dev-id="0403:6010" --input=/home/junmin/tock/target/riscv32imc-unknown-none-elf/release/earlgrey-nexysvideo-app.bin
Running SPI flash update.
Image divided into 212 frames.
frame: 0x00000000 to offset: 0x00000000
frame: 0x00000001 to offset: 0x000003d8
...
frame: 0x800000d3 to offset: 0x00032b08
```

**UART output**
```
I00000 boot_rom.c:52] Version:    opentitan-snapshot-20191101-1-3779-g754c60f5
Build Date: 2021-01-06, 16:11:20

I00001 bootstrap.c:153] Bootstrap requested, initialising HW...
I00002 bootstrap.c:177] HW initialisation completed, waiting for SPI input...
I00003 bootstrap.c:101] Processing frame #0, expecting #0
I00004 bootstrap.c:124] Flash erase successful
I00005 bootstrap.c:101] Processing frame #0, expecting #1
I00006 bootstrap.c:101] Processing frame #1, expecting #1
I00007 bootstrap.c:101] Processing frame #0, expecting #2
...
I00424 bootstrap.c:101] Processing frame #210, expecting #210
I00425 bootstrap.c:101] Processing frame #0, expecting #211
I00426 bootstrap.c:101] Processing frame #211, expecting #211
I00427 bootstrap.c:134] Bootstrap: DONE!
I00428 boot_rom.c:61] Boot ROM initialisation has completed, jump into flash!
```

alistair23, 2021-01-07:
You are going to have to connect GDB and see what is happening. Do you know if Tock is even being loaded?

One thing to try first though. Tock currently is using an older OT memory layout, check the OT README in the boards directory to see what commit we are using. There is a PR to update this, which you might need if you are using a new SHA: https://github.com/tock/tock/pull/2289

Note that the above PR doesn't use the latest OT SHA either as there are some regressions there. You should at least see prints though if you are using the latest OT SHA and the above Tock PR.

jackieke724, 2021-01-08:
@alistair23 Thanks for the help! The problem is indeed caused by the memory layout mismatch. I am now using the OT repo @`24ab6c`  and the hello world example works.

However, all other examples (leds, timer...) only print `LowLevelDebug: App 0x0 alert code 0x1`. There are leds, switches and timer available on Genesys 2. Do you have any clue on that?

jrvanwhy, 2021-01-08:
`LowLevelDebug` alert code `0x1` indicates `panic!` was called ([alert code table](https://github.com/tock/tock/blob/master/doc/syscalls/00008_low_level_debug.md#predefined-alert-codes)). I don't have time to help debug the source of the panic further, sorry.

alistair23, 2021-01-08:
The devices might not be wired up in Tock, which will return an error to the app.

Maybe try setting these to `true` and see if the logs help: https://github.com/tock/tock/blob/master/kernel/src/config.rs#L48

jackieke724, 2021-01-10:
@alistair23 Could you elaborate a bit more on "not wired up"?

Here is the output of additional logs. The panic seems to happen before the app code. For some other tests (i.e. led, button), the 0x1 alert code happens when loading flash. 
```
I00411 bootstrap.c:134] Bootstrap: DONE!
I00412 boot_rom.c:61] Boot ROM initialisation has completed, jump into flash!
Loading processes from flash=0x20030000-0x200FFFFF into sram=0x10004000-0x1000FFFF
Loaded process[0] from flash=0x20030000-0x20030FFF into sram=0x10004000-0x10005413 = "timer_subscribe"
OpenTitan initialisation complete. Entering main loop
[0] function_LowLevelDebug: App 0x0 alert code 0x1
                                                  call @0x20030080(0x20030040, 0x10004000, 0x1414, 0x10004c00)
[0] memop(0, 0x10004864) = 0x0 = SUCCESS
[0] memop(10, 0x10004800) = 0x0 = SUCCESS
[0] memop(11, 0x10004864) = 0x0 = SUCCESS
[0] memop(0, 0x10004c64) = 0x0 = SUCCESS
[0] cmd(0x0, 0, 0x0, 0x0) = 0x1 = SuccessWithValue { value: 1 }
[0] cmd(0x0, 1, 0x0, 0x0) = 0x2710 = SuccessWithValue { value: 10000 }
[0] remove_pending_callbacks[0x0:0] = 0 callback(s) removed
[0] subscribe(0x0, 0, @0x200307ec, 0x1000470c) = 0x0 = SUCCESS
[0] cmd(0x0, 2, 0x0, 0x0) = 0xffff0048 = SuccessWithValue { value: 4294901832 }
[0] remove_pending_callbacks[0x0:0] = 0 callback(s) removed
[0] subscribe(0x0, 0, @0x0, 0x0) = 0x0 = SUCCESS
[0] cmd(0x8, 1, 0x1, 0x0) = 0x0 = SUCCESS
[0] yield

```

alistair23, 2021-01-10:
Hmmm.... You probably want to figure out if any syscalls are an error.

Look at the command and the returns. Something must be triggering a panic in the app.

jackieke724, 2021-01-11:
@alistair23 You are correct that the panic is from the app. I have learned that the weird order of the debug message is probably due to the console mux. 

As you have suggested, I am trying to associate the log with the source code. For this particular `timer_subscribe` example shown below, I've got some questions.
```
async fn main() -> TockResult<()> {
    let mut drivers = libtock::retrieve_drivers()?;

    drivers.console.create_console();

    let mut with_callback = drivers.timer.with_callback(|_, _| {
        println!("This line is printed 2 seconds after the start of the program.");
    });

    let mut timer = with_callback.init()?;
    timer.set_alarm(Duration::from_ms(2000))?;

    future::pending().await
}
```
In the previous log I posted, The first 2 `cmd` show that the alarm driver exists with frequency set at 10 kHz. Following that, is the `subscribe` assigning the println call back function to the alarm driver? 

Next, `cmd(0x0, 2, 0x0, 0x0)` reads the current counter tic value which is 4294901832 in this case, so it is probably from `let mut timer = with_callback.init()?;`. I do not understand what the next subscribe is intending to accomplish, but then panic happens at `cmd(0x8, 1, 0x1, 0x0)`. Which line of source code do you think is triggering panic?

If I should go deeper with gdb, how can I insert a breakpoint right before panic? Thank you!

jackieke724, 2021-01-11:
Have all the examples in libtock-rs been tested on opentitan earlgrey? Is there any configuration you think might have been out-dated?

alistair23, 2021-01-12:
Have you tried commenting out lines to figure out where the problem is?

I would recommend connecting GDB to figure out what is going on.

I don't think all the examples have been specifically tested, but they should all work.

jackieke724, 2021-01-12:
`timer.set_alarm(Duration::from_ms(2000))?;` is causing the problem.

jackieke724, 2021-01-12:
When I tried to debug with GDB, I set a breakpoint at `let mut timer = with_callback.init()?;`, and then continued from there, the line printed out fine. I have no idea why the code only works this way. Any clues?

I have also tested the leds and gpios (switches) which are working fine without the timer.

alistair23, 2021-01-12:
If it works in GDB but not without then it normally indicates a race condition. I don't have any simple next steps unfortunately. You are going to have to dig into the interrupt handling to see what is going wrong.

jackieke724, 2021-01-13:
In this case, how would I be able to debug a race condition in GDB? Is there a way to print out interrupts?

alistair23, 2021-01-13:
If you run on QEMU you can trace the interrupts, but that probably won't help in your case.

Maybe put break points in the interrupt handler and see if that sheds any light?

jackieke724, 2021-01-13:
Hey @alistair23 , I wanted to try if the recent commits could solve this issue, so I pulled your merged PR and updated the OT repo to the latest commit. Now, even the hello_world example is giving me fatal exception, shown below.
```
I00411 bootstrap.c:134] Bootstrap: DONE!
I00412 boot_rom.c:61] Boot ROM initialisation has completed, jump into flash!

panicked at 'fatal exception: LoadFault: 0x10000b5c', chips/earlgrey/src/chip.rs:248:13
        Kernel version release-1.6-578-g4449c9efc

---| Global debug writer not registered.
     Call `set_debug_writer_wrapper` in board initialization.

---| App Status |---
```

How can I add a debug writer in board initialization? Thanks!

alistair23, 2021-01-13:
Hmm... That seems like a memory mismatch.

Let me check again today.

alistair23, 2021-01-13:
Ok, it looks like you are using the wrong commit if you are getting that error.

You want to use the bitstream from: lowRISC/opentitan@99cb19827ebbd5dcee097c061a271ac09be5b523

The OT bitstream can *NOT* be any newer! There is a regression in the hardware and I see errors similar to what you mentioned above using a newer commit.

At the same time you want to use the latest Tock commit. I just checked with 584b323fbbf3b6346b0d8d823b12dc9093049494 and I don't see any issues.

I haven't yet tested the timer app, I'll test that now.

alistair23, 2021-01-13:
Ok, `libtock-rs` is broken since the new Alarm went in.

@phil-levis did you test with libtock-rs?