tock/tock #2344: Tracking: LoRa Support

bradjc, 2021-01-11:
Tock needs LoRa support, and this issue is a marker that we would like to see support for a stack.

Generally there are two versions:

1. Implement a full LoRaWAN stack in Tock (in kernel or userspace). #1760 is a start on this.
2. Use an external chip with a full LoRaWAN stack already, and send it AT commands.

bradjc, 2021-08-31:
Another LoRa driver is here: #2615

bradjc, 2023-05-26:
We merged #3360

alistair23, 2023-08-11:
I think this is now fully supported with: https://github.com/tock/tock/tree/master/boards/apollo3/lora_things_plus#radiolib

closed 2023-08-11 by alistair23

alistair23, 2023-08-23:
If anyone is interested this is another promising development environment

https://wiki.seeedstudio.com/Wio-WM1110_Dev_Kit_Hardware_Overview/

It's the nRF52840, which [Tock supports](https://github.com/tock/tock/blob/master/boards/nordic/nrf52840dk/README.md) with a LR1110. Unfortunately RadioLib doesn't support the [LR1110 yet](https://github.com/jgromes/RadioLib/issues/679)

Seeed have announced a product based on the development kit as well: https://www.kickstarter.com/projects/seeed/sensecap-t1000-track-what-you-care-with-no-boundaries

brghena, 2023-08-23:
Wow, this board has a lot of features for pretty cheap: https://www.seeedstudio.com/Wio-WM1110-Dev-Kit-p-5677.html

 * LoRaWAN
 * GPS
 * Antennas for LoRa and GPS
 * WiFi AP Scanning (detects nearby AP MAC addresses for positioning)
 * nRF52840 (so BLE/Thread)
 * Temp/Humidity and Accelerometer over I2C
 * AA battery holder AND a solar panel

bradjc, 2023-08-23:
> Wow, this board has a lot of features for pretty cheap: https://www.seeedstudio.com/Wio-WM1110-Dev-Kit-p-5677.html

Wow! Is this the future of Tock boards??

The issue is how to program it. Seems like by default USB is attached to UART (yuck):

[image]

We _can_ work with that, but we need some way to install the bootloader the first time.

bradjc, 2023-08-23:
Also, it doesn't seem to have actual GPS, but it can save GPS info which can be later processed to determine location.