tock/tock #2347: ARM: Support floating point

alistair23, 2021-01-12:
Currently Tock requires that floating point is disabled and disables it on platforms where it was enabled by a previous state.

It would be nice to allow user space apps to use the hardware's floating point support.

It's possible to enable floating point in the kernel by doing this in `arch/cortex-m/src/scb.rs` file:

```rust
/// Enable the FPU
#[cfg(all(target_arch = "arm", target_os = "none"))]
pub unsafe fn enable_fpca() {
    SCB.cpacr
        .modify(CoprocessorAccessControl::CP10::SET + CoprocessorAccessControl::CP11::SET);

    asm!("dsb", "isb", options(nomem, nostack, preserves_flags));

    if SCB.cpacr.read(CoprocessorAccessControl::CP10) != 3
        || SCB.cpacr.read(CoprocessorAccessControl::CP11) != 3
    {
        panic!("Unable to enable FPU");
    }
}

// Mock implementation for tests on Travis-CI.
#[cfg(not(any(target_arch = "arm", target_os = "none")))]
pub unsafe fn enable_fpca() {
    unimplemented!()
}
```

That causes issues when switching to/from apps using floating point. I'm assuming that's because we need to save/restore something.

Does anyone have comments/thoughts on enabling floating point for ARM and any hints as to do it?

hudson-ayers, 2021-01-12:
There was a decent amount of discussion around this here: https://github.com/tock/tock/pull/1010

and there is a tracking issue for cortex-m floating point here: https://github.com/tock/tock/issues/1246

alistair23, 2021-01-12:
Strange, those don't come up when I search "float" on GitHub. I'll close this in favour of #1246.

closed 2021-01-12 by alistair23