tock/tock #2352: Panic in LowLevelDebug

jettr, 2021-01-13:
The incoming `tx_buffer` is not necessarily the same size as `MESSAGE`. This will throw a panic from `copy_from_slice`.

https://github.com/tock/tock/blob/06ae6fb970ed991fa045e92dba2bb4f69dc41f40/capsules/src/low_level_debug/mod.rs#L65

An alternative would be `tx_buffer[..MESAGE.len()].copy_from_slice(MESSAGE);` but that panics if `tx_buffer` is not large enough. We probably want to add some bounds checking.

jrvanwhy, 2021-01-13:
I think the fix is to use your suggestion `tx_buffer[..MESSAGE.len()].copy_from_slice(MESSAGE);` and redefine `BUF_LEN` as:

```
pub const BUF_LEN: usize = max(fmt::BUF_LEN, MESSAGE.len());
```

You would have to move or change the visibility of `max` and move `MESSAGE` to module scope. Maybe rename `fmt::BUF_LEN` -- or move `MESSAGE` into `fmt` and include `MESSAGE` in the `fmt` length calculation (which wouldn't change the length calculation at all).

jrvanwhy, 2021-01-13:
This issue highlights a lack of testing: I'm not sure whether there is currently a mechanism that would allow me to test the grant allocation failure case. I was able to test the other failure case (queue filled up) by hand but not grant failure.

jettr, 2021-01-13:
I ran into this issue when I didn't have enough HEAP space in my app if that help for a manually reproduction.

You should be able to add a test mod here and using `cargo test`, right and just pass in a buffer that isn't the same size as MESSAGE

commit 0e333fcc (2021-10-05): capsules: lld: Copy only length of message

We know buffer will be long enough because the min BUF_LEN is longer
than the const MESSAGE.

Resolves #2352
PR #2848 (2021-10-05): capsules: lld: Copy only length of message
### Pull Request Overview

We know buffer will be long enough because the min BUF_LEN is longer than the const MESSAGE.

Resolves #2352.

### Testing Strategy

travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2021-10-05 by bors[bot]