tock/tock #2366: vDSO style syscalls

alistair23, 2021-01-19:
Many modern operations systems use something similar to Linux's [vDSO](https://man7.org/linux/man-pages/man7/vdso.7.html) style syscalls.

That is for some common syscall operations instead of requiring userspace to trap into the kernel, which can take a long time, they can just read the value from memory.

The Linux kernel documenation describes this well:

```
       Why does the vDSO exist at all?  There are some system calls the
       kernel provides that user-space code ends up using frequently, to
       the point that such calls can dominate overall performance.  This
       is due both to the frequency of the call as well as the context-
       switch overhead that results from exiting user space and entering
       the kernel.
```

I think it's worth adding support to Tock for vDSO style syscalls. The general idea being:

 1. All applications are made aware of a memory address used for vDSO (where should this be?)
 2. When the kernel returns to an app it writes all of the configured values to the vDSO memory location
 3. The applications libtock library can then check the existence of vDSO values before performing a syscall. This should be handled in the library, not by the users application code.

The values in memory can be stored in simple key/value pairs, where the key matches with the existing syscall values. That way an application can easily determine what vDSO syscalls are supported.

Two good vDSO syscall candidates would be the current time and are any callbacks pending. Capsules can add their own vDSO syscalls as well. For example maybe the ADC capsule allows an application to setup the ADC value in a vDSO syscall, then allowing the application to read the value without trapping.

bradjc, 2021-01-19:
This could be global or per-process. Global would be simpler to setup, but putting the memory location inside of the process's already accessible memory would be easier to make work with the MPU.

bradjc, 2021-01-19:
> For example maybe the ADC capsule allows an application to setup the ADC value in a vDSO syscall, then allowing the application to read the value without trapping.

I think this is the same thing as `allow`.

alistair23, 2021-01-19:
I think this would need to be local, to ensure we don't leak information. We could require the app to call `allow` to get a region of memory or always use a constant offset from the start of the apps memory.

> > For example maybe the ADC capsule allows an application to setup the ADC value in a vDSO syscall, then allowing the application to read the value without trapping.
> 
> I think this is the same thing as `allow`.

Yes, this is pretty much the same as allow without a callback. There might be some occasions where we want this to update, even if an interrupt hasn't triggered (such as writing the current time) in which case it ends up a little different to allow.

lschuermann, 2021-01-21:
> Two good vDSO syscall candidates would be the current time and are any callbacks pending. Capsules can add their own vDSO syscalls as well. For example maybe the ADC capsule allows an application to [...]

I like this idea. I think `allow` could actually be a good fit here, although I'm slightly concerned about the preemptive scheduling nature. How would we guarantee that userspace read a value without it being overwritten by the kernel in the meantime?

I imagine only a select number of capsules would want to do this, and every capsule should still be fully operational without using vDSO-ish operating mechanisms. I can imagine adding a `VDSODriver: Driver` trait and providing a method in the `Platform` trait to iterate over all vDSO drivers before the kernel returns to a process. With an upcoming PR about a Platform struct helper macro that should be possible.

alistair23, 2021-01-21:
I don't think we need to guarantee anything. Whatever user space reads is fine, we don't need the value to be 100% accurate (it anyways will only be as accurate as the last trap).

Yep, if userspace never swaps out then it won't be accurate. That should be up the the applications libtock though to determine if it wants to do this or a full syscall.

alistair23, 2021-01-25:
I have looked at this a bit more.

I don't think we can allocate a read only vDSO region in the app process. I was looking at having something up near or part of the grant region. This doesn't end up working though as there are MPU alignment constraints that become hard to meet without wasting space. For example squeezing in 16 bytes (4 32-bit values) between the app heap and the grant region would require a full 256 bytes for the ARM MPU. RISC-V is more forgiving when using TOR though. This doesn't include space wasted by the next region being unaligned as we have extended past the apps allocated memory. Cutting dowwards into the app space seems like an option, but again it has to allign with the grant and I think ends up wasting space.

I think that leaves two options.

A global region that is readable by all apps. This means the kernel can just blindly update this region when returning to an app without caring what app is running. This would probably be implemented kind of like a special app that is created first. The problem here again is MPU alignment issues and I think we could waste space, even if nothing is using it.

The other option is just a special capsule that an app has to call `allow` on and it's "interrupt handler" is triggered before swapping to an app. That seems the easiest and fits in with Tock's model so I'll have a look at that.

alistair23, 2021-01-25:
I have opened a draft PR: https://github.com/tock/tock/pull/2381

phil-levis, 2021-01-26:
A global region that's readable by all apps won't be implementable by some platforms. IIRC, the OpenTitan MPU is already all spoken for by Tock's process model?

That being said, this seems like a potentially very useful piece of functionality. But I'm a little wary of adding a piece of functionality -- especially one dealing with the kernel/userspace boundary -- without some concrete use cases and design goals. Tock is used in very different situations than Linux; when processes spend most of their time idle, the drift of ticks can be much bigger than systems making tens of thousands of system calls per second. 

Use cases that seem valuable to me:
  - process ID
  - prior execution exit code (if any)
  - an incrementing counter of invocation count
 
Is there anything in Linux that uses vDSO besides `gettimeofday`?

It's worth noting vDSO was designed in part to address security issues of the previous mechanism (`vsyscall`); we probably want to pay very close attention to what those issues were so we don't repeat mistakes.

alistair23, 2021-01-26:
Yes, I think you are right and we would have trouble fitting in all PMP regions. We could change the PMP implementation to not use TOR though.

I went with just having apps pass allow. That way we don't have to worry about another MPU region, where to put it or how to communicate that information to the app. I was worried that MPU allignment constraints would waste a large amount of memory to support a single read only region.

Those are good cases, we can look at adding them if the current design is approved.

Linux uses vDSO for time, CPU information, res (not sure what that is) as well as architecture specific information, like sigtrap (powerPC). RISC-V has a [flush icache vDSO](https://github.com/bminor/glibc/blob/21c3f4b5368686ade28d90d8c7d79c4c95c72c1b/sysdeps/unix/sysv/linux/riscv/flush-icache.c#L40)  as well.

Yep, by using `allow` I think we don't expose anything as we aren't sharing information between apps.

bradjc, 2021-02-01:
> We could change the PMP implementation to not use TOR though.

Just one note which is not super relevant but important to keep in mind: we do need to maintain some level of stability with how MPUs work on each platform. Since outside-the-kernel tools (particularly tockloader and elf2tab) need to know MPU restrictions to properly create binaries and flash them, we rely on MPU requirements not really changing to make that easier and prevent mismatches between how apps are built and what the kernel expects. We may also want to formalize this better. I might write up an RFC to that effect.

bradjc, 2021-09-28:
Implemented in #2381

closed 2021-09-28 by bradjc