tock/tock #2369: Process::memory violates aliasing requirements (undefined behavior)

jrvanwhy, 2021-01-20:
[`Process::memory`](https://github.com/tock/tock/blob/2a729631fca67e9eb7c1b6dac3ec0a89c9ff92b5/kernel/src/process.rs#L821) is a `&'static mut [u8]` that points to a process' memory region. Although Rust's memory model is not yet 100% formally specified, it seems very unlikely that holding a `&mut [u8]` reference into memory that is mutated by the userspace process is not undefined behavior. It is certainly undefined behavior under the stacked borrows model.

hudson-ayers, 2021-01-20:
Looking at this briefly, `Process::memory` is only ever used via calls to `self.memory.as_ptr()` and `self.memory.len()`, so it should be pretty straightforward to instead store a pointer and length (in fact, I wonder if we used to do exactly that).

bradjc, 2021-01-20:
To the untrained eye, storing a pointer and length is exactly what a slice does. Can we keep a reference to a non-mutable slice?

jrvanwhy, 2021-01-20:
> To the untrained eye, storing a pointer and length is exactly what a slice does. Can we keep a reference to a non-mutable slice?

They are structurally equivalent but not semantically equivalent.

If we store a non-`mut` slice, then we would generate undefined behavior when we produce a `&mut [u8]` as a result of an `allow` system call (I think the kernel does that). Under the stacked borrows model, if we never generate a `&mut [u8]` pointing into application memory, the app mutating that memory still causes undefined behavior.

Using a pointer + length does not carry the same semantics around mutation.

lschuermann, 2021-01-21:
Ha! Just wanted to open this issue. Agree with the proposed solution of storing a pointer and length. In general I feel we should really enforce only a single mutable slice reference to any process-available memory section existing in the kernel. Given a capsule can deference an AppSlice at any time, it's hard to uphold that guarantee if we also store slices in the kernel.

bradjc, 2021-01-28:
We also have a mutable slice in main.rs. Is that OK?

jrvanwhy, 2021-01-28:
> We also have a mutable slice in main.rs. Is that OK?

I believe that is problematic as well, as it could overlap with mutable references obtained via `AppSlice`.

That said, I don't expect it to result in bugs in the near future as long as the slice isn't used after processes start executing -- the compiler doesn't really have a reason to "notice" the presence of undefined behavior.

bradjc, 2021-01-28:
This is then where things get a little messier in my mind: what is less likely to cause bugs? Human programmers writing C-esque code with pointers and lengths, or compilers with non-guaranteed behavior?

Not storing `memory: [u8]` in Process is a pretty simple change. Unfactoring loading processes to not use a slice is a bigger change, and one that can easily create bugs.

jrvanwhy, 2021-01-28:
> Not storing `memory: [u8]` in Process is a pretty simple change. Unfactoring loading processes to not use a slice is a bigger change, and one that can easily create bugs.

Would you mind pointing the slice out to me? I looked through the hifive1 board file but most types are inferred so I didn't find it.

hudson-ayers, 2021-01-28:
https://github.com/tock/tock/blob/af1efe87f3f52f9dc923f0ca936091f7681ce59a/boards/hifive1/src/main.rs#L245 constructs a mutable static slice to app memory, which is then used in process.rs to load processes here: https://github.com/tock/tock/blob/af1efe87f3f52f9dc923f0ca936091f7681ce59a/kernel/src/process.rs#L139 .

I suspect that if we restore `Process::memory` to being a pointer and a length, we could get away with passing a non-static slice to `load_processes()`, which might relieve any concerns about UB: No `AppSlices` are created until after loading processes is finished.

jrvanwhy, 2021-01-28:
Unless I am missing something, that slice disappears when `load_processes()` terminates, except for `Process::memory`. If we change `Process::memory` to use a raw pointer, then it will disappear entirely after `load_processes()`. Because that happens before processes start executing, I don't see how that can produce undefined behavior.

Am I missing something?

hudson-ayers, 2021-01-28:
Yeah, that was the point I was trying to make. I'm just unsure of whether we would *technically* need to change the signature of `load_processes()` to take a non-static slice (which should be easy if we change `Process::memory` -- nothing else requires it to be `static`!). In practice I cannot imagine it would ever produce undefined behavior either way.

jrvanwhy, 2021-01-29:
> I'm just unsure of whether we would *technically* need to change the signature of `load_processes()` to take a non-static slice (which should be easy if we change `Process::memory` -- nothing else requires it to be `static`!).

Although we *technically* need a Rust specification to answer this, I am confident the answer is "no". None of the aliasing rules I am aware of for Rust care about lifetimes -- lifetimes exist purely for the borrow checker. As a result, changing lifetimes only affects whether code compiles, not whether it would produce undefined behavior.

I think it would be beneficial to change it to a non-`'static` reference to prevent future mistakes.

hudson-ayers, 2021-01-29:
makes sense, thanks for explaining

commit d6059d52 (2021-03-24): Merge #2494

2494: kernel: remove memory slice from Process struct r=ppannuto a=bradjc

### Pull Request Overview

This pull request removes the mutable slice to process memory that the Process object holds. This helps avoid undefined behavior since there are many cases (like AppSlices) where we generate new slices to the same process memory.

I also was able to remove an `unsafe` by changing `add`->`wrapping_add`.

Fixes #2369.

### Testing Strategy

Running an app on Hail. The change is pretty mechanical.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant fi [... cut]
PR #2494 (2021-03-24): kernel: remove memory slice from Process struct
### Pull Request Overview

This pull request removes the mutable slice to process memory that the Process object holds. This helps avoid undefined behavior since there are many cases (like AppSlices) where we generate new slices to the same process memory.

I also was able to remove an `unsafe` by changing `add`->`wrapping_add`.

Fixes #2369.

### Testing Strategy

Running an app on Hail. The change is pretty mechanical.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are requir [... cut]
closed 2021-03-24 by bors[bot]