tock/tock #2396: trd-syscall: subscribe callback cancellation ambiguity

ppannuto, 2021-01-29:
The juxtaposition of `immediately return` and `must not invoke the previous callback after the call to subscribe` is possibly ambiguous:

> If the passed callback is not valid (is outside process executable
> memory and is not the Null Callback described below), the kernel MUST
> NOT invoke the requested driver and MUST immediately return a failure
> with a return code of EINVAL.
> 
> A passed callback MUST be valid until the next invocation of `subscribe`
> with the same syscall and driver identifier. When userspace invokes
> subscribe, the kernel MUST cancel all pending callbacks for that driver
> and subscribe identifier: it MUST NOT invoke the previous callback after
> the call to subscribe, and MUST NOT invoke the new callback for events
> that occurred before the call to subscribe.

What is the expected behavior in the case that a second call to subscribe passes an invalid callback? Is the original callback cancelled (and replaced with the null callback?) or is the original callback still active?

lschuermann, 2021-01-29:
Juxtaposition, what a great new word to learn. :)

> What is the expected behavior in the case that a second call to subscribe passes an invalid callback? Is the original callback cancelled (and replaced with the null callback?) or is the original callback still active?

I think this is a valid concern and could be specified better. As far as I know -- in the current implementation -- we refuse the operation entirely and don't swap out the value from the Driver, so an invalid payload to allow or subscribe has no observable side-effect other than the context switch to the kernel. I think this is fine. Would you rather have it do something different?

ppannuto, 2021-01-29:
I think that's perfectly valid (and probably my preferred) behavior, but also think it would be nice for the documentation to assert that intentionally / specifically.

---

I blame my father for the sometimes gratuitous lexicon, we played a lot of word games as a family growing up :)

lschuermann, 2021-02-03:
I've opened a PR to clarify this. Thinking about this, it's actually an important issue with many consequences regarding the implementation and the side effects of scheduling a Callback, which I tried to outline in the description. Let me know what you think.

phil-levis, 2021-02-15:
#2405

commit c3ac6a9e (2021-02-15): Tock 2.0: clarify subscribe behavior w.r.t. invalid/refused callback

The description of the Subscribe system call semantics was possibly
contradicting itself by stating that, in case of passing an invalid
Callback both

- the driver would never be invoked and
- the previously registered callback must not be invoked after the
  call to subscribe.

This resolves this issue by clearly stating that only in case of
accepted callbacks this guarantee is made.

Fixes: #2396 (https://github.com/tock/tock/issues/2396)
Reported-by: Pat Pannuto <pat.pannuto@gmail.com>
commit e4dc8b72 (2021-02-15): Merge #2405

2405: Tock 2.0: clarify subscribe behavior w.r.t. invalid/refused callback r=phil-levis a=lschuermann

### Pull Request Overview

The description of the Subscribe system call semantics was possibly contradicting itself by stating that, in case of passing an invalid Callback, both

- the driver would never be invoked and
- the previously registered callback must not be invoked after the call to subscribe.

This resolves this issue by clearly stating that only in case of "accepted" callbacks (i.e. the kernel has invoked the driver's `subscribe`method AND the driver has returned the  [... cut]
PR #2405 (2021-02-15): Tock 2.0: clarify subscribe behavior w.r.t. invalid/refused callback
### Pull Request Overview

The description of the Subscribe system call semantics was possibly contradicting itself by stating that, in case of passing an invalid Callback, both

- the driver would never be invoked and
- the previously registered callback must not be invoked after the call to subscribe.

This resolves this issue by clearly stating that only in case of "accepted" callbacks (i.e. the kernel has invoked the driver's `subscribe`method AND the driver has returned the old callback through the `Ok` variant) this gua [... cut]
closed 2021-02-15 by phil-levis