tock/tock #2413: TicKV Hash Algorithm

alistair23, 2021-02-05:
TicKV has made some progress and is now at the point where we need a hash algorithm. The question now is which algorithm and which implementation.

The algorithm is not used for cryptography, so we don't need a crypto one.

Properties:
 * Fast
 * Easy-ish to implement on embedded systems
 * Good for kv stores and resistant to collisions
 * Produces a 64-bit hash

I have no intentions of inventing another algorithm. My plan was to use an existing one and probably also an existing implementation, unless there is a good reason to re-implement it (there rarely is).

I have found two main contenders, although I'm option to other options.

The main options I have found are:
 - [SipHasher](https://github.com/veorq/SipHash/blob/master/README.md): Provided by [Rust core](https://doc.rust-lang.org/core/hash/struct.SipHasher.html) and now a [standalone crate](https://github.com/jedisct1/rust-siphash)
 - [SwissTable](https://abseil.io/blog/20180927-swisstables) used by Rust for it's [HashMap](https://doc.rust-lang.org/std/collections/struct.HashMap.html)

Other options:
 - [Spooky hash](https://burtleburtle.net/bob/hash/spooky.html)

I'm leaning towards SipHasher, but would like to know what others think.

alistair23, 2021-02-05:
@phil-levis thoughts?

jrvanwhy, 2021-02-05:
SwissTable is a hash table, not a hash function.

You should probably consider Fnv as well.

alistair23, 2021-02-06:
Thanks! I skimmed through the SwissTable section too quickly.

alistair23, 2021-02-06:
At a quick look FNV opens up the possibility of a DOS from malicious apps with large keys. For smaller operations it does seem to be faster.

l162, 2021-02-09:
Likely unrelated, but would it be possible to use set approximation (like a bloom filter) here? Have a `needs_alloc` (const generic) version of [Xor Filters](https://arxiv.org/abs/1912.08258) sitting around at home. Fuse filters may be particularly good for large number of keys (opt-in depending on how a board intends to use the kv). Could potentially decrease object overhead, but also lead to different design.

wright-joe, 2021-03-15:
[Murmur3](https://docs.rs/murmur3/0.5.1/murmur3/)

l162, 2021-05-03:
https://github.com/nickbabcock/highway-rs

l162, 2021-05-03:
[Highway](https://docs.rs/highway/0.6.4/highway/)

closed 2021-12-16 by bors[bot]