tock/tock #2415: Confusing behaviour when trying to invoke SPI operations before calling set_client

mmurray22, 2021-02-07:
**Summary of Issue:** When calling configure on a VirtualSpiMasterDevice before setting the SpiComponent client results in a silent error that prevents SPI from working. I was wondering if the client can be added to the MUX when the VirtualSpiMasterDevice is created in `::new`, which I think would be more intuitive. 

**Explanation:**
While developing and testing kernel code which invokes remote system calls, I ran into the following error: 

I instantiated the remote system call capsule in the imix/sr/main.rs as such:
```
let remote_mux_spi = components::spi::SpiMuxComponent::new(&sam4l::spi::SPI)
        .finalize(components::spi_mux_component_helper!(sam4l::spi::SpiHw));
let remote_spi = SpiComponent::new(remote_mux_spi, 2)
        .finalize(components::spi_component_helper!(sam4l::spi::SpiHw));
let remote_system_call = static_init!(capsules::system_call_interface::RemoteSystemCall<'static>,
                                          RemoteSystemCall::new(&mut BUF, &mut BUF_CLI, &mut CLIENT, remote_spi));
remote_spi.set_client(remote_system_call);
```
where the remote_system_call object above is the element being included in the Imix struct. Inside the `RemoteSystemCall::new` function, I have the following code where I configure the VirtualSpiMasterDevice:
```
spi.configure(
          spi::ClockPolarity::IdleLow,
          spi::ClockPhase::SampleLeading,
          400_000
);
```
However, I configure the VirtualSpiMasterDevice before `remote_spi.set_client(remote_system_call)` which means I was attempting to configure the device before it is added to the MUX, as seen in the `set_client` function, which is copied below:
```
pub fn set_client(&'a self, client: &'a dyn hil::spi::SpiMasterClient) {
        self.mux.devices.push_head(self);
        self.client.set(client);
 }
```
While I now know this error can be fixed by making sure to reserve invoking SPI operations until after the client is set, the current setup does not seem as intuitive as it could be and not invoking the commands in the correct order leads to silent errors. 

That way, upon creation it is automatically added to the MUX and we don't have to wait until the object is set as a client to start running SPI operations.
I was wondering if the client can be added to the MUX when the VirtualSpiMasterDevice is created in `::new`. That way, upon creation it is automatically added to the MUX and we don't have to wait until the object is set as a client to start running SPI operations.

Note: I've been developing on an older version of the tock kernel, so I do not have any master changes made in the last few months or so.

phil-levis, 2021-02-08:
There seem to also be some deeper issues with the virtualizer, e.g., calls to configure pass through directly to reconfigure the SPI. Because configure is not a split-phase operation, this is a bit of an issue: calling `configure` pushes an op, which might execute asynchronously. Per-client configuration should be stored in each client, then set when that client performs an operation. If the RAM costs for this are an issue, you can do a callback to fetch the values from the client (this is what ICEM did on TinyOS).

bradjc, 2021-02-09:
I think the issue with `set_client()` is just a fundamental issue in Tock. That pattern is everywhere, and a lot of silent errors occur if you do not call set_client properly. Components are our main tool to address this, as then one person just has to write it correctly and any board can use it.

But as @phil-levis notes the SPI stack is clearly not well implemented. In addition to the SPI settings not being virtualized per-client, the SpiDevice HILs should have `set_client()` methods.

closed 2021-04-23 by bradjc