tock/tock #2425: RFC: Fixing unnecessary stack use on all platforms

hudson-ayers, 2021-02-10:
The new peripheral instantiation approach has forced us to increase the stack size on several platforms. Here is why:

`static_init!()` initializes items on the stack before copying them into the relevant global. As a result, when the peripherals are created using `static_init!(Sam4lDefaultPeripherals, Sam4lDefaultPeripherals::new(pm))`, they are allocated on the stack in `reset_handler()`. Unfortunately, `reset_handler()` never returns, so as far as I can tell all local variables in `reset_handler()` take up stack space forever. Notably, this issue actually applies to *everything* we allocate using `static_init!()` directly in `reset_handler()` -- the new peripheral approach just exposed it because the peripherals are big.

**In summary: using `static_init!()`in `reset_handler()` uses twice as much memory as the thing being statically allocated: the memory for the global is used, and the stack memory is used forever**

Another note: `lazy_static` and `Box::new()` have this same issue, though of course they do not affect us.

I have thought of 4 potential solutions to this, and would like people's thoughts on each.

## Solutions

I have been able to come up with several solutions, each with a unique set of tradeoffs.

### Solution 1
By convention, create peripherals in a separate, `#[inline(never)]` function:
```rust
#[inline(never)]
unsafe fn create_peripherals(
    pm: &'static sam4l::pm::PowerManager,
) -> &'static mut Sam4lDefaultPeripherals {
    static_init!(Sam4lDefaultPeripherals, Sam4lDefaultPeripherals::new(pm))
}

#[no_mangle]
pub unsafe fn reset_handler() {
    sam4l::init();
    let pm = static_init!(sam4l::pm::PowerManager, sam4l::pm::PowerManager::new());
    let peripherals = create_peripherals(pm);
    ...
```
Using `-Z emit-stack-sizes`, this drops the stack frame required for `reset_handler()` from 1496 bytes --> 248 bytes.

This also prevents stack overflows in practice. If I add a 6450 byte stack allocation to the end of reset_handler, the kernel still runs with this code on the board, but there is a stack overflow with the current approach.

Pros: 
- stack space used to allocate the peripherals is freed immediately

Cons:
- stack still has to be at least as large as the `DefaultPeripherals` struct
- gross boilerplate for creating peripherals with non-obvious reason for why it is needed

### Solution 2
Create a new `static_init_drop_stack!` macro that could be used anytime a large object needs to be allocated via `static_init` in `reset_handler()`. Look something like this:
```rust
#[macro_export]
macro_rules! static_init_drop_stack {
    ($T:ty, $constructor: expr, $arg:expr, $T2:ty $(,)?,) => {{
        #[inline(never)]
        unsafe fn create_thing(arg: $T2) -> &'static mut $T {
            let mut buf = $crate::static_buf!($T);
            buf.initialize($constructor(arg))
        }
        create_thing($arg)
    }};
}
```
Usage:
```rust
    let peripherals = kernel::static_init_drop_stack!(
        Sam4lDefaultPeripherals,
        Sam4lDefaultPeripherals::new,
        pm,
        &'static sam4l::pm::PowerManager,
    );
```

Currently this only works for a single argument to the constructor, but I believe I could support any number of arguments with some effort.

Pros:
- one line replacement in main
- could be used for stuff other than peripherals

Cons: 
- using the macro is more verbose than normal `static_init!()`, with somewhat more confusing syntax
- stack still has to be at least as large as the `DefaultPeripherals` struct

### Solution 3
Make each `DefaultPeripheral` struct provide its own macro for instantiating itself, instead of a `new()` function. Have the `DefaultPeripheral` struct store references to peripherals instead of owned types. New macro would internally call `static_init!()` on every peripheral individually, and assign these fields to the larger struct.

Pros:
- removes requirement that stack be as large as total size used for peripherals
- single line replacement in main

Cons:
- authors of new chips have to write a macro

### Solution 4
Move almost everything in `reset_handler()` into a `setup_board()` function with its own stack frame.

as a result, `reset_handler` now looks like this:
```rust
#[no_mangle]
pub unsafe fn reset_handler() {
    let (board_kernel, imix, chip, scheduler) = board_setup();
    let main_cap = create_capability!(capabilities::MainLoopCapability);
    board_kernel.kernel_loop(&imix, chip, Some(&imix.ipc), scheduler, &main_cap);
}

#[inline(never)]
unsafe fn board_setup() -> (
    &'static Kernel,
    Imix,
    &'static Sam4l<Sam4lDefaultPeripherals>,
    &'static RoundRobinSched<'static>,
) {
    sam4l::init();
    ...
```

pros:
- no prolonged stack use from any `static_init()` calls -- now `reset_handler` only has 200 byte stack frame on Imix.
- no hacky macros needed
- Also reduces code size by ~2kB on Imix, though I have no idea why

cons:
- big diff for every board when we make the change
- still feels sorta hacky

## Questions you may have

> Why can' we just use a block scope, rather than a dedicated `#[inline(never)]` function?

AFAICT, LLVM does not move the stack pointer to free local variables when they go out of scope, only at the end of a function

> But can't LLVM reuse stack space dedicated for local variables that have gone out of scope?

In theory yes, but it does not seem to here. I am not sure if this is something that LLVM does not support at all, or if this is just an optimization we do not trigger.

> Why can't we just initialize directly in the static, rather than allocating on the stack and then copying in?

If you know of a way to do this please let me know. I was not able to figure out a way to do so and suspect it is not possible.

> Can we modify `static_init!()` so that anything passed to it is deallocated at the end of the scope of the macro?

I tried several approaches to this and failed to find anything that would not require changing the interface.

Using a function internally means that it is impossible to capture elements from the callers environment, (e.g. `pm` in the sam4l example). Using an anonymous closure works, but marking an anonymous closure as `[inline(never)]` requires a nightly feature that seems unlikely to be stabilized, and complicates lifetime checking for elements passed into the closure in a way that breaks a lot of existing code.

> What is the stack frame size for `reset_handler()` for all upstream Tock boards today?

Glad you asked!
```
File: acd52832.elf
         1504     reset_handler
File: clue_nrf52840.elf
         2240     reset_handler
File: hail.elf
          928     reset_handler
File: imix.elf
         1496     reset_handler
File: imxrt1050-evkb.elf
         2192     reset_handler
File: microbit_v2.elf
         1696     reset_handler
File: msp-exp432p401r.elf
         2368     reset_handler
File: nano33ble.elf
         2216     reset_handler
File: nrf52840dk.elf
         2240     reset_handler
File: nrf52840_dongle.elf
         2224     reset_handler
File: nrf52dk.elf
         1496     reset_handler
File: nucleo_f429zi.elf
         2760     reset_handler
File: nucleo_f446re.elf
         2744     reset_handler
File: redboard_artemis_nano.elf
          552     reset_handler
File: stm32f3discovery.elf
         2120     reset_handler
File: stm32f412gdiscovery.elf
         2752     reset_handler
File: teensy40.elf
         2384     reset_handler
```

bradjc, 2021-02-10:
Thanks for your great summary. I know that peripheral instantiation has made this a pressing issue, but really any large structure that is statically init-ed could cause problems, so I'm just considering all static_init-ed objects. It seems like there are four general ways to address this:

- A: Do all `static_init` in a function that is not the reset handler.
- B: Replace `new()` calls with macros, so things aren't passed on the stack.
- C: Change static init so that LLVM can free stack space.
- D: Figure out how to get LLVM to delete reset_handler's stack frame (I take it it is not clear how to do this).

My preference in order is D, C, A. It seems like D might not be possible. Updating static_init seems fine to me, but it seems like it would complicate calling the macro quite a bit, which is probably too ergonomically unfriendly to be viable. That leaves A.

I think splitting things into functions in main.rs as convention would be fine. Alternatively, can components help? What if we make peripheral instantiation a component?

hudson-ayers, 2021-02-12:
Making peripheral instantiation a component is a possibility. Unfortunately making a generic component that could create any peripheral struct is pretty tricky (how can you call `xxxDefaultPeripheral::new()` from within `finalize()` if your component is generic over the `DefaultPeripheral` type?).

Making a per-chip component would work, but it could not go in the components folder without making the `components` crate have a dependency on every chip. This would be unfortunate -- someone building `nrf52dk` probably does not want to have to compile every chip in the Tock repository. The component could be placed in each specific `chip` directory, but then we would be breaking our "rule" that `static_init!()` only be called from within `boards/`.

If breaking that rule is okay, I think the best approach is probably to change `xxxDefaultPeripherals::new()` to individually `static_init!()` each peripheral internally. Lets call this Option 5. For Imix, the result would look like this:

```rust
pub struct Sam4lDefaultPeripherals {
    pub acifc: &'static mut crate::acifc::Acifc<'static>,
    pub adc: &'static mut crate::adc::Adc,
    pub aes: &'static mut crate::aes::Aes<'static>,
    // ...
}

impl Sam4lDefaultPeripherals {
    pub fn new(pm: &'static crate::pm::PowerManager) -> Self {
        use crate::dma::{DMAChannel, DMAChannelNum};
        Self {
            static_init!(crate::acifc::Acifc, acifc: crate::acifc::Acifc::new()),
            adc: static_init!(crate::adc::Adc, crate::adc::Adc::new(crate::dma::DMAPeripheral::ADCIFE_RX, pm)),
            aes: static_init!(crate::aes::Aes, crate::aes::Aes::new()),
            //...
        }
    }
}
```
If you need a stack smaller than the sum of all your peripherals, no problem -- say GPIO is your biggest peripheral (lots of pins):

```rust
#[inline(never)]
create_gpio_peripherals() -> &'static mut crate::gpio::ReallyBigGpio {
    static_init!(crate::gpio::ReallyBigGpio, crate::gpio::ReallyBigGpio::new())
}

impl Sam4lDefaultPeripherals {
    pub fn new(pm: &'static crate::pm::PowerManager) -> Self {
        use crate::dma::{DMAChannel, DMAChannelNum};
        Self {
            static_init!(crate::acifc::Acifc, acifc: crate::acifc::Acifc::new()),
            adc: static_init!(crate::adc::Adc, crate::adc::Adc::new(crate::dma::DMAPeripheral::ADCIFE_RX, pm)),
            aes: static_init!(crate::aes::Aes, crate::aes::Aes::new()),
            //...
            gpio: create_gpio_peripherals(),
        }
    }
```

bradjc, 2021-02-12:
We could just put the components in each board crate. We also already have an nrf52 components crate. Copying them wouldn't be ideal, but is there any reason the component code would change or need to be updated really?

hudson-ayers, 2021-02-15:
It seems possible that the component code would need to be updated when new peripherals are added, at least when adding peripherals changes the function signature for `SomeChipDefaultPeripherals::new()`.

Personally I would probably prefer `static_init!()` in chips to duplicated components in each board crate.

hudson-ayers, 2021-03-16:
So it turns out that updating to the newest Rust nightly (2021-03-15) has at least partially fixed this issue, thanks to https://github.com/rust-lang/rust/pull/82806 . Thanks to the upgrade to LLVM 12, LLVM now can optimize away `memcpy`s across blocks, which means that `static_init!()` no longer allocates items on the stack, then copies them to the global -- instead LLVM allocates directly into the memory of the global! At least, https://github.com/rust-lang/rust/issues/53827#issuecomment-797759625 suggests that is what happens for `Box::new()` now, which I believe to be the same basic problem.

stack frame sizes of all `reset_handler()`s on latest nightly:
```
File: acd52832.elf
         1400     reset_handler
File: clue_nrf52840.elf
         2056     reset_handler
File: hail.elf
          456     reset_handler
File: imix.elf
          520     reset_handler
File: imxrt1050-evkb.elf
         2384     reset_handler
File: microbit_v2.elf
         1776     reset_handler
File: msp-exp432p401r.elf
          440     reset_handler
File: nano33ble.elf
         2040     reset_handler
File: nrf52840dk.elf
         2064     reset_handler
File: nrf52840_dongle.elf
         2048     reset_handler
File: nrf52dk.elf
         1400     reset_handler
File: nucleo_f429zi.elf
         3208     reset_handler
File: nucleo_f446re.elf
         3192     reset_handler
File: redboard_artemis_nano.elf
          288     reset_handler
File: stm32f3discovery.elf
         2472     reset_handler
File: stm32f412gdiscovery.elf
         3224     reset_handler
File: teensy40.elf
         2384     reset_handler
File: earlgrey-nexysvideo.elf
          272     reset_handler (was 944 before, though I left it out in my top post)
File: litex_arty.elf
          144     reset_handler (was 144 before, though I left it out in my top post)
File: arty_e21.elf
          112     reset_handler
File: hifive1.elf
          112     reset_handler
```

`Imix`, `Hail`, `Msp-exp432`, the `redboard_artemis_nano`, and `earlgrey` see substantial reductions in stack frame size. All of the nrf52 boards see modest reductions of 100-150 bytes. I suspect this could be because there are two layers of initialization code for the nrf chips, because there is some shared initialization between the nrf52840 and nrf52832, and some not shared. The `teensy40`, `litex_arty`, and `arty_e21`/`hifive1` are unchaged. The stm based boards and the `imxrt1050`, however, all seem to see substantial increases in stack frame size, which I do not immediately know the cause of. The STM boards also have the two-layer initialization, but I would think that would lead to no change, not an increase.

In general, it remains the case that this "fix" is just a compiler optimization that is not guaranteed to apply. So we can try to write code that makes the optimization possible, or stick with the fix we decided on before this was an option.

I also tried to look at the assembly, but disassembly with opt-level=z can be a bit of a nightmare. Disassembly with source (`arm-none-eabi-objdump -D -S imix.elf`) only works on the elf compiled on the newest nightly (I think this is because LLVM-12 also fixed #1922 ). That said, the assembly is definitely different! I had a hard time fully understanding what is happening, but a couple observations:

- LLVM 12 is smarter about dead code elimination, and figuring out what variables are never called anywhere. Peripherals that are never used are not allocated at all. For example, on sam4l, usart0, usart1, and usart2 are not allocated, but usart3 (the only one that is used) is. On a related note, the newer nightly reduces Imix code size by 6kB!
- The old asm has lots of calls to `__aeabi_memclr4__` that are not present in the new asm

If anyone wants to take a closer look, I have the objdump output, let me know.

bradjc, 2021-03-16:
Do you have a tool for printing the size of the reset handler stack frames?

hudson-ayers, 2021-03-16:
To print the stack frame sizes, I follow these steps:
1. Add `-Z emit-stack-sizes` to `RUSTC_FLAGS` in `boards/Makefile.common`
2. `make allboards`
3. `$ cd tock/target`
4. `$ shopt -s globstar` (makes recursive glob work on bash 4+)
5. `$ llvm-readelf --stack-sizes **/*.elf | grep -E '.*.elf|reset_handler'`

bradjc, 2023-06-21:
This seems to have regressed? Looking at `make stack-analysis` on current master:

```
❯ make stack-analysis
microbit_v2
----------------------
   main stack frame:
         1784     main

   5 largest stack frames:
         2408     _ZN7cortexm24kernel_hardfault_arm_v7m17h88793f49ade3c395E
         1784     main
          960     _ZN13capsules_core15process_console31ProcessConsole$LT$_$C$A$C$C$GT$11write_state17hf1a8799619718071E
          832     _ZN95_$LT$kernel..process_standard..ProcessStandard$LT$C$GT$$u20$as$u20$kernel..process..Process$GT$18print_full_process17haf8c56e314683244E
          672     _ZN117_$LT$capsules_core..process_console..ProcessConsole$LT$_$C$A$C$C$GT$$u20$as$u20$kernel..hil..uart..TransmitClient$GT$18transmitted_buffer17hb97c69890c491033E

```

While `main` is not the biggest (because for some reason now the fault handler is), it is very large. From older PRs, it looks like the microbit main should be around ~600 bytes of stack. Also, the create peripherals function is not on this list, and it looks like it is ~200 bytes of stack.

I tried going back in time to just before various upgrades of rustc, but I couldn't find a commit where the stack usage looked like it used to (ie main in second with get_periph first) and then `make stack-analysis` stopped working.

```
❯ make stack-analysis | grep create
          288     _ZN11microbit_v218create_peripherals17hebd30012208189deE
```

```
❯ git co 9496135f48f4c1755da1d120f51eaf2522000655
Note: switching to '9496135f48f4c1755da1d120f51eaf2522000655'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 9496135f4 Merge #3317
❯ make stack-analysis
microbit_v2
----------------------
   main stack frame:
         1536     main

   5 largest stack frames:
         1536     main
          712     _ZN13capsules_core15process_console31ProcessConsole$LT$_$C$A$C$C$GT$11write_state17h5afdb94c9c89b668E
          680     _ZN117_$LT$capsules_core..process_console..ProcessConsole$LT$_$C$A$C$C$GT$$u20$as$u20$kernel..hil..uart..TransmitClient$GT$18transmitted_buffer17h4e3b271955154ccaE
          608     _ZN7cortexm24kernel_hardfault_arm_v7m17h0be95cf01a9033f3E
          520     _ZN95_$LT$kernel..process_standard..ProcessStandard$LT$C$GT$$u20$as$u20$kernel..process..Process$GT$11try_restart17hde572bf7c2d45cafE

   WARNING! main is the largest stack frame!
   See https://github.com/tock/tock/issues/2425 for an explanation of
   why this is an issue, and https://github.com/tock/tock/pull/2715 for
   an example of how to fix it.

❯ cat ../../rust-toolchain
nightly-2022-10-22
❯ git log ../../rust-toolchain
❯ git co b6e2ddb0ed063894f4e3e0a446ce028267c6f745
Previous HEAD position was 9496135f4 Merge #3317
[... 102 lines cut ...]
HEAD is now at b0a08ec24 Merge #2856
❯ make stack-analysis
microbit_v2
----------------------
   main stack frame:
         1304     main

   5 largest stack frames:
         1744     _ZN11microbit_v215get_peripherals17h6b5b0b38f88f2affE
         1304     main
          704     _ZN8capsules15process_console27ProcessConsole$LT$A$C$C$GT$11write_state17h73e5852c154a1671E
          640     _ZN108_$LT$capsules..process_console..ProcessConsole$LT$A$C$C$GT$$u20$as$u20$kernel..hil..uart..TransmitClient$GT$18transmitted_buffer17h984908d8ef1d313eE
          608     _ZN7cortexm24kernel_hardfault_arm_v7m17h9ab982dacb7bda30E

❯ git log ../../rust-toolchain
❯ git co af7502c03fe094ed6f9940923d00470d35d519cc
Previous HEAD position was b0a08ec24 Merge #2856
HEAD is now at af7502c03 Merge #2804
❯ make stack-analysis
microbit_v2
----------------------
make: *** [stack-analysis] Error 2
```

I suspect this is related to #3478.

bradjc, 2023-06-21:
I found an old comment showing what the stack frame size looked like at one point, we seem to be nowhere near that now: https://github.com/tock/tock/pull/2503#issuecomment-816801396

> ```
> microbit_v2
> ----------------------
> reset_handler stack frame:
>
> 5 largest stack frames:
 >        1712     _ZN11microbit_v215get_peripherals17h3b4c27a2b28b64b4E
>           608     _ZN7cortexm24kernel_hardfault_arm_v7m17hf8c557e3e1a7d761E
>           608     _ZN6kernel7process14load_processes17hc18cff78f42d2f2dE
>           568     _ZN82_$LT$kernel..process..Process$LT$C$GT$$u20$as$u20$kernel..process..ProcessType$GT$11try_restart17hb6f2e8c6d2b2887bE
>           296     _ZN6kernel5grant21AppliedGrant$LT$T$GT$15get_or_allocate17hae61da99ed6c3574E
> ```

bradjc, 2023-09-05:
Moved to https://github.com/tock/tock/issues/3518

closed 2023-09-05 by bradjc