tock/tock #2426: Tock 2.0: Update Changelog

bradjc, 2021-02-12:
For the upcoming release we need to update `CHANGELOG.md`.

hudson-ayers, 2021-02-12:
I think it would be nice to have a single place with a list of all the things we need done as part of the release. AFAIK all there really is just the list of capsules that need to be changed, but there are a bunch of other things like:

- update changelog
- remove SuccessWithValue
- testing certain things at release time
- updating all the docs
- removing the legacy system call interface in the kernel, and updating all the boards to reflect this
- removing the legacy system call interface in userspace
- renaming "command2, subscribe2" etc. to command, subscribe in userspace
- probably much more .... would be good to have a central place with a list like this

hudson-ayers, 2021-02-12:
I'm gonna go ahead and make a new issue for this ^

phil-levis, 2021-02-15:
#2429

closed 2021-02-15 by phil-levis