tock/tock #2429: Tock 2.0: release blocker tracking

hudson-ayers, 2021-02-12:
This issue serves as a list of all tasks that need to be completed before an official Tock 2.0 release can be made. Please edit this list to add new tasks when they come up, or to check off tasks once they are completed. I am not including any tasks completed before I created this issue.

- [ ] #2235 port all system call drivers to new API
- [ ] #1796 remove guarantees on memory size for userspace apps
- [ ] #2270 -- prevent callbacks from violating appslice / callback swap invariants
- [ ] remove all uses of and references to `SuccessWithValue`
- [ ] update `docs/`
- [ ] update the tock book to reflect tock 2.0 system calls
- [ ] remove the legacy system call interface in the kernel, and update all the boards to reflect this
- [ ] remove the legacy system call interface in userspace
- [ ] rename "command2, subscribe2" etc. to command, subscribe in `libtock-c`
- [ ] Update callbacks to pass errorcodes instead of returncodes (non-blocking -- see #2320)
- [ ] update changelog
- [ ] merge master into `tock-2.0-dev` and halt all development on `master`. create a release candidate for testing, test thoroughly, including any capsules not tested when originally merged into `tock-2.0-dev`
- [ ] replace `master` with `tock-2.0-dev`

I am not putting this on #1607 because that issue was more "what should we do for Tock 2.0" and has lots of irrelevant discussion to this.

cc @phil-levis

bradjc, 2021-03-10:
What version of rustc do we want to use? Stick with nightly-2021-01-07? Or, we can probably update now and get plenty of testing in.

hudson-ayers, 2021-03-10:
Ooo if we update now we can use https://github.com/rust-lang/rust/pull/79208#event-4433248803

bradjc, 2021-04-30:
TRD104 (https://github.com/tock/tock/blob/master/doc/reference/trd104-syscalls.md#44-read-write-allow-class-id-3) says

> When userspace shares a buffer, it can no longer access it.

How are we enforcing that? Can a capsule rely on that?

lschuermann, 2021-04-30:
> How are we enforcing that? Can a capsule rely on that?

As far as I know, we don't and capsules can't. It might be worth putting a notice in there -- though I'm not sure of the precise target audience of this document. If it's only a reference to the system call interface from purely a userspace perspective, it might be fine just how it is now.

phil-levis, 2021-05-05:
I think we should remove #1143 -- this is an internal implementation, rather than an interface question. We can do it after 2.0.

hudson-ayers, 2021-07-09:
Per the discussion on the core call today, I wanted to organize release testing. No one should actually start testing yet -- there are a couple important PRs still outstanding -- but this list will help us get an idea of what platforms we have testers for and which we do not.

For boards which I believe few people have access to hardware for, I tagged the person who contributed the board to the repo -- please reply indicating you can in fact test that board, and I will give it a check.

For other boards, please sign up if you can, by editing this comment or replying that you can test a given board.

@tock/core-wg @tock/opentitan-wg 

General release testing / board-specific tests:

- [ ] acd52832
- [ ] arty_e21
- [ ] clue_nrf52840 
- [ ] earlgrey-nexysvideo
- [ ] hail
- [x] imix (@hudson-ayers )
- [ ] imxrt1050-evkb (@valexandru )
- [ ] litex/sim (@lschuermann )
- [ ] litex/fpga (@lschuermann )
- [ ] microbit_v2
- [ ] msp_exp432p401r (@lebakassemmerl )
- [ ] nano33ble
- [x] nrf52dk (@hudson-ayers )
- [ ] nrf52840dk
- [ ] nrf52840_dongle
- [ ] nucleo_f429zi
- [ ] nucleo_f446re
- [ ] raspberry pi pico
- [ ] redboard artemis nano
- [ ] stm32f3discovery
- [ ] stm32f412gdiscovery
- [ ] swervolf (@alistair23 )
- [ ] teensy40 (@mciantyre )
- [ ] weact_f401ccu6 (@yusefkarim )

Additional 2.0 specific testing:
- [ ] OT QEMU tests (@alistair23 )
- [ ] HMAC testing
- [ ] CTAP testing
- [ ] any other untested 2.0 capsules?

I will post the full testing regiment once 2.0 is ready-to-go.

mciantyre, 2021-07-10:
I'm available for Teensy 4.0 testing.

alexandruradovici, 2021-07-10:
I am available to test the Clue, Microbit, STMs, Raspberry Pi and Nano 33

alistair23, 2021-07-12:
I can test: OT, Redboard and SweRVolf. I have also already tested HMAC (but can test again) and OT QEMU.

The CTAP test I can also do but might be stuck on libtock-c support.

alexandruradovici, 2021-07-16:
I can test the Nucleo 429zi also.

alexandruradovici, 2021-07-19:
We got an nucleo_f446re, so we can test that too.

valexandru, 2021-07-19:
I can test the imxrt1050-evkb.

alexandruradovici, 2021-07-19:
We got Arty A7 FPGA, we think we can use it to test arty_21.

lschuermann, 2021-07-20:
> We got Arty A7 FPGA, we think we can use it to test arty_21.

Be aware that, as far as I know, you're going to need the Arty A7-100T. I also have an Arty A7-35T, and use that for LiteX, but it probably won't work for the `arty_21`.

yusefkarim, 2021-07-26:
I can test the weact_f401ccu6.

lschuermann, 2021-07-28:
I'm not sure whether this is on anyone's radar, but we might want to remove the Tock 2.0 breaking changes warning from the README prior to tagging a release, or at least rewording it to say the switch already happened (currently says "in progress"): https://github.com/tock/tock#tock-20-breaking-changes

bradjc, 2021-07-30:
Who is going to update the changelog file? Should we just delete it? It's already out of date.

bradjc, 2021-08-02:
Testing template. Please paste this to a new comment and mark off tests as they are confirmed to work. Also feel free to add more tests, this is just a pretty good starting point. And note, not all tests are expected to work on all boards.

```
# <Board Name>

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
```

bradjc, 2021-08-02:
# Hail

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`
  - [ ] `examples/tests/hail` and `examples/sensors`

alistair23, 2021-08-04:
libtock-c is failing to build due to https://github.com/tock/libtock-c/issues/217

alexandruradovici, 2021-08-04:
Will modify this today.

phil-levis, 2021-08-06:
I can update the changelog file.

brghena, 2021-08-06:
# Microbit v2

Tested with bootloader.

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

hudson-ayers, 2021-08-06:
If you want to look at what tests were run for your platform during the 1.6 release (to confirm regressions or just get a list of runnable tests), see here: https://github.com/tock/tock/issues/2030

hudson-ayers, 2021-08-06:
# Imix

- Tests run on most Boards
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - Fails when run simultaneously (expected: console not virtualized)
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

 
- Imix-specific
  - [ ] `examples/tests/imix`
  - [ ] `examples/ble-uart`
  - [ ] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - [ ] `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  -  [ ] **In-Kernel Tests**
      - [ ] `boards/imix/src/test/aes_ccm_test.rs`
      - [ ] `boards/imix/src/test/aes_test.rs`
      - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
      - [ ] `boards/imix/src/test/linear_log_test.rs`
      - [ ] `boards/imix/src/test/log_test.rs`
      - [ ] `boards/imix/src/test/rng_test.rs`
      - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
          - [ ] `start()`
          - [ ] `start_rx()`
          - [ ] `start_with_app()`
          - [ ] `start_dual_rx()`
      - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`
      - [ ] `boards/imix/src/test/multi_alarm_test.rs`

- Miscellaneous
  - [ ] **Scheduler Tests**
    - [ ] `whileone` + `blink`
    - [ ] restart `whileone` using process console
    - [ ] stop and start `whileone`
    - [ ] `whileone` + `blink` + `hello_loop` using MLFQ scheduler

alexandruradovici, 2021-08-07:
# Arty e21

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

alexandruradovici, 2021-08-07:
# Clue nrf52840

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/screen`
  - [ ] `examples/lvgl`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

lschuermann, 2021-08-09:
# LiteX on Arty A7-35T

Tests are performed on the pinned [`tock-litex` release 2021072001](https://github.com/lschuermann/tock-litex/releases/tag/2021072001), with the Verilog source and bitstream available for download [here](https://github.com/lschuermann/tock-litex/releases/download/2021072001/digilent_arty_a7-35t.zip).

Required non-PIC Tock target definition for libtock-c, first app:
```
rv32imc|rv32imc.0x41000060.0x42004000|0x41000060|0x42004000
```

Tested app upload through both serial boot (`litex_term --speed 1000000 --kernel $KERNEL_BINARY /dev/$TTY_DEV`) and network boot via TFTP server hosting `$KERNEL_BINARY` as `/boot.bin` on `192.168.1.100/24`.

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

lschuermann, 2021-08-09:
# LiteX Simulator

~Tests are performed on the pinned [`tock-litex` release 2021072001](https://github.com/lschuermann/tock-litex/releases/tag/2021072001), with the Verilog source and bitstream available for download [here](https://github.com/lschuermann/tock-litex/releases/download/2021072001/digilent_arty_a7-35t.zip).~

*The pinned tock-litex release pins a VexRiscv CPU version which has a broken PMP implementation with TOR addressing unimplemented. These tests are performed on a local tock-litex version with an older VexRiscv version. Once I'm somewhat confident that the PMP works completely on this version, I'll publish a new tock-litex release and PR to Tock.*

Required non-PIC Tock target definition for libtock-c, first app:
```
rv32i|rv32i.0x00080060.0x40004000|0x00080060|0x40004000
```

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
  - [x] `examples/tests/mpu` (tested all cases one-by-one, all permitted accesses work, all invalid accesses fault)

alistair23, 2021-08-10:
- [ ] earlgrey-nexysvideo
  - [ ] Unit tests
  - [ ] Board/app tests
- [ ] redboard artemis
  - [ ] Board/app tests
- [X] swervolf
  - [X] Manual tests
- [ ] ESP32-C3
  - [ ] Unit tests
- [X] OT QEMU tests
- [X] HMAC testing
- [ ] CTAP testing -- blocked on libtock-rs

irina-b-dev, 2021-08-10:
# Raspberry Pi Pico

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

brghena, 2021-08-11:
I just realized that I don't actually have the right hardware to program nRF52840 dongles. I forgot that they need a 10-pin tagconnect programmer. Does anyone else have the right hardware to test them? It shouldn't be of very high importance, since multiple other nRF52 platforms have been successfully tested already.

ppannuto, 2021-08-12:
I have the 10-pin programmer in my office. I'm OoO until Monday of next week, but hopefully should be a quick run through to test things on the dongle if all the other nRF boards are cleared by then.

lschuermann, 2021-08-12:
```
# nrf52840dk

@ppannuto I'll be sure to have the nRF52840DK checked by then.

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
```

hudson-ayers, 2021-08-12:
# nrf52dk

* All Boards
  * [ ]  `examples/sensors`
  * [ ]  `examples/c_hello` and `examples/tests/printf_long`
  * [ ]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    Only individually - as expected
  * [ ]  `examples/blink`
  * [ ]  `examples/rot_client` and `examples/rot_service`
  * [ ]  `examples/blink` and `examples/c_hello` and `examples/buttons`
  * [ ]  `examples/lua-hello`
  * [ ]  `examples/tests/console_timeout`
  * [ ]  `examples/tests/malloc_test01`
  * [ ]  `examples/tests/stack_size_test01`
  * [ ]  `examples/tests/stack_size_test02`
  * [ ]  `examples/tests/mpu_stack_growth`
  * [ ]  `examples/tests/mpu_walk_region`
     regular loop does not crash, button presses crash both reads as expected
  * [ ]  `examples/tests/multi_alarm_test`
     slow, peaceful blinking of each LED in order
  * [ ]  `examples/tests/adc`
     `No ADC driver!`
  * [ ]  `examples/tests/adc_continuous`
     `No ADC driver!`
  * [ ]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
     `Number of LEDs: 4` .. then each LED randomly turns on and off
  * [ ]  `examples/tests/gpio` with mode set to 0
    Touch `P0.03` to Vdd (randomly tested with a few other pins to _not_ fire an interrupt):
 
* nRF specific
  * [ ]  `examples/ble_advertising`
  * [ ]  `examples/ble_passive_scanning`

hudson-ayers, 2021-08-12:
@mciantyre @yusefkarim @lebakassemmerl 

Several platforms have passed all tests at this point, so it should be safe to start testing your platforms when you have the chance! Just checkout release-2.0-rc2 and libtock-c master, and run/report the release tests as shown above.

Thank you!

alexandruradovici, 2021-08-13:
# Nucleo F429ZI

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

alexandruradovici, 2021-08-14:
# Nano 33 BLE

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

yusefkarim, 2021-08-14:
# WeAct STM32F401CCU6
I have not completed tests requiring multiple apps flashed at once. I will try to find time to those soon as well.

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] ~~`examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`~~
    - ~~sense_to_ble: Application that samples a sensor and advertises value with BLE~~
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
    - NOTE: Each work independently, must still test together
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - NOTE: Each work independently, must still test together
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
    - NOTE: Each work independently, must still test together
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - NOTE: Several channels gave back `Error sampling ADC: -10`, but others were successful (e.g. single sampling on Channel 5). This likely might just be a hardware thing, I have little knowledge on ADCs. My initial support of the  weact board (kernel) may have incorrectly configured ADC (I might have time in a few weeks to look into this).
  - [x] `examples/tests/adc_continuous`
    - NOTE: received `set buffer error: -10`. Again, might just be hardware specific thing (I did try changing `ADC_CHANNEL` but any number resulted in the same error).
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] ❌ `examples/tests/gpio` with mode set to 0
    - NOTE: weact does not have pin 0, so I instead chose pin 11. Regardless, when `mode = 0`, nothing gets printed when I jump the pin high. I tried `mode = 2` and that works, so something likely wrong with interrupt configuration?
  - [x] `examples/tests/whileone`

mciantyre, 2021-08-14:
[Placeholder comment for Teensy 4 test results. I'm traveling without hardware; can test within a week.]

alexandruradovici, 2021-08-15:
>         * NOTE: weact does not have pin 0, so I instead chose pin 11. Regardless, when `mode = 0`, nothing gets printed when I jump the pin high. I tried `mode = 2` and that works, so something likely wrong with interrupt configuration?
I have the same issue with the STM boards, I will take a look at it

valexandru, 2021-08-16:
# i.MX RT 1052 EVKB

- Tests
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and  ❌ `examples/buttons`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth` 
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0 ?? — cred ca nu 
  - [x] `examples/tests/whileone`

hudson-ayers, 2021-08-16:
@yusefkarim don't worry about testing `console_recv_short` and `console_recv_long` together, we expect them to fail when flashed together,

alexandruradovici, 2021-08-16:
# STM32F412G Discovery

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with `[SHORT] Error doing UART receive: -2`
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/screen`
  - [x] `examples/lvgl`
  - [x] `examples/touch` (requires #2784)
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
  - [x] `examples/tests/adc_continuous`
    - Fails as continuous sampling is not supported
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/tests/gpio` with mode set to 0 (requires #2784)
  - [x] `examples/tests/whileone`

mciantyre, 2021-08-17:
# Teensy 4.0

- [x] `examples/c_hello` and `examples/tests/printf_long`
  - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
- [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
  - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - *Passed when run separately; known issues when running concurrently.*
- [x] `examples/blink`
- [x] `examples/rot13_client` and `examples/rot13_service`
  - *Requires #2779, otherwise panics.*
- [x] `examples/blink` and `examples/c_hello` ~~and `examples/buttons`~~
  - *No buttons available at the moment.*
- [x] `examples/lua-hello`
- [x] `examples/tests/console_timeout`
- [x] `examples/tests/malloc_test01`
- [x] `examples/tests/stack_size_test01`
- [x] `examples/tests/stack_size_test02`
- [x] `examples/tests/mpu_stack_growth`
- [x] `examples/tests/mpu_walk_region`
  - *Didn't press a button to force overrun.*
- [x] `examples/tests/multi_alarm_test`
  - *Only one configured LED on board; combined with `whileone` to make it interesting.*
- [x] `examples/tests/whileone`

Tested on Tock `release-2.0-rc2` tag, libtock-c `0a4eadbd3`.

alexandruradovici, 2021-08-17:
# STM32F303 Discovery

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
    - temperature is always 0
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
    - Fails with `[LONG] Error doing UART receive: -2`
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
     - has no `rng ` peripheral driver, but the service and applications seems to work
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

bradjc, 2021-08-25:
# 2.0 Sign-off

- [ ] @hudson-ayers
- [ ] @bradjc
- [ ] @brghena
- [ ] @phil-levis
- [ ] @alevy
- [ ] @ppannuto
- [ ] @lschuermann
- [ ] @jrvanwhy

hudson-ayers, 2021-08-30:
https://github.com/tock/tock/releases/tag/release-2.0  🎉

closed 2021-08-30 by hudson-ayers