tock/tock #2438: Tock panicked with LoadFault with OpenTitan 99cb19827 bitstream

gemarcano, 2021-02-17:
I built the OpenTitan FPGA bitstream using Vivado 2020.2 and the OpenTitan commit 99cb19827 (for use on the Nexys Video FPGA board). I then attempted to flash the earlgrey board Tock build (commit e5c422ba23) by setting up the environment and invoking `make flash` from the earlgrey-nexys video board folder in Tock. I am attaching the output logs to this issue, showing a LoadFault at address 0x20030001. Just in case, I'm also attaching the bitstream I built.

Output log: [opentitan-99cb19827-tock-panic.log](https://github.com/tock/tock/files/5998491/opentitan-99cb19827-tock-panic.log).
Bitstream I built: [lowrisc_systems_top_earlgrey_nexysvideo_0.1.zip](https://github.com/tock/tock/files/5998537/lowrisc_systems_top_earlgrey_nexysvideo_0.1.zip)

I'll try turning on tracing and debugging to see if I can get any more useful information.

gemarcano, 2021-02-18:
I tried re-building the bitstream with Vivado 2020.1, but I still see the same issue.

As an experiment, I tried uploading an image to the board with an application. If I do this, there isn't a crash... but nothing happens also (no entering main loop message). This is looking to be a problem in the bitstream generation, but I'd have to debug Tock somehow just to see how far it gets and what causes it to fail to work with my bitstream.

alistair23, 2021-02-18:
As discussed on Slack this is probably a bitstream generation issue and should be raised with lowRISC/opentitan

gemarcano, 2021-02-18:
The OpenTitan folk told me the reduce size script for OpenTitan wasn't necessary for this commit, so I built one more bitstream without running this script beforehand, and it appears to work!

Should we document this somewhere, at least while we're stuck on OpenTitan commit 99cb19827?

alistair23, 2021-02-18:
Great! I'm glad the issue has been figured out at least.

I would say that the documentation should probably be in upstream OT, but if you want to add something to the README in Tock just open a PR.

gemarcano, 2021-02-18:
OK. I know latest upstream does require running a newer version of the reducing script, and from what I can tell some work has been done to it since the commit we have to use for Tock, so it's unclear whether the bug is still there or not.

As for Tock, at least while we're on 99cb19827 OpenTitan, sure, I'll look into making a PR updating our documentation.

With that, I think I'll close this issue. Thanks for the help!

closed 2021-02-18 by gemarcano

alistair23, 2021-02-18:
Great! I'm glad it's working.

When I'm back from holidays I was planning on looking into fixes the SRAM scrambling issue and then updating the bitstream.