tock/tock #2536: Rename `kernel/` to `core/`?

ppannuto, 2021-04-16:
I was re-reading #1905's thread during the call, and I didn’t really want to interrupt the primary discussion for this, but from https://github.com/tock/tock/issues/1905#issuecomment-639522753:

> > I think so we can have a coherent discussion, we have to use the same terminology. From https://github.com/tock/tock/blob/master/doc/Design.md, everything under the syscall interface is the kernel. [...] An easy way to separate out capsules is to refer to the “core kernel”.

> Well, in practice I interact more with the “kernel” as in everything in the kernel/ folder, than with the design documentation. So it would make sense to apply this terminology to the code as well, and `git mv kernel core` so that the good terminology is clear for everyone.

I think this suggestion might’ve gotten lost at the time, but it is actually a pretty good one, and 2.0 is a good opportunity for a big rename that would otherwise be noise in git history. Thoughts?

lschuermann, 2021-04-16:
This is an interesting proposal, because it on the one hand seems natural, given that `/kernel` is widely referred to the core kernel already. I do however think that `/chips` and `/arch` must equally be counted to the core kernel (at least w.r.t. to the platform they're targeting)? Having a `core` crate would (rightfully?) elevate that and might also _cause_ some confusion? I'm not opposed to this, it's just rather interesting.

In any case, I think this could be a change that should probably be applied close to releasing Tock 2.0, to not "unnecessarily" break the current development efforts (given it's suspected low implementation complexity of `git mv` and some import renaming).

jrvanwhy, 2021-04-19:
I don't think we want to call the crate `core`, as that would overlap with Rust's `core` library. Maybe `tock_core`?

bradjc, 2021-04-19:
For the core team, the folder/crate name doesn't really matter. We know what is there. People new to tock are likely to find more value in the names, and I think "core" provides little value, as every project can have core code. However, "kernel" is at least somewhat descriptive, and a new user shouldn't be surprised to find the scheduler and syscall handling there. I'm in favor of leaving it.

alevy, 2021-05-02:
What about calling it "tock"?

There isn't actually a "tock" crate atm

bradjc, 2021-05-03:
> What about calling it "tock"? There isn't actually a "tock" crate atm

Wouldn't that add to the confusion? Do you mean "tock", the org, repo, OS, or crate?

bradjc, 2021-07-06:
Should we close this?

lschuermann, 2021-07-06:
I think `kernel` seems like a good name.

closed 2021-07-06 by brghena